esphome/issues #10: Unable to connect to wifi with sensor in configuration

DonHugo, 2019-01-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266/D1_min

**Affected component:**
Wifi

**Description of problem:**
if i have a configuration with only wifi and mqtt settings it works, but as soon as i add any sensor it cant connect to the wifi anymore.

first logentry is from executing first yaml configuration and second logentry is from executing second yaml configuration

**Problem-relevant YAML-configuration entries:**
```yaml
#Does not work
esphomeyaml:
  name: varmevaxlare
  platform: ESP8266
  board: d1_mini
  arduino_version: latest
  esphomelib_version: latest
  

wifi:
  ssid: 'Parlan'
  password: 'password'
  hostname: 'D1_mini'

# Enable logging
logger:
  level: DEBUG

mqtt:
  broker: '192.168.0.111'
  username: 'name'
  password: 'password'
  topic_prefix: 'test/esphome/1'
  
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 10s

```

```yaml
#This one works
esphomeyaml:
  name: varmevaxlare
  platform: ESP8266
  board: d1_mini
  arduino_version: latest
  esphomelib_version: latest
  

wifi:
  ssid: 'Parlan'
  password: 'password'
  hostname: 'D1_mini'

# Enable logging
logger:
  level: DEBUG

mqtt:
  broker: '192.168.0.111'
  username: 'name'
  password: 'password'
  topic_prefix: 'test/esphome/1'
```
**Traceback (if applicable):**

```
Checking size .pioenvs/varmevaxlare/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  45.2% (used 37040 bytes from 81920 bytes)
PROGRAM: [===       ]  31.2% (used 325884 bytes from 1044464 bytes)
======================== [SUCCESS] Took 308.36 seconds ========================
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB1 write_flash 0x0 /config/esphomeyaml/varmevaxlare/.pioenvs/varmevaxlare/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:27:01:b0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 330032 bytes to 230411...
Wrote 330032 bytes (230411 compressed) at 0x00000000 in 20.8 seconds (effective 126.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[10:46:48]�b��oo�lgg���cx��drd{lp�n��l��cn�|l��c��nn�l��d`�\033ngl`g{���nc�d{��nc�d�ccc���l`��o�[I][logger:071]: Log initialized
[10:46:48][I][application:053]: Running through setup()...
[10:46:48][C][wifi:029]: Setting up WiFi...
[10:46:48][D][wifi:260]: Starting scan...
[10:46:57]
[10:46:57] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[10:46:57]
[10:46:57]wdt reset
[10:46:57]load 0x4010f000, len 1384, room 16 
[10:46:57]tail 8
[10:46:57]chksum 0x2d
[10:46:57]csum 0x2d
[10:46:57]vbb28d4a3
[10:46:57]~ld
[10:46:57][I][logger:071]: Log initialized
[10:46:57][I][application:053]: Running through setup()...
[10:46:57][C][wifi:029]: Setting up WiFi...
[10:46:57][D][wifi:260]: Starting scan...
```

```
[17:48:59][I][application:053]: Running through setup()...
[17:48:59][C][wifi:029]: Setting up WiFi...
[17:48:59][D][wifi:260]: Starting scan...
[17:49:05][D][wifi:275]: Found networks:
[17:49:05][I][wifi:308]: - 'Parlan' (40:16:7E:EB:46:48) ▂▄▆█
[17:49:05][D][wifi:309]:     Channel: 11
[17:49:05][D][wifi:310]:     RSSI: -32 dB
[17:49:05][D][wifi:312]: - '' (FA:8F:CA:63:F0:AF) ▂▄▆█
[17:49:05][I][wifi:179]: WiFi Connecting to 'Parlan'...
[17:49:06][I][wifi:349]: WiFi connected!
[17:49:06][C][wifi:240]:   SSID: 'Parlan'
[17:49:06][C][wifi:241]:   IP Address: 192.168.0.192
[17:49:06][C][wifi:243]:   BSSID: 40:16:7E:EB:46:48
[17:49:06][C][wifi:245]:   Hostname: 'D1_mini'
[17:49:06][C][wifi:250]:   Signal strength: -36 dB ▂▄▆█
[17:49:06][C][wifi:251]:   Channel: 11
[17:49:06][C][wifi:252]:   Subnet: 255.255.255.0
[17:49:06][C][wifi:253]:   Gateway: 192.168.0.1
[17:49:06][C][wifi:254]:   DNS1: 192.168.0.129
[17:49:06][C][wifi:255]:   DNS2: 192.168.0.1
[17:49:06][C][mqtt.client:024]: Setting up MQTT...
[17:49:06][I][mqtt.client:170]: Connecting to MQTT...
[17:49:06][I][mqtt.client:209]: MQTT Connected!
[17:49:06][I][application:089]: setup() finished successfully!
[17:49:06][I][application:097]: You're running esphomelib v1.10.1 compiled on Jan 21 2019, 17:43:49
[17:49:06][C][wifi:341]: WiFi:
[17:49:06][C][wifi:240]:   SSID: 'Parlan'
[17:49:06][C][wifi:241]:   IP Address: 192.168.0.192
[17:49:06][C][wifi:243]:   BSSID: 40:16:7E:EB:46:48
[17:49:06][C][wifi:245]:   Hostname: 'D1_mini'
[17:49:06][C][wifi:250]:   Signal strength: -36 dB ▂▄▆█
[17:49:06][C][wifi:251]:   Channel: 11
[17:49:06][C][wifi:252]:   Subnet: 255.255.255.0
[17:49:06][C][wifi:253]:   Gateway: 192.168.0.1
[17:49:06][C][wifi:254]:   DNS1: 192.168.0.129
[17:49:06][C][wifi:255]:   DNS2: 192.168.0.1
[17:49:06][C][logger:099]: Logger:
[17:49:06][C][logger:100]:   Level: DEBUG
[17:49:06][C][logger:101]:   Log Baud Rate: 115200
[17:49:06][C][mqtt.client:058]: MQTT:
[17:49:06][C][mqtt.client:060]:   Server Address: 192.168.0.111:1883 (192.168.0.111)
[17:49:06][C][mqtt.client:061]:   Username: 'hugo'
[17:49:06][C][mqtt.client:062]:   Client ID: 'varmevaxlare-b4e62d2701b0'
[17:49:06][C][mqtt.client:064]:   Discovery prefix: 'homeassistant'
[17:49:06][C][mqtt.client:065]:   Discovery retain: YES
[17:49:06][C][mqtt.client:067]:   Topic Prefix: 'test/esphome/1'
[17:49:06][C][mqtt.client:069]:   Log Topic: 'test/esphome/1/debug'
[17:49:06][C][mqtt.client:072]:   Availability: 'test/esphome/1/status'
[17:49:06][I][application:114]: Running through first loop()
[17:49:06][I][application:141]: First loop finished successfully!
```

**Additional information:**

OttoWinter, 2019-03-24:
Closing due to inactivity

closed 2019-03-24 by OttoWinter

DonHugo, 2019-03-26:
Hey @OttoWinter , i still have this problem with my D1 mini, even after upgrading to latest version(1.12.1). do you need any more information from me to troubleshoot?
i would really like if this issue was re-opend again!

nowismine, 2019-04-10:
Hey All,

Just wanted to confirm that I saw the same issue (interestingly only on Basic, but not the Dual). Re flashed the yaml without the sensor then with sensor again and it started working.

Cheers

DonHugo, 2019-04-11:
Hey @nowismine, i have trid that as well with no luck. did you use the API or MQTT?
what sensors did you test with?

nowismine, 2019-04-12:
@DonHugo - Using Api, mqtt is not enabled, ota enabled, with a dht sensor.

 I have seen the issue at some point on an esp-12e, but dismissed it as something related to the fact I was using my mobile as a hotspot to test.

I am wondering if it’s not something that’s caused by an issue in compilation or flashing. I’ll try a few more tests and let you know.