esphome/issues #15: BME680 reports bad pressure value

thomaskr, 2019-02-01:
# Description

Esphomeyaml reports unreasonably low pressure value retrieved via esphomelib from Bosch BME680 sensor.

The pressure value is reported as 740 hPa while a value of 993 hPa is expected.

# Environment

esphomeyaml==1.9.3, Python 2.7.15rc1, Ubuntu 18.04 x86_64, [MASL](https://en.wikipedia.org/wiki/Metres_above_sea_level): 65

Dependency Graph
```
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <Hash> v1.0
|-- <Wire> v1.0
|-- <esphomelib> v1.9.3 #9d5e290
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Hash> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <FastLED> v3.2.0
|   |   |-- <EspSoftwareSerial> v3.4.1
|   |-- <Wire> v1.0
|   |-- <ArduinoOTA> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
```

# Log output

```
[10:12:20][VV][component:132]: Running interval 'update':0 with interval=15000 last_execution=75656 (now=90663)
[10:12:20][VV][i2c:073]: Beginning Transmission to 0x77:
[10:12:20][VV][i2c:111]:     Writing 0b01110100 (0x74)
[10:12:20][VV][i2c:111]:     Writing 0b10100001 (0xA1)
[10:12:20][VV][i2c:078]:     Transmission ended. Status code: 0x00
[10:12:20][VV][component:079]: set_timeout(name='data', timeout=250)
[10:12:20][VV][component:132]: Running timeout 'data':1 with interval=250 last_execution=90684 (now=90950)
[10:12:20][VV][i2c:073]: Beginning Transmission to 0x77:
[10:12:20][VV][i2c:111]:     Writing 0b00011101 (0x1D)
[10:12:20][VV][i2c:078]:     Transmission ended. Status code: 0x00
[10:12:20][VV][i2c:100]: Requesting 15 bytes from 0x77:
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b01110101 (0x75)
[10:12:20][VV][i2c:132]:     Received 0b01111000 (0x78)
[10:12:20][VV][i2c:132]:     Received 0b11010000 (0xD0)
[10:12:20][VV][i2c:132]:     Received 0b01011111 (0x5F)
[10:12:20][VV][i2c:132]:     Received 0b01100010 (0x62)
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b01001010 (0x4A)
[10:12:20][VV][i2c:132]:     Received 0b11111001 (0xF9)
[10:12:20][D][sensor.bme680:313]: Got temperature=19.5°C pressure=740.0hPa humidity=59.9% gas_resistance=18588.0Ω
[10:12:20][V][sensor.sensor:020]: 'Temperatur Flur Komode': Received new state 19.519823
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff3234)::input(19.519823)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff3234)::new_value(19.519823) -> 19.520716
[10:12:20][V][sensor.sensor:020]: 'Luftdruck': Received new state 739.966492
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff34fc)::input(739.966492)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff34fc)::new_value(739.966492) -> 739.967468
[10:12:20][V][sensor.sensor:020]: 'Luftfeuchtigkeit Flur': Received new state 59.895885
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff37c4)::input(59.895885)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff37c4)::new_value(59.895885) -> 59.895073
[10:12:20][V][sensor.sensor:020]: 'Gassensor Flur': Received new state 18588.000000
[10:12:20][D][sensor.sensor:126]: 'Gassensor Flur': Sending state 18588.00000 with 0 decimals of accuracy
[10:12:20][V][mqtt.client:239]: Publish(topic='esp_hallway/sensor/gassensor_flur/state' payload='18588' retain=1)

```

andreas-bulling, 2019-02-02:
same problem here

S3NS375, 2019-03-05:
same problem here too. Has someone figured out something?
Can i do a workaround?

S3NS375, 2019-03-07:
Hello again ;)

Today i have figured out that the shown pressure looks identical to the temperature curve. Maybe a good hint to find the problem:

[image]

[image]

thomaskr, 2019-03-07:
Here they are very similar as well:

[image]

james-davis-protonmail, 2019-03-12:
Hi, I have also (many) issues with the BME680. I bought three of them which I run on a izokee (wemos d1 mini clone).

They are all setup at the same location with the same configuration, however report different (crazy) values. This is the configuration which I use for the three of them:

```
substitutions:
  devicename: izokee_1
  ip: 192.168.0.203
esphome:
  name: ${devicename}
  platform: ESP8266
  board: d1_mini
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: ${ip}
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.50
i2c:
  sda: 4
  scl: 5
  scan: False
sensor:
  - platform: bme680
    temperature:
      name: ${devicename}_temperature
      filters:
        - filter_out: 100
        - filter_out: 0
        - filter_nan:
      oversampling: 2x
    pressure:
      name: ${devicename}_pressure
      filters:
        - filter_out: 100
        - filter_out: 0
        - filter_nan:
      oversampling: 2x
    humidity:
      name: ${devicename}_humidity
      filters:
        - filter_out: 100
        - filter_out: 0
        - filter_nan:
      oversampling: 2x
    gas_resistance:
      name: ${devicename}_gas_resistance
      filters:
        - filter_nan:
    address: 0x76
    update_interval: 90s
    iir_filter: 3x  

debug:

# Enable logging
logger:
  level: debug
# Enable Home Assistant API
api:
ota:
```

those are my horrible results :
[image]

Would you mind sharing your yaml conf with me so I can try it? I am specially interested in the oversampling values (tried none up to x16), the iir filter (tried OFF up to x127) and filters if you have any.

Thanks :)

thomaskr, 2019-03-13:
Your problems are not related to the original issue. Please stick to the topic or open a new issue.

gon0, 2019-03-18:
Same here. I am using Adafruit BME680 Library on TTGO V2 LoRa OLED ESP32 and on firebeetle ESP32. On both i saw drops of pressure, which recover somehow after some time. Also the BME680 cannot be read after some minutes or after some days and a new bme.begin() fails. I have china BME680 boards in purple (CJCMU-680), running via I2C. [image]

closed 2019-05-15 by OttoWinter