esphome/issues #19: remote_receiver error: Data is coming in too fast

spattinson, 2019-02-02:
**Env:** Home Assistant, 433Mhz generic binary window/door sensors. 
**Board:** Wemos D1 mini
**Component:** remote_receiver
**Description:** Code cannot receive codes using rc_switch, get:

```bash
[15:26:32][W][remote.receiver:357]: Data is coming in too fast!
```

This may be significant:

```bash
[15:26:18][W][remote.receiver:320]: Remote Receiver Signal starts with a HIGH value. Usually this 
means you have to invert the signal using 'inverted: True' !
```

Where is this supposed to be set?
The sample sketch for rc_switch ReceiveDemo_Advanced works fine
As does OpenMQTTGateway

```yaml
esphomeyaml:
  name: rf_bridge
  platform: ESP8266
  board: d1_mini
wifi:
  ssid: 'home'
  password: 'fileybay'

mqtt:
  broker: '192.168.1.122'
  username: 'scott'
  password: 'XXXXXXX'

# Enable logging
logger:

ota:
  password: 'fileybay'

remote_receiver:
  pin: D2
  dump: rc_switch
```

Log:

```bash
[15:26:18][I][logger:057]: Log initialized
[15:26:18][C][ota:465]: There have been 1 suspected unsuccessful boot attempts.
[15:26:18][I][application:053]: Running through setup()...
[15:26:18][C][remote.receiver:296]: Setting up Remote Receiver...
[15:26:18][C][wifi:027]: Setting up WiFi...
[15:26:18][I][wifi:279]: WiFi Connecting to 'home'...
[15:26:18][I][wifi:347]: WiFi connected!
[15:26:18][C][wifi:325]:   SSID: 'home'
[15:26:18][C][wifi:326]:   IP Address: 192.168.1.203
[15:26:18][C][wifi:328]:   BSSID: D8:50:E6:A8:1E:20
[15:26:18][C][wifi:330]:   Hostname: 'rf_bridge'
[15:26:18][C][wifi:332]:   Channel: 3
[15:26:18][C][wifi:333]:   Subnet: 255.255.255.0
[15:26:18][C][wifi:334]:   Gateway: 192.168.1.1
[15:26:18][C][wifi:335]:   DNS1: 192.168.1.1
[15:26:18][C][wifi:336]:   DNS2: 0.0.0.0
[15:26:18][C][ota:133]: Over-The-Air Updates:
[15:26:18][C][ota:134]:   Address: 192.168.1.203:8266
[15:26:18][C][ota:136]:   Using Password.
[15:26:18][C][mqtt.client:020]: Setting up MQTT...
[15:26:18][I][mqtt.client:079]: Connecting to MQTT...
[15:26:18][I][mqtt.client:115]: MQTT Connected!
[... 88 lines cut ...]
The sensor send each trigger about ten times
If I change my yaml to raw/all instead of rc_switch I get this with esphomeyaml

```bash
[13:32:43][D][remote.raw:051]: Received Raw: 1578, -470, 251, -1832, 426, -1665,
 366, -402, 285, -414, 170, -475, 184, -421, 269, -417, 265, -549, 269, -1823, 289, -1805, 275, -427, 260, -441, 238, -477, 51, -2043, 231, -1866, 229, -435, 256, -433, 252, -504, 249, -449, 250, -452, 
[13:32:43][D][remote.raw:064]:   245, -464, 245, -1860, 173, -1930, 164, -1942, 145, -478, 245, -464, 250, -524, 238, -468, 240, -475, 237, -469, 234, -1859, 102, -524, 231, -486, 235, -530, 230, -509, 236, -501, 189, -3245, 235, -505, 230, -510, 230, -10000
[13:32:43][D][remote.raw:051]: Received Raw: 184, -667, 245, -1014, 165, -4345, 245, -728, 242, -740, 240, -737, 228, -777, 224, -4455, 233, -817, 212, -5099, 227, -1427, 50, -1312, 226, -987, 181, -1633, 226, -5135, 214, -1179, 222, -1295, 175, -3487, 166, -4627, 145, -4184, 200, 
[13:32:43][D][remote.raw:064]:   -1807, 162, -10000
[13:32:43][W][remote.receiver:357]: Data is coming in too fast!
[13:32:43][D][remote.raw:064]: Received Raw: 96, -4345, 186, -10000
[13:32:43][D][remote.raw:051]: Received Raw: 77, -3667, 41, -91, 43, -380, 70, -504, 228, -613, 60, -537, 121, -716, 47, -523, 46, -427, 158, -455, 23, -284, 73, -101, 193, -103, 69, -122, 49, -168, 121, -28, 57, -457, 222, -77, 112, -509, 95, -72, 182, -41, 51, -169, 129, -428, 99, 
[13:32:44][D][remote.raw:051]:   -67, 159, -271, 94, -85, 172, -27, 128, -108, 50, -50, 145, -155, 158, -200, 97, -58, 44, -101, 161, -95, 57, -394, 457, -176, 502, -85, 178, -88, 1323, -58, 74, -67, 103, -34, 349, -51, 421, -32, 508, -65, 104, -74, 416, -98, 1567, -59, 13, -97, 1433, 
[13:32:44][D][remote.raw:064]:   -17, 10000
[13:32:44][W][remote.receiver:357]: Data is coming in too fast!
[13:32:44][D][remote.raw:064]: Received Raw: -26, 97, -56, 201, -52, 45, -37, 1141, -67, 320, -31, 206, -108, 18, -45, 10000
```

Things I have tried:

different GPIOs, I started off using D4 GPIO2, because it has an LED and I could see when data arrived, This worked with other code not esphomeyaml. I am currently using D2 GPIO4 - same issue.
swapped over rf receiver module: same issue
tried increasing the buffer, esp8266 seems to crash mostly if you set it greater than 2kb, still crashes sometimes with 2kb, but when it didn't crash I get the same issue.
Disabled  ota in case they caused overhead. No Difference
Thanks in advance. I'm moving from OPenMQTTGateway to esphomeyaml because I wish to standardise, have deployed one project.

HFsi, 2019-04-10:
I'm getting an error while compiling with the branch "remote-receiver"
Is it still active?

OttoWinter, 2019-04-25:
> I'm getting an error while compiling with the branch "remote-receiver"

No, it is not. In the latest esphome dev (not esphome-core dev, the python dev) though this should be gone. Incoming data that is coming in too fast is just discarded (and should happen less often due to misc improvements)

closed 2019-04-25 by OttoWinter

escoand, 2019-05-01:
I've similar problems and raw is constantly dumping useless data without even one silent second. With the dev branch it's not getting better but the real data isn't recognized better. Is there something we can do to help you?

OttoWinter, 2019-05-01:
> I've similar problems and raw is constantly dumping useless data without even one silent second.

Well if there's useless raw data on the wire it will be dumped. That's the whole point of raw. If the protocol is supported you can also enable all dumpers except for the raw one.

escoand, 2019-05-02:
The issue history implied some kind of generic RF or rc_switch problem to me and your mentioned branch did not solve my problem.
So the noise coming in is more a problem of my concrete setup, right?

matthijsv93, 2019-05-03:
I have the same problem. When set to only dump rcswitch decoded data, the only thing that is displayed is the error "data coming in too fast".  I'm using d1mini with RX470C 433mhz rf receiver.  When i use tasmo, I get the decoded data without any noise . However I do prefer to work with Esphome.

bphermansson, 2019-05-03:
Same problem here. The receivers have an AGC, if there's no strong signal they raise their gain to try to get a signal. So if you look at the signal in an oscilloscope there's a constant noise on the output. Is that why the buffer is filled up?

escoand, 2019-05-03:
So far so good. What is Tasmota (and others) doing to avoid the noise?

HFsi, 2019-05-04:
Did you flashed the RF chip or used the original firmware?

escoand, 2019-05-04:
@HFsi I myself have a XD-RF-5V. But if different modules have the same problem with ESPHome there seems to be something strange going on.
I've used my module successfully with the 433utils.

Gastondamian, 2019-05-04:
I have the same issue with XD-RF-5V receiver. I will try the module with OpenMQTTGateway.

Gastondamian, 2019-05-04:
I uploaded this example https://github.com/sui77/rc-switch/tree/master/examples/ReceiveDemo_Advanced from RCSwitch to my nodemcu

It works perfect and gave me this code when my KERUI PIR Sensor is activated:

`Decimal: 14769002 (24Bit) Binary: 111000010101101101101010 Tri-State: not applicable PulseLength: 299 microseconds Protocol: 1`
`Raw data: 9299,915,301,915,302,914,304,302,905,301,905,301,905,302,904,911,306,300,906,908,308,299,907,909,308,908,308,298,908,908,310,907,310,296,909,906,311,905,312,295,910,904,312,295,911,905,312,294,912,`

But with esphome using remote_receiver on the same nodemcu, the code is truncated or not displayed with a lot of "Data is coming in too fast! Try increasing the buffer size." messages.
```
[17:39:10][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:10][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:11][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:12][D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='11000010101101101101010'
[17:39:12][D][remote.rc_switch:183]: Received RCSwitch: protocol=4 data='11000010101101101101010'
[17:39:13][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:14][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:14][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:15][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:16][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:16][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:17][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:18][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:19][D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='1000010101101101101010'
[17:39:19][D][remote.rc_switch:183]: Received RCSwitch: protocol=4 data='1000010101101101101010'
[17:39:20][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:20][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:21][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:22][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:24][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:24][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:25][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:26][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:27][D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='01101101010'
[17:39:27][D][remote.rc_switch:183]: Received RCSwitch: protocol=4 data='01101101010'
[17:39:28][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:28][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:29][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:30][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:30][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:31][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
[17:39:32][W][remote.receiver:313]: Data is coming in too fast! Try increasing the buffer size.
```
I played with filter, idle and buffer parameters.

HFsi, 2019-05-04:
@Gastondamian Did you tried OMG? That will be my next attemp since I have Sonoff Bridge

escoand, 2019-05-06:
@HFsi I tried Tasmota and it's working in general without the garbage. The only problem I have with it is that my [remote](https://www.brennenstuhl.com/de-DE/produkte/funksteckdosen/comfort-line-funkschalt-set-4x-ip20) is sending some kind of rolling code:

```
[19:41:06]20:41:07 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x23FB6C","Bits":24,"Protocol":3,"Pulse":103}}
[19:41:08]20:41:09 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x28801C","Bits":24,"Protocol":5,"Pulse":523}}
[19:41:11]20:41:12 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x2F37AC","Bits":24,"Protocol":5,"Pulse":526}}
[19:41:13]20:41:14 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x23FB6C","Bits":24,"Protocol":5,"Pulse":525}}
[19:41:20]20:41:21 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x23FB6C","Bits":24,"Protocol":5,"Pulse":525}}
[19:41:22]20:41:23 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x27035C","Bits":24,"Protocol":3,"Pulse":103}}
[19:41:29]20:41:29 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x28801C","Bits":24,"Protocol":5,"Pulse":519}}
[19:41:32]20:41:32 RSL: tele/sonoff/RESULT = {"RfReceived":{"Data":"0x23FB6C","Bits":24,"Protocol":5,"Pulse":524}}
```

OttoWinter, 2019-05-07:
No that's incorrect. The difference is that Tasmota just does not have a raw dumper, where esphome has one. If you don't want to see raw dumps just disable them in the `dumpers` key. Problem solved.

Raw is especially useful for IR codes (and RF codes too) because the application doesn't need to support the protocol.

If the rc_switch codes are not decoded correctly that would be a bug. The getting lots of raw messages is correct (and a feature).

xiaolei0125, 2019-05-12:
This is my configuration, but I received a number of different codes. These codes seem to be truncated. Can someone tell me why?  how can I get the correct code of 24 bits?

remote_receiver:
    pin: D1
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
    dump: [rc_switch]

`[D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='1000100000'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=2 data='1000100000'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='10001000000000001011'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=2 data='10001000000000001011'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=3 data='10001000000000001011'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=2 data='100010000000000010111110'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=3 data='100110000000000010111110'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='10001000000000001'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=2 data='10001000000000001'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=3 data='10001000000000001'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=1 data='1000100000000000101'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=2 data='1000100000000000101'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=3 data='1000100000000000101'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=4 data='1000100000000000101'
[D][remote.rc_switch:183]: Received RCSwitch: protocol=5 data='1000100000000000101'`

OttoWinter, 2019-05-12:
> but I received a number of different codes.

That is because there are different protocols with different timings that sometimes overlap. So one message can sometimes be decoded by several protocols but only one is correct. You need to check which one is correct my trial and error.

> how can I get the correct code of 24 bits?

There are two lines with a code of 24 bits, use those (if you know that the code is 24 bits long)

xiaolei0125, 2019-05-14:
@OttoWinter Thanks for your reply, I will try again later.

Gastondamian, 2019-05-18:
> This is my configuration, but I received a number of different codes. These codes seem to be truncated. Can someone tell me why? how can I get the correct code of 24 bits?
> 
> remote_receiver:
> pin: D1
> tolerance: 50%
> filter: 250us
> idle: 4ms
> buffer_size: 2kb
> dump: [rc_switch]
> 
> Received RCSwitch: protocol=1 data='1000100000'
> Received RCSwitch: protocol=2 data='1000100000'
> Received RCSwitch: protocol=1 data='10001000000000001011'
> Received RCSwitch: protocol=2 data='10001000000000001011'
> Received RCSwitch: protocol=3 data='10001000000000001011'
> Received RCSwitch: protocol=2 data='100010000000000010111110'
> Received RCSwitch: protocol=3 data='100110000000000010111110'
> Received RCSwitch: protocol=1 data='10001000000000001'
> Received RCSwitch: protocol=2 data='10001000000000001'
> Received RCSwitch: protocol=3 data='10001000000000001'
> Received RCSwitch: protocol=1 data='1000100000000000101'
> Received RCSwitch: protocol=2 data='1000100000000000101'
> Received RCSwitch: protocol=3 data='1000100000000000101'
> Received RCSwitch: protocol=4 data='1000100000000000101'
> Received RCSwitch: protocol=5 data='1000100000000000101'

I have the same issue, the code is truncated and I receive al lot of "Data is coming in too fast! Try increasing the buffer size."