esphome/issues #21: ESP32 BLE Tracker causes API Connection Issues

njbuch, 2019-02-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows, platformIO, newest components as of today.
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP32
**Affected component:**

api

**Description of problem:**
It seems that the board looses connection for unknown reasons and reboots. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: tracker
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: 'xx'
  password: 'xx'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
```

**Traceback (if applicable):**

```
[17:50:08][W][ota:136]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[17:50:08][I][application:114]: Running through first loop()
[17:50:08][I][application:141]: First loop finished successfully!
[17:52:03][I][ota:154]: Boot seems successful, resetting boot loop counter.
[17:52:34][D][esp32_ble_tracker:441]: Found device 12:3B:6A:1B:7E:0C RSSI=-76
[17:52:34][D][esp32_ble_tracker:462]:   Address Type: PUBLIC
[17:55:04][D][esp32_ble_tracker:165]: Starting scan...
[17:55:05][D][esp32_ble_tracker:441]: Found device 22:A3:0C:A2:BE:98 RSSI=-56
[17:55:05][D][esp32_ble_tracker:462]:   Address Type: RANDOM
[17:55:08][E][api:063]: No client connected to API. Rebooting...
[17:55:08][I][helpers:277]: Forcing a reboot... Reason: 'api'
[17:55:08]ets Jun  8 2016 00:22:57
[17:55:08]
[17:55:08]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:55:08]configsip: 0, SPIWP:0xee
[17:55:08]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:55:08]mode:DIO, clock div:2
[17:55:08]load:0x3fff0018,len:4
[17:55:08]load:0x3fff001c,len:952
[17:55:08]load:0x40078000,len:6084
[17:55:08]load:0x40080000,len:7944
[17:55:08]entry 0x40080310
[17:55:08][I][logger:071]: Log initialized
[17:55:08][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[17:55:08][I][application:053]: Running through setup()...
[17:55:09][D][esp32_ble_tracker:165]: Starting scan...
[17:55:09][C][wifi:029]: Setting up WiFi...
[17:55:09][D][wifi:260]: Starting scan...
[17:55:09][D][esp32_ble_tracker:441]: Found device 22:A3:0C:A2:BE:98 RSSI=-53
[17:55:09][D][esp32_ble_tracker:462]:   Address Type: RANDOM
[17:55:11][D][wifi:275]: Found networks:
[17:55:11][I][wifi:308]: - 'xx' (00:1E:80:76:8B:E4) ÔûéÔûäÔûåÔûê
[17:55:11][D][wifi:309]:     Channel: 11
[17:55:11][D][wifi:310]:     RSSI: -64 dB
[17:55:11][D][wifi:312]: - 'DIRECT-dLC410 Series' (02:15:99:09:76:E6) ÔûéÔûäÔûåÔûê
[17:55:11][D][wifi:312]: - 'sniffer' (18:31:BF:45:CE:B0) ÔûéÔûäÔûåÔûê
[17:55:11][I][wifi:179]: WiFi Connecting to '1421530'...
[17:55:13][I][wifi:349]: WiFi connected!
[17:55:13][C][wifi:240]:   SSID: 'xx'
[17:55:13][C][wifi:241]:   IP Address: 192.168.0.143
[17:55:13][C][wifi:243]:   BSSID: 00:1E:80:76:8B:E4
[17:55:13][C][wifi:245]:   Hostname: 'tracker'
[17:55:13][C][wifi:250]:   Signal strength: -63 dB ÔûéÔûäÔûåÔûê
[17:55:13][C][wifi:251]:   Channel: 11
[17:55:13][C][wifi:252]:   Subnet: 255.255.255.0
[17:55:13][C][wifi:253]:   Gateway: 192.168.0.1
[17:55:13][C][wifi:254]:   DNS1: 192.168.0.1
[17:55:13][C][wifi:255]:   DNS2: 0.0.0.0
[17:55:13][C][api:023]: Setting up Home Assistant API server...
[17:55:13][C][ota:129]: Over-The-Air Updates:
[17:55:13][C][ota:130]:   Address: 192.168.0.143:3232
[17:55:13][I][application:089]: setup() finished successfully!
[17:55:13][I][application:097]: You're running esphomelib v1.10.1 compiled on Feb  1 2019, 17:14:29
[17:55:13][C][wifi:341]: WiFi:
[17:55:13][C][wifi:240]:   SSID: 'xx'
[17:55:13][C][wifi:241]:   IP Address: 192.168.0.143
[17:55:13][C][wifi:243]:   BSSID: 00:1E:80:76:8B:E4
[17:55:13][C][wifi:245]:   Hostname: 'tracker'
[17:55:13][C][wifi:250]:   Signal strength: -63 dB ÔûéÔûäÔûåÔûê
[17:55:13][C][wifi:251]:   Channel: 11
[17:55:13][C][wifi:252]:   Subnet: 255.255.255.0
[17:55:13][C][wifi:253]:   Gateway: 192.168.0.1
[17:55:13][C][wifi:254]:   DNS1: 192.168.0.1
[17:55:13][C][wifi:255]:   DNS2: 0.0.0.0
[17:55:13][C][logger:099]: Logger:
[17:55:13][C][logger:100]:   Level: DEBUG
[17:55:13][C][logger:101]:   Log Baud Rate: 115200
[17:55:13][C][esp32_ble_tracker:744]: BLE Tracker:
[17:55:13][C][esp32_ble_tracker:745]:   Scan Interval: 300 s
[17:55:13][C][api:072]: API Server:
[17:55:13][C][api:073]:   Address: 192.168.0.143:6053
[17:55:13][C][ota:129]: Over-The-Air Updates:
[17:55:13][C][ota:130]:   Address: 192.168.0.143:3232
[17:55:13][I][application:114]: Running through first loop()
[17:55:13][I][application:141]: First loop finished successfully!
[17:57:08][I][ota:154]: Boot seems successful, resetting boot loop counter.
[18:00:09][D][esp32_ble_tracker:165]: Starting scan...
[18:00:10][D][esp32_ble_tracker:441]: Found device 22:A3:0C:A2:BE:98 RSSI=-55
[18:00:10][D][esp32_ble_tracker:462]:   Address Type: RANDOM
[18:00:13][E][api:063]: No client connected to API. Rebooting...
[18:00:13][I][helpers:277]: Forcing a reboot... Reason: 'api'
[18:00:13]ets Jun  8 2016 00:22:57
[18:00:13]
[18:00:13]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:00:13]configsip: 0, SPIWP:0xee
[18:00:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:00:13]mode:DIO, clock div:2
[18:00:13]load:0x3fff0018,len:4
[18:00:13]load:0x3fff001c,len:952
[18:00:13]load:0x40078000,len:6084
[18:00:13]load:0x40080000,len:7944
[18:00:13]entry 0x40080310
[18:00:13][I][logger:071]: Log initialized
```

**Additional information:**

MichelLap, 2019-03-06:
Hi, 

I am experiencing a very similar issue.  

Here are the details:

ESPHome Version: 1.11.2
Hardware: ESP32-WROOM-32
HA platform: Hass.io on RPI3b+
HA version: 0.88.1

Description:

My ESP32 modules stay "connected" in HA for days without issues until I add the "esp32_ble_tracker:" into the configuration.  At which point, it becomes unreachable  (for 8 to 10 minutes) many times per days.  

Did many tests:

- Reproduced on two distinct ESP32 modules.
- Problem happens using either dynamic or Static IP config.
- Tried different physical locations (around the Wi-Fi router)
- Happen with the basic default config.  No issues with any other configuration (status, uptime, version, hall sensor every 10 sec,...).  All are working fines and reports to HA for days without a glitch.  Problem appear as soon as I start using the BLE tracker component (with or without any BLE sensors configured).

If i can be of any help to assist on this issue, let me know. 

Currently trying to migrate all my Xiaomi Mijia from RPI to ESP32 using ESPHome, but need to resolve this issue first.

Really love this application and greatly appreciate the time and effort this require.

Thanks

brandond, 2019-03-06:
@MichelLap please open a new issue and fill out the issue template.

Michel-Lapointe, 2019-03-20:
Thanks,

Issue seams to have been completely fixed by latest 1.12.0 release.  Did not get any disconnect in the past 48 hours since the upgrade.

Thanks for the great work.

OttoWinter, 2019-03-20:
Amazing! I will close the issue then. New issues of this type should be reported as new issues in this tracker.

closed 2019-03-20 by OttoWinter