esphome/issues #23: Installing two DS18B20 sensors

dbasberg, 2019-02-02:
I have two DHT22s running on separate pins so, much is working.  I added two DS18B20s on GPIO25 and found their addresses by method in 'Dallas Temperature Sensor' (ESPHome 1.10.1) example.  Then I added two sensors according to that write-up. The log looks good except error message 'Reading scratchpad from Dallas Sensor failed'.  The scratchpad contains " FF FF FF FF FF FF'.  On an oscilloscope, we have messages exchanged at good logic levels near 0 & 3.3 volts.  The addresses report perfectly at all times.  So some works (addresses) but not values for temperature.  I am happy to do more testing as requested.  I have not found others discussing this.

OttoWinter, 2019-04-25:
Closing due to inactivity.

closed 2019-04-25 by OttoWinter

jikama, 2019-06-12:
Connecting two configured DS18B20 will freeze the NodeMcu with ESPHOME. If one of the sensors is disconnected, the other one is functional and I see the value.

dallas:

pin: D5
update_interval: 10s
sensor:
platform: dallas
address: 0x3F030997792F5828
name: "Temperature #1"
platform: dallas
address: 0xD703139779A28828
name: "Temperature #2"

brandond, 2019-06-12:
@jikama please don't necro-post issues. Open a new one and fill out the template.