esphome/issues #27: 4CH Pro R1 not working (bootloop)

peterforeman, 2019-02-02:
**Operating environment (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266/Board/Sonoff):**

Sonoff 4CH Pro R1

```
ESP Chip Id 10442015
Flash Chip Id 0x144051
Flash Size 1024kB
```

**Affected component:**

"Generic Sonoff"

**Description of problem:**
Uploading compiled firmware on my Tasmota-flash Sonoff 4CH Pro R1 creates a bootloop. The device never connects to WiFi anymore. Tried flashing via OTA (first time), and several times via serial. I tried the ESP8285 board (which is in the 4CH Pro R1) as well as the esp8266 board config. Nothing worked. Flashed Tasmota binary again which worked instantly (even remembered the settings).

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: 4chpro
  platform: ESP8266
  board: esp8285
  board_flash_mode: dout
```

**Traceback (if applicable):**

```
Not available yet, will update asap with serial logging.
```

**Additional information:**

peterforeman, 2019-02-23:
Did some further testing. It crashes on connecting to wifi, using a static IP. Switching to DHCP fixes the problem.

Static config:

```
wifi:
  ssid: 'xxx'
  password: '12345678901234567890123456789012'
  fast_connect: on
  manual_ip:
    static_ip: 10.0.100.89
    subnet: 255.255.255.0
    gateway: 10.0.100.1
    dns1: 10.0.100.1
    dns2: 10.0.100.1
```

Serial output:
```
csum 0x2d
vbb28d4a3
~ld
␛[0;32m[I][logger:071]: Log initialized␛[0m
␛[0;35m[C][ota:461]: There have been 2 suspected unsuccessful boot attempts.␛[0m
␛[0;32m[I][application:053]: Running through setup()...␛[0m
␛[0;36m[D][binary_sensor:027]: 'Sonoff 4CH Status': Sending state OFF␛[0m
␛[0;35m[C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Sonoff 4CH Button 1'...␛[0m
␛[0;36m[D][binary_sensor:027]: 'Sonoff 4CH Button 1': Sending state OFF␛[0m
␛[0;35m[C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Sonoff 4CH Button 2'...␛[0m
␛[0;36m[D][binary_sensor:027]: 'Sonoff 4CH Button 2': Sending state OFF␛[0m
␛[0;35m[C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Sonoff 4CH Button 3'...␛[0m
␛[0;36m[D][binary_sensor:027]: 'Sonoff 4CH Button 3': Sending state OFF␛[0m
␛[0;35m[C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Sonoff 4CH Button 4'...␛[0m
␛[0;36m[D][binary_sensor:027]: 'Sonoff 4CH Button 4': Sending state OFF␛[0m
␛[0;35m[C][switch.gpio:023]: Setting up GPIO Switch 'Sonoff 4CH Relay 1'...␛[0m
␛[0;36m[D][switch:055]: 'Sonoff 4CH Relay 1': Sending state OFF␛[0m
␛[0;35m[C][switch.gpio:023]: Setting up GPIO Switch 'Sonoff 4CH Relay 2'...␛[0m
␛[0;36m[D][switch:055]: 'Sonoff 4CH Relay 2': Sending state OFF␛[0m
␛[0;35m[C][switch.gpio:023]: Setting up GPIO Switch 'Sonoff 4CH Relay 3'...␛[0m
␛[0;36m[D][switch:055]: 'Sonoff 4CH Relay 3': Sending state OFF␛[0m
␛[0;35m[C][switch.gpio:023]: Setting up GPIO Switch 'Sonoff 4CH Relay 4'...␛[0m
␛[0;36m[D][switch:055]: 'Sonoff 4CH Relay 4': Sending state OFF␛[0m
␛[0;35m[C][wifi:029]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:179]: WiFi Connecting to 'xxx'...␛[0m

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
```

And eventually into safe mode:

```
ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
␛[0;32m[I][logger:071]: Log initialized␛[0m
␛[0;35m[C][ota:461]: There have been 10 suspected unsuccessful boot attempts.␛[0m
␛[1;31m[E][ota:466]: Boot loop detected. Proceeding to safe mode.␛[0m
␛[0;35m[C][wifi:029]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:179]: WiFi Connecting to 'xxx'...␛[0m

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
␛[0;32m[I][logger:071]: Log initialized␛[0m
␛[0;35m[C][ota:461]: There have been 0 suspected unsuccessful boot attempts.␛[0m
␛[0;32m[I][application:053]: Running through setup()...␛[0m
```
(etc)

peterforeman, 2019-03-16:
@OttoWinter For me this can be closed, since the DHCP workaround is ok for me. But I can imagine others have this problem as well.

OttoWinter, 2019-03-17:
I'll close this then. The issue is definitely somewhere in the internals of the ESP SDK though, which we can't change because it's closed-source and not known to us.

closed 2019-03-17 by OttoWinter

deece, 2019-04-15:
I see the same problem, with DHCP enabled (same reset cause)

I have a pair of 4ch Sonoffs, flashing the same binary, one works as expected, and one has a boot loop.
The Wifi network they connect to are hidden.

```
esphome:
  name: sonoff4ch_attic
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "redacted"
  password: "redacted"
  fast_connect: true
  domain: .lan

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: status
    name: "Sonoff 4CH Status"

switch:
  - platform: gpio
    name: "Sonoff 4CH Relay 2"
    pin: GPIO5
    id: out2
  - platform: gpio
    name: "Sonoff 4CH Relay 3"
    pin: GPIO4
    id: out3
  - platform: gpio
    name: "Sonoff 4CH Relay 4"
    pin: GPIO15
    id: out4

status_led:
  pin:
    number:  GPIO13
    inverted: True

output:
  - platform: gpio
    id: out1
    pin: GPIO12

light:
  - platform: binary
    name: "Workshop Lights"
    output: out1
```

OttoWinter, 2019-06-18:
Moving the discussion to #455