esphome/issues #28: LICENSE

sh0rez, 2019-02-02:
Esphomeyaml is missing a LICENSE file or any other kind of declaration.
Is this intended? Should all rights be reserved? And does this comply with the Apache-License of homeassistant?

OttoWinter, 2019-02-28:
Ok, it took some time but I finally studied up on licensing and GPLv3 - I am far from an expert on this and not a lawyer, so this may not (and probably is not) 100% accurate.

So I believe esphome does not violate GPLv3 due to two reasons:

1. GPLv3 has a special "System Library" exception (see section 1 and 6) that allows linking against static libraries:

> A separable portion of the object code, whose source code is excluded from the Corresponding Source as a System Library, need not be included in conveying the object code work.

> The “System Libraries” of an executable work include anything, other than the work as a whole, that (a) is included in the normal form of packaging a Major Component, but which is not part of that Major Component, and (b) serves only to enable use of the work with that Major Component, or to implement a Standard Interface for which an implementation is available to the public in source code form. A “Major Component”, in this context, means a major essential component (kernel, window system, and so on) of the specific operating system (if any) on which the executable work runs, or a compiler used to produce the work, or an object code interpreter used to run it.

The definition there is pretty hard to understand (at least for me) - this [stackoverflow answer](https://opensource.stackexchange.com/a/2217) helped. 

> The final paragraph of section 1 revises the exception to the source code distribution requirement in GPLv2 that we have sometimes called the system library exception. This exception has been read to prohibit certain distribution arrangements that we consider reasonable and have not sought to prevent, such as distribution of gcc linked with a non-free C library that is included as part of a larger non-free system. This is not to say that such non-free libraries are legitimate; rather, preventing free software from linking with these libraries would hurt free software more than it would hurt proprietary software.
> 
> As revised, the exception has two parts. Part (a) rewords the GPLv2 exception for clarity but also removes the words ``unless that component itself accompanies the executable.'' By itself, (a) would be too permissive, allowing distributors to evade their responsibilities under the GPL. We have therefore added part (b) to specify when a system library that is an adjunct of a major essential operating system component, compiler, or interpreter does not trigger the requirement to distribute source code. The more low-level the functionality provided by the library, the more likely it is to be qualified for this exception.

And I think the ESP8266 nonos/rtos SDK and esp-idf both apply with that description of a system library (especially for esp-idf, it also *is* the operating system). But this whole "system library" exception is probably a bit flakey, and this is where i'm not 100% sure if it applies.

The extensive GPLv3 FAQ also has an entry about it too: https://www.gnu.org/licenses/gpl-faq.html#SystemLibraryException

2. The other reason is that as far as I understand, the "give me your entire source" rules only apply when you redistribute compiled versions of the software - and more specifically for all statically linked libraries.

But... It would be nonsense to distribute an ESPHome binary, since that contains your WiFi/MQTT credentials etc. At least I have not seen anybody redistribute an esphome binary yet.

So in conclusion I think GPLv3 is not violated here for those two reasons. I may be wrong, so if someone who knows more about this comes across this, I'd like to hear your opinion.

For now, I'm closing this issue.

Ref: https://github.com/xoseperez/espurna/issues/948

closed 2019-02-28 by OttoWinter

sh0rez, 2019-02-28:
I have no real idea about legal things tbh, but your explanation sounds reasonable. Thank you for looking into this 👍