esphome/issues #29: Problem with OTA updates

tomdee, 2019-02-02:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP32

**Affected component:**
OTA

**Description of problem:**
I've never been able to get OTA working. I was using the last release 1.9.x and have just tried with 1.10 now. I've also switched to static IPs but it's still not working.

Usually, I don't see it get past 2 or 3 %, but this time it got a bit higher
```
22:25:58][D][ota:180]: Starting OTA Update from 192.168.10.142...
[22:25:58][D][ota:336]: OTA in progress: 0.1%
[22:25:59][D][ota:336]: OTA in progress: 5.3%
[22:26:00][D][ota:336]: OTA in progress: 10.6%
[22:26:01][D][ota:336]: OTA in progress: 14.8%
[22:26:02][D][ota:336]: OTA in progress: 20.0%
[22:26:03][D][ota:336]: OTA in progress: 22.6%
[22:26:04][D][ota:336]: OTA in progress: 28.1%
[22:26:05][D][ota:336]: OTA in progress: 33.2%
[22:26:06][W][ota:398]: Error reading data!
[22:26:06][W][ota:369]: Update end failed! Error: No Error
[22:26:06]
[22:26:06][E][WiFiClient.cpp:338] write(): 12
[22:26:06][D][esp32_ble_tracker:363]: Xiaomi MiJia 4C:65:A8:DB:A6:01 Got temperature=18.7°C, humidity=72.0%
```

Any thoughts on what else I can try?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```

**Traceback (if applicable):**

```

```

**Additional information:**

legomind, 2019-02-13:
Just wanted to say that I am not using the BLE tracker, and I was unable to flash via OTA (connection kept dropping) until I switched to using the dev arduino version.

OttoWinter, 2019-02-13:
> I'm pretty sure I couldn't get OTA updates working on my ESP32 even without the BLE tracker component enabled. Could this be a more general problem?

@tomdee With what esphomeyaml version was that? With dev? Because I recently tried to upgrade the ESP32 core to the newest version in dev, but that had some stability issues and I reverted that (https://github.com/esphome/esphome/pull/422). So if you saw this stability issue with dev version (not 1.10.1) that could explain it.

@legomind That is interesting. I do not see any [recent commits](https://github.com/espressif/arduino-esp32/commits/master) in arduino-esp32 repo that would have fixed anything related to that. Will try that soon.

danps1, 2019-02-25:
Just a +1 that this is almost certainly related to BLE.
I have 2 identical ESP32 devices, 1 of which I use BLE on and the other I don't.
The one with BLE never gets beyond 6% when doing an OTA, whereas the other updates flawlessly every time.

rb2k, 2019-02-25:
Same here, ESP32 device + esp32_ble_tracker with xiaomi_miflora sensor and the OTA update does indeed not work

OttoWinter, 2019-03-05:
Fixed in latest dev version (esp-sdk has been upgraded, which fixes this issue)

closed 2019-03-05 by OttoWinter

tomdee, 2019-03-06:
Thanks @OttoWinter ESP32 OTA updates are working for me now!

marcinkowalczyk, 2019-03-06:
Hi,

 I've just tried with dev and no luck
`kowal@ubuntu:~$ sudo esphome termometry-dol.yaml logs
INFO Reading configuration...
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (CP2102 USB to UART Bridge Controller)
  [2] Over The Air (192.168.1.14)
(number): 2
INFO Starting log output from 192.168.1.14 using esphome API
INFO Connecting to 192.168.1.14:6053 (192.168.1.14)
INFO Successfully connected to 192.168.1.14
[11:56:50][I][application:097]: esphome-core version 1.12.0-dev compiled on Mar  6 2019, 11:33:04
[11:56:50][C][wifi:332]: WiFi:
[11:56:50][C][wifi:228]:   SSID: 'SAGEM_6230_IoT'
[11:56:50][C][wifi:229]:   IP Address: 192.168.1.14
[11:56:50][C][wifi:231]:   BSSID: 7A:8A:20:8D:AA:B1
[11:56:50][C][wifi:232]:   Hostname: 'termometrydol'
[11:56:50][C][wifi:236]:   Signal strength: -66 dB ▂▄▆█
[11:56:50][C][wifi:237]:   Channel: 1
[11:56:50][C][wifi:238]:   Subnet: 255.255.255.0
[11:56:50][C][wifi:239]:   Gateway: 192.168.1.1
[11:56:50][C][wifi:240]:   DNS1: 1.1.1.1
[11:56:50][C][wifi:241]:   DNS2: 1.0.0.1
[11:56:50][C][logger:142]: Logger:
[11:56:50][C][logger:143]:   Level: DEBUG
[11:56:50][C][logger:144]:   Log Baud Rate: 115200
[11:56:50][C][logger:145]:   Hardware UART: UART0
[11:56:50][C][esp32_ble_tracker:703]: BLE Tracker:
[11:56:50][C][esp32_ble_tracker:704]:   Scan Interval: 300 s
[11:56:50][C][esp32_ble_tracker:712]:   Xiaomi 00004C65Axxxxxxx
[11:56:50][C][esp32_ble_tracker:713]:     Temperature  'Kuchnia temperatura'
[11:56:50][C][esp32_ble_tracker:713]:       Unit of Measurement: '°C'
[11:56:50][C][esp32_ble_tracker:713]:       Accuracy Decimals: 1
[11:56:50][C][esp32_ble_tracker:714]:     Humidity  'Kuchnia wilgotnosc'
[11:56:50][C][esp32_ble_tracker:714]:       Unit of Measurement: '%'
[11:56:50][C][esp32_ble_tracker:714]:       Accuracy Decimals: 1
[11:56:50][C][esp32_ble_tracker:714]:       Icon: 'mdi:water-percent'
[11:56:50][C][esp32_ble_tracker:718]:     Battery Level  'Kuchnia bateria'
[11:56:50][C][esp32_ble_tracker:718]:       Unit of Measurement: '%'
[11:56:50][C][esp32_ble_tracker:718]:       Accuracy Decimals: 0
[11:56:50][C][esp32_ble_tracker:718]:       Icon: 'mdi:battery'
[11:56:50][C][api:089]: API Server:
[11:56:50][C][api:090]:   Address: 192.168.1.14:6053
[11:56:50][C][ota:127]: Over-The-Air Updates:
[11:56:50][C][ota:128]:   Address: 192.168.1.14:3232
[11:56:50][W][ota:134]: Last Boot was an unhandled reset, will proceed to safe mode in 4 restarts
[11:56:51][I][ota:152]: Boot seems successful, resetting boot loop counter.
^Ckowal@ubuntu:~$ sudo esphome termometry-dol.yaml upload
INFO Reading configuration...
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (CP2102 USB to UART Bridge Controller)
  [2] Over The Air (192.168.1.14)
(number): 2
INFO Connecting to 192.168.1.14
INFO Uploading termometrydol/.pioenvs/termometrydol/firmware.bin (1378656 bytes)
Uploading: [=========                                                   ] 15%
ERROR Error sending data: timed out
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:50484
INFO Upload size: 1378656
INFO Sending invitation to 192.168.1.14
..........
ERROR No response from the ESP
kowal@ubuntu:~$
`

OttoWinter, 2019-03-06:
What esphome (yaml) version is this? The fix is in both core and python projects.

marcinkowalczyk, 2019-03-06:
@OttoWinter installed from pip couple of minutes back:
esphome in /usr/local/lib/python2.7/dist-packages (1.11.2)

OttoWinter, 2019-03-06:
see faq, install esphome dev version.

egon0, 2019-03-08:
esphome dev-version seems to have fixed the issue. at least for me and just tried it one-time. but ota update finished successful for the first time for me.

buzzeddesign, 2019-03-17:
Appreciate all your work on esphome Otto!

Still a no go for me.  Using 1.13.0-dev of esphome.  Consistently getting a timed out error.

Do we still need this flag in the YAML?
`arduino_version: dev`

```
$ esphome ble_hub_2.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Updating esphome-core copy from git (ble_hub_2/.piolibdeps/esphome-core)
Already up to date.
INFO Compiling app...
INFO Running:  platformio run -d ble_hub_2
Processing ble_hub_2 (platform: espressif32@1.6.0; framework: arduino; board: node32s)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/node32s.html
PLATFORM: Espressif 32 > Node32s
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphome-core> 1.13.0-dev #fe04ee5
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.0.3
|   |-- <Preferences> 1.0
|   |-- <Update> 1.0
Retrieving maximum program size .pioenvs/ble_hub_2/firmware.elf
Checking size .pioenvs/ble_hub_2/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  18.1% (used 59304 bytes from 327680 bytes)
PROGRAM: [========= ]  89.0% (used 1458806 bytes from 1638400 bytes)
============================================================================ [SUCCESS] Took 5.16 seconds ============================================================================
INFO Successfully compiled program.
INFO Resolving IP address of ble_hub_2.local
INFO  -> 192.168.0.253
INFO Uploading ble_hub_2/.pioenvs/ble_hub_2/firmware.bin (1458928 bytes)
Uploading: [=                                                           ] 1% 
ERROR Error sending data: timed out
```