esphome/issues #30: esphomeyaml Hass.io addon Pi Zero W not starting

wutr, 2019-02-02:
**Operating environment (Hass.io/Docker/pip/etc.):**
Hass.io on Raspberry Pi Zero W

**Affected component:**
https://esphomelib.com/esphomeyaml/guides/getting_started_hassio.html

**Description of problem:**
After successfully adding the repository to the Hass.io add-on store I have installed esphomeyaml v1.9.3 which took a while but as far as I'm aware it was successful. When clicking "start" however, it appears the add-on is starting, and the "open web-ui" link appears. There is no webserver running though and a connection can't be made. When leaving the add-on page and returning to it, the add-on is in the "stopped" state and clicking on "start" doesn't actually do anything from then on.

**Additional information:**
* Hass.io syslog doesn't show anything useful whatsoever.
* Rebooting so far hasn't helped
* There is no log information in the add-on page in the Hass.io dashboard like there is for other add-ons

kodavn, 2019-03-20:
i have the same problem when install esphome 1.12 on Hass.io - Raspberry Pi Zero W. I don't think RPZ W lack resources because my install on Hassio - Orange pi zero worked well. Anyone sold it?

OttoWinter, 2019-07-11:
Closing as won't fix - esphome does not work with the resources of the tiny rpi zero

closed 2019-07-11 by OttoWinter