esphome/issues #33: Removal of MQTT when API is active breaks binary_sensors

davericher, 2019-02-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Linux ra 4.20.3-042003-generic #201901171122 SMP Thu Jan 17 16:25:26 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic

**ESP (ESP32/ESP8266/Board/Sonoff):**

PLATFORM: Espressif 32 (Stage) > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)

**Affected component:**

binary_sensor
**Description of problem:**

It seems your removal of mqtt when API is enabled has stopped Home Assistant from seeing binary sensor results, the sensors do not show up in ESP homes web interface, but they do show up in the Log output 

**Problem-relevant YAML-configuration entries:**
```yaml
...
  - platform: gpio
    pin:
      number: GPIO26
      mode: INPUT_PULLUP
      inverted: True
    name: Sirius lightswitch
...
```

**Traceback (if applicable):**

```
```

**Additional information:**

OttoWinter, 2019-02-03:
@davericher Then this is a duplicate of https://github.com/esphome/issues/issues/26 - One moment, I'll upload the patch shortly.

davericher, 2019-02-03:
Your the man @OttoWinter

davericher, 2019-02-03:
Confirmed working

closed 2019-02-03 by davericher