esphome/issues #34: Assert failed in esp_err_t nvs::PageManager::requestNewPage()

mvturnho, 2019-02-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I use Windows 10 PC with pip to build and flash esphomelib.

**ESP ESP32:**

Board is an nodemcu ESP32 with an 8led fastled strip connected to pin gpio4
**Affected component:**

https://esphomelib.com/esphomeyaml/components/light/fastled_clockless.html

**Description of problem:**

I use the addressable_lambda for FastLed clockless. After subitting the folowing MQTT message a number of times to the esp it crashes with the  boot: Assert failed in esp_err_t nvs message.

```json
{"effect":"statustest","state":"ON","brightness":100,"color":{"r":255,"g":255,"b":0}}
```

Here the exact error message that I see when connected to the esp over serial.

```
[13:35:55][D][statustest:032]: switch on led 1
[13:35:55][D][light.state:454]: 'LED Strip' Turning ON
[13:35:55][D][light.state:461]:   Brightness: 39%
[13:35:55][D][light.state:486]:   Red=100%, Green=100%, Blue=0%
[13:35:55][D][light.state:496]:   Transition Length: 1000 ms
[13:35:55][D][light.state:503]:   Effect: 'statustest'
[13:35:55]E (14371) boot: Assert failed in esp_err_t nvs::PageManager::requestNewPage(), /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp:175 (usedEntries == newPage->getUsedEntryCount())
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: controlpanel
  platform: ESP32
  board: nodemcu-32s
  esphomelib_version:
    branch: master

wifi:
  ssid: 'XXXXXXXXXXX'
  password: 'XXXXXXXXXXXXXXXXXXXXXXXXXXXX'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

# Example configuration entry
i2c:
  sda: 21
  scl: 22
  scan: True

light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: GPIO4
    num_leds: 8
    rgb_order: GRB
    name: 'LED Strip'
    effects:
      - addressable_lambda:
          name: statustest
          update_interval: 500ms
          lambda: |-
            ESP_LOGD("statustest", "switch on led 1");
            for (int i = 0; i < it.size(); i++) 
              it[i] = it[i].get() - 10;
            it[1] = light::ESPHSVColor(0, 255, 100);
```

**Traceback (if applicable):**

```
INFO Starting log output from controlpanel/debug
[13:48:04][C][mqtt.client:067]:   Topic Prefix: 'controlpanel'
[13:48:04][C][mqtt.client:069]:   Log Topic: 'controlpanel/debug'
[13:48:04][C][mqtt.client:072]:   Availability: 'controlpanel/status'
[13:48:04][C][light.mqtt_json:072]: MQTT Light 'LED Strip':
[13:48:04][C][light.mqtt_json:073]:   State Topic: 'controlpanel/light/led_strip/state'
[13:48:04][C][light.mqtt_json:073]:   Command Topic: 'controlpanel/light/led_strip/command'
[13:48:04][I][application:114]: Running through first loop()
[13:48:04][D][statustest:030]: switch on led 1
[13:48:04][I][application:141]: First loop finished successfully!
[13:48:04][D][statustest:030]: switch on led 1
[13:48:05][D][statustest:030]: switch on led 1
[13:48:05][D][statustest:030]: switch on led 1
[13:48:06][D][statustest:030]: switch on led 1
[13:48:06][D][statustest:030]: switch on led 1
[13:48:07][D][statustest:030]: switch on led 1
[13:48:07][D][statustest:030]: switch on led 1
[13:48:08][D][statustest:030]: switch on led 1
[13:48:08][D][statustest:030]: switch on led 1
[13:48:09][D][statustest:030]: switch on led 1
[13:48:09][D][statustest:030]: switch on led 1
[13:48:10][D][statustest:030]: switch on led 1
[13:48:10][D][light.state:454]: 'LED Strip' Turning ON
[13:48:10][D][light.state:461]:   Brightness: 39%
[13:48:10][D][light.state:486]:   Red=100%, Green=100%, Blue=0%
[13:48:10][D][light.state:496]:   Transition Length: 1000 ms
[13:48:10][D][light.state:503]:   Effect: 'statustest'
[13:48:10][D][statustest:030]: switch on led 1
[13:48:11][D][statustest:030]: switch on led 1
[13:48:11][D][statustest:030]: switch on led 1
[13:48:12][D][light.state:454]: 'LED Strip' Turning ON
[13:48:12][D][light.state:461]:   Brightness: 39%
[13:48:12][D][light.state:486]:   Red=100%, Green=100%, Blue=0%
[13:48:12][D][light.state:496]:   Transition Length: 1000 ms
[13:48:12][D][light.state:503]:   Effect: 'statustest'
[13:48:12][D][statustest:030]: switch on led 1
[13:48:12][D][statustest:030]: switch on led 1
[13:48:13][D][light.state:454]: 'LED Strip' Turning ON
[13:48:13][D][light.state:461]:   Brightness: 39%
[13:48:13][D][light.state:486]:   Red=100%, Green=100%, Blue=0%
[13:48:13][D][light.state:496]:   Transition Length: 1000 ms
[13:48:13][D][light.state:503]:   Effect: 'statustest'
[13:48:13]E (14576) boot: Assert failed in esp_err_t nvs::PageManager::requestNewPage(), /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp:175 (usedEntries == newPage->getUsedEntryCount())

```

**Additional information:**

OttoWinter, 2019-02-05:
Yeah I've also seen this issue a couple of times. And it was always fixed by a complete erase+flash.

It's bad that esp-idf chose to make this an assert, as that obviously makes the whole system fail. Going to flag this as an external issue.

OttoWinter, 2019-03-24:
I believe this has been fixed in recent esp-idf update

closed 2019-03-24 by OttoWinter