esphome/issues #35: compiling gives a error

pvdw0310, 2019-02-05:
**Operating environment/Installation (Hass.io.):**

Using Hass.io

**ESP (ESP8266-12e Board):**

esp8266-12e

**Affected component:**

**Description of problem:**

Yesterday the compiling went well, now I get a error !

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphomeyaml/living_inkom
Processing living_inkom (platform: espressif8266@1.8.0; framework: arduino; board: esp12e)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing v1.10.1
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=346 @ 0.8.2
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=306 @ 1.1.1
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=126 @ 3.2.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=547 @ 2.4.1
Downloading...
Unpacking  [####################################]  100%
Traceback (most recent call last):
SCons import failed. Unable to find engine files in:
/root/.platformio/packages/tool-scons/script/../engine
File "/root/.platformio/packages/tool-scons/script/scons", line 190, in <module>
/root/.platformio/packages/tool-scons/script/scons-local-2.5.1
import SCons.Script
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/__init__.py", line 75, in <module>
/root/.platformio/packages/tool-scons/script/scons-local
import SCons.Action
/usr/lib/scons-2.5.1
ImportError: No module named Action
/usr/local/lib/scons-2.5.1
/usr/lib/python2.7/site-packages/scons-2.5.1
/usr/local/lib/python2.7/site-packages/scons-2.5.1
/usr/lib/scons-2.5.1
/usr/lib/scons
/usr/local/lib/scons
/usr/lib/python2.7/site-packages/scons
/usr/local/lib/python2.7/site-packages/scons
/usr/lib/scons
========================== [ERROR] Took 13.29 seconds ==========================
**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: living_inkom
  platform: ESP8266
  board: esp12e
  
wifi:
  ssid: 'xxxxxx'
  password: 'xxxxxxxxxxxx'
  
binary_sensor:
  - platform: gpio
    name: "Pin GPIO0"
    pin: GPIO0
    filters:
    - invert:
    on_press:
      then:
        - switch.toggle: lamp
      
    
switch:
  - platform: gpio
    name: "lamp"
    inverted: True
    pin: 2
    id: lamp

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2019-02-05:
And on what system is this? RPi, VM, etc? And what version/architecture?

pvdw0310, 2019-02-05:
Rpi 3B+  hass.io version 0.86.4

OttoWinter, 2019-03-24:
Closing due to inactivity

closed 2019-03-24 by OttoWinter