esphome/issues #36: MQTT lights are incompatible with home assistant 0.87

brandond, 2019-02-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip virtualenv - esphomelib v1.11.0-dev Jan 23 2019, 20:30:08

**ESP (ESP32/ESP8266/Board/Sonoff):**
esp8266

**Affected component:**
mqtt light

**Description of problem:**
MQTT lights are incompatible with home assistant 0.87, specifically this change: https://github.com/home-assistant/home-assistant/pull/20562.

Other MQTT components (switches, sensors, etc) seem to be OK.

**Traceback (if applicable):**
home assistant error log:
```
hass_1   | 2019-02-06 22:41:06 WARNING (MainThread) [homeassistant.components.mqtt.discovery] "platform": "mqtt_json" is deprecated, replace with "schema":"json"
hass_1   | 2019-02-06 22:41:06 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: light fixture01 fixture_01_light
hass_1   | 2019-02-06 22:41:06 ERROR (MainThread) [homeassistant.components.mqtt.light] Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'flash': True, 'name': 'Fixture 01 Light', 'platform': 'mqtt', 'state_topic': 'fixture01/light/fixture_01_light/state', 'command_topic': 'fixture01/light/fixture_01_light/command', 'availability_topic': 'fixture01/status', 'unique_id': 'ESPlightfixture_01_light', 'device': {'identifiers': '3c71bf2bd91a', 'name': 'fixture01', 'sw_version': 'esphomelib v1.11.0-dev Jan 23 2019, 20:30:08', 'model': 'PLATFORMIO_ESP01_1M', 'manufacturer': 'espressif'}, 'schema': 'json'},)
hass_1   | Traceback (most recent call last):
hass_1   |   File "/usr/src/app/homeassistant/components/mqtt/light/__init__.py", line 60, in async_discover
hass_1   |     config = PLATFORM_SCHEMA(discovery_payload)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/validators.py", line 207, in __call__
hass_1   |     return self._exec((Schema(val) for val in self.validators), v)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/validators.py", line 286, in _exec
hass_1   |     raise e if self.msg is None else AllInvalid(self.msg, path=path)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/validators.py", line 282, in _exec
hass_1   |     v = func(v)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 267, in __call__
hass_1   |     return self._compiled([], data)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 811, in validate_callable
hass_1   |     return schema(data)
hass_1   |   File "/usr/src/app/homeassistant/components/mqtt/light/__init__.py", line 36, in validate_mqtt_light
hass_1   |     return schemas[value[CONF_SCHEMA]](value)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 267, in __call__
hass_1   |     return self._compiled([], data)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
hass_1   |     return base_validate(path, iteritems(data), out)
hass_1   |   File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
hass_1   |     raise er.MultipleInvalid(errors)
hass_1   | voluptuous.error.MultipleInvalid: extra keys not allowed @ data['flash']
```

**Additional information:**

closed 2019-02-10 by OttoWinter

nicop4, 2019-02-12:
Hi @OttoWinter 
Any ETA for milestone 1.11.0 ? A lot of lights and sensor are broken at home :( Or a partial release with what is already fixed?
I don't want to put pressure, thanks for your great work 👍

OttoWinter, 2019-02-13:
@nicop4 I do not have ETAs for releases, I release things when they are ready. And there a few things I wish to get done before a release (some major refactors, like renaming the project). A partial release is also not really possible because these major refactors would make cherry picking commits a complete merge conflict hell.

The workaround is a) use native API or b) use dev version of esphome or c) use Home Assistant config script in the ESPHome dashboard to auto-generate the configuration.yaml entries for you HA config.

edif30, 2019-02-13:
@OttoWinter how does using native API fix the issue?  I'm using the API and I get the same tracebacks as the OP.

brandond, 2019-02-13:
The API does not use MQTT JSON so I'm not sure how you'd get a traceback from `homeassistant.components.mqtt.light`. Are you sure it's the same?

edif30, 2019-02-13:
@brandond I'm sure, I didn't have any mqtt: on the light I have.  Maybe it was an old mqtt log entry?  I can test later on.

OttoWinter, 2019-02-13:
@edif30 that error can also come from old retained discovery message and/or other devices.

edif30, 2019-02-13:
@OttoWinter thats what I'm thinking.  Since I replied to this thread I had deleted the mqtt db and cleaned out old entries but did not add the light: back in yet.  I will test.  Will probably work fine with API.

nicop4, 2019-02-16:
@OttoWinter thanks for the alternatives, I need the mqtt data because it is stored in an influx database for other purpose so API is not really an option now. I'll try to get dev version.

asadenton, 2019-02-16:
This is a one line change: https://github.com/brandond/esphome-core/commit/5b32af87e4f68e295bfb58bb19088635c3d8cbdb

After updating everything I just searched for and changed the file in the build directory directly and that fixed it. The problem with waiting to release this change is that everybody who has updated HA and uses this just lost all their lights.

choffee, 2019-02-17:
That fixed it for me.
Just ran the following to fix up all my lights, then build and upload

	find . -iname mqtt_json_light_component.cpp -exec sed -i '/flasd/d' {} \;