esphome/issues #38: compilation terminated fatal error: Wire.h: No such file or directory

mvturnho, 2019-02-08:
**Operating environment (Hass.io/Docker/pip/etc.):**

Windows 10 commandline and Visual C++
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266 nosemcu
**Affected component:**

https://esphomelib.com/esphomeyaml/custom/custom_component.html

**Description of problem:**
I added the library `I2Cdevlib-esphomelib` to the yaml file to use the MPR121 driver from my custom component.
Then when compiling the yaml file I get the compile errors.

```
**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.h:36:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.cpp:33:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

```

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: controlpanel
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version:
    branch: dev
  includes:
    - controlpanel/src/my_custom_sensor.h
  libraries:
    - I2Cdevlib-esphomelib

wifi:
  ssid: 'XXXXXXXXXX'
  password: 'XXXXXXXXXXXXXXXXX'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

i2c:
  sda: GPIO12
  scl: GPIO14
  scan: False

custom_component:
    - lambda: |-
        auto my_custom = new MyCustomComponent();
        return {my_custom};
```

Custom component `my_custom_sensor.h`
```cpp
#include "esphomelib/application.h"

using namespace esphomelib;
#include "MPR121.h"

class MyCustomComponent : public Component {
 public:
  void setup() override {
    // This will be called once to set up the component
    // think of it as the setup() call in Arduino

  }
  void loop() override {
    // This will be called very often after setup time.
    // think of it as the loop() call in Arduino

    ESP_LOGD("custom", "Message");
  }
};
```

**Traceback (if applicable):**

```
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|-- <FastLED> 3.2.0
|-- <Hash> 1.0
|-- <I2Cdevlib-esphomelib> 1.1.0
|-- <Wire> 1.0
|-- <esphomelib> 1.11.0-dev #0bea62d
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
Compiling .pioenvs\controlpanel\src\main.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\ADS1115.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\BMP085.cpp.o
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.h:38:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.cpp:35:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\ADS1115.h:40:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\ADS1115.cpp:36:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.h:36:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.cpp:33:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\ADS1115.cpp.o] Error 1
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o] Error 1
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\BMP085.cpp.o] Error 1
 [ERROR] Took 3.95 seconds

```

**Additional information:**

mvturnho, 2019-02-09:
@brandond and @OttoWinter , tried your solution, but this has not the deired effect.

I added i2c to my controlpanel.yaml file.

```yaml
esphomeyaml:
  name: controlpanel
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version:
        branch: dev
  includes: 
        - controlpanel/src/custom_i2c.h
  libraries:
        - Wire
        - I2Cdevlib-esphomelib

wifi:
  ssid: 'XXXXXXXXXXX'
  password: 'XXXXXXXXXXXXXXXXXX'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

i2c:
  sda: D2
  scl: D4
  scan: True
  
light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: D3
    num_leds: 16
    rgb_order: GRB
    name: 'LED Strip'
```

Compilation gives:

```
$ esphomeyaml controlpanel.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Updating esphomelib copy from git (controlpanel\.piolibdeps\esphomelib)
Already up to date.
INFO Compiling app...
INFO Running:  platformio run -d controlpanel
Processing controlpanel (platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 58 compatible libraries
Scanning dependencies...
Dependency Graph
[... 184 lines cut ...]
    -DUSE_MQTT
    -DUSE_NEW_OTA
    -DUSE_OTA
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -Wl,-Teagle.flash.4m.ld
    ${common.build_flags}
framework = arduino
lib_deps =
    AsyncMqttClient@0.8.2
    ESP8266WiFi
    ESP8266mDNS
    FastLED@3.2.0
    Hash
    I2Cdevlib-esphomelib
    Wire
    https://github.com/OttoWinter/esphomelib.git#dev
    ${common.lib_deps}
lib_ldf_mode = off
platform = espressif8266@1.8.0
upload_speed = 115200
; =========== AUTO GENERATED CODE END ============
; ========= YOU CAN EDIT AFTER THIS LINE =========
```

mvturnho, 2019-02-09:
With `lib_ldf_mode = on` comipaltion works fine.

How can I set this from the yaml file?

OttoWinter, 2019-02-09:
@mvturnho See https://esphome.io/components/esphomeyaml.html#platformio-options

closed 2019-03-17 by OttoWinter