esphome/issues #42: SHT3X-D Sensor Broken in DEV

mtl010957, 2019-02-09:
**Operating environment (Hass.io/Docker/pip/etc.):**
DEV branch (1.11.0-dev) using pip on Ubuntu 18.10

**ESP (ESP32/ESP8266/Board/Sonoff):**
Adafruit Huzzah ESP8266

**Affected component:**
[SHT3X-D](https://esphome.io/components/sensor/sht3xd.html)

**Description of problem:**
Reading of the SHT3X-D sensor is failing with the following message in the log:

[11:42:02][E][sensor.sht3xd:059]: Communication with SHT3xD failed!

See below for USB connected logging that has some additional details.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: ah02_th
  upper_devicename: AH02_TH

esphomeyaml: !include adafruit_huzzah/device.yaml
wifi: !include connection/wifi.yaml
mqtt: !include connection/mqtt.yaml
# Enable logging
logger: !include base/logger.yaml
# Enable Home Assistant API
#api: (broken with MQTT for now)
ota:
# Need I2C bus for sensors (defaults are good)
i2c:
# SHT31-D Temperature and Humidity
sensor: !include sensor/sht3x_d.yaml
# Blue LED (either output+light or status_led, not both)
output: !include adafruit_huzzah/output.yaml
light: !include adafruit_huzzah/light.yaml
#status_led: !include adafruit_huzzah/status_led.yaml
# Flash Button
binary_sensor: !include adafruit_huzzah/binary_sensor.yaml
```

sensor/sht3x_d.yaml contains:
```yaml
- platform: sht3xd
  temperature:
    name: "${upper_devicename} Temperature"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    expire_after: 600s
    filters:
      - or:
        - delta: 0.5
        - heartbeat: 300s
  humidity:
    name: "${upper_devicename} Humidity"
    unit_of_measurement: "%"
    accuracy_decimals: 1
    expire_after: 600s
    filters:
      - or:
        - delta: 0.5
        - heartbeat: 300s
  address: 0x44
  update_interval: 10s
```

**Traceback (if applicable):**

```
[12:04:01][C][ota:462]: There have been 1 suspected unsuccessful boot attempts.
[12:04:01][I][application:055]: Running through setup()...
[12:04:01][C][output.esp8266_pwm:023]: Setting up ESP8266 PWM Output...
[12:04:01][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'AH02_TH Button'...
[12:04:01][D][binary_sensor:037]: 'AH02_TH Button': Sending state OFF
[12:04:01][C][light.state:149]: Setting up light 'AH02_TH Blue LED'...
[12:04:01][C][sensor.sht3xd:038]: Setting up SHT3xD...
[12:04:01][E][sensor.sht3xd:131]: CRC8 Checksum invalid! 0xFF != 0x0E
[12:04:01][E][component:163]: Component was marked as failed.
[12:04:01][C][wifi:029]: Setting up WiFi...
[12:04:01][D][wifi:259]: Starting scan...
[12:04:07][D][wifi:274]: Found networks:
[12:04:07][I][wifi:307]: - 'REDACTED' (C8:3A:35:FB:18:41) ▂▄▆█
[12:04:07][D][wifi:308]:     Channel: 9
[12:04:07][D][wifi:309]:     RSSI: -74 dB
[12:04:07][I][wifi:178]: WiFi Connecting to 'REDACTED'...
[12:04:11][I][wifi:348]: WiFi connected!
[12:04:11][C][wifi:239]:   SSID: 'REDACTED'
[12:04:11][C][wifi:240]:   IP Address: REDACTED
[12:04:11][C][wifi:242]:   BSSID: REDACTED
[12:04:11][C][wifi:244]:   Hostname: 'ah02_th'
[12:04:11][C][wifi:249]:   Signal strength: -77 dB ▂▄▆█
[12:04:11][C][wifi:250]:   Channel: 9
[12:04:11][C][wifi:251]:   Subnet: 255.255.255.0
[12:04:11][C][wifi:252]:   Gateway: REDACTED
[12:04:11][C][wifi:253]:   DNS1: 75.75.75.75
[12:04:11][C][wifi:254]:   DNS2: 75.75.76.76
[12:04:11][C][ota:129]: Over-The-Air Updates:
[12:04:11][C][ota:130]:   Address: REDACTED
[12:04:11][C][mqtt.client:028]: Setting up MQTT...
[12:04:11][D][mqtt.client:115]: Resolving MQTT broker IP address...
[12:04:12][D][mqtt.client:148]: Resolved broker IP address to REDACTED
[12:04:12][I][mqtt.client:174]: Connecting to MQTT...
[12:04:12][I][mqtt.client:214]: MQTT Connected!
[12:04:12][I][application:090]: setup() finished successfully!
[12:04:12][I][application:098]: You're running esphomelib v1.11.0-dev compiled on Feb  9 2019, 11:44:05
[12:04:12][C][wifi:340]: WiFi:
[12:04:12][C][wifi:239]:   SSID: 'REDACTED'
[12:04:12][C][wifi:240]:   IP Address: REDACTED
[12:04:12][C][wifi:242]:   BSSID: REDACTED
[12:04:12][C][wifi:244]:   Hostname: 'ah02_th'
[12:04:12][C][wifi:249]:   Signal strength: -77 dB ▂▄▆█
[12:04:12][C][wifi:250]:   Channel: 9
[12:04:12][C][wifi:251]:   Subnet: 255.255.255.0
[12:04:12][C][wifi:252]:   Gateway: REDACTED
[12:04:12][C][wifi:253]:   DNS1: 75.75.75.75
[12:04:12][C][wifi:254]:   DNS2: 75.75.76.76
[12:04:12][C][i2c:045]: I2C Bus:
[12:04:12][C][i2c:046]:   SDA Pin: GPIO4
[12:04:12][C][i2c:047]:   SCL Pin: GPIO5
[12:04:12][C][output.esp8266_pwm:028]: ESP8266 PWM:
[12:04:12][C][output.esp8266_pwm:029]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[12:04:12][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[12:04:12][C][binary_sensor.gpio:022]: GPIO Binary Sensor 'AH02_TH Button'
[12:04:12][C][binary_sensor.gpio:023]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[12:04:12][C][logger:126]: Logger:
[12:04:12][C][logger:127]:   Level: DEBUG
[12:04:12][C][logger:128]:   Log Baud Rate: 115200
[12:04:12][C][light.state:266]: Light 'AH02_TH Blue LED'
[12:04:12][C][light.state:268]:   Default Transition Length: 1000 ms
[12:04:12][C][light.state:269]:   Gamma Correct: 2.80
[12:04:12][C][sensor.sht3xd:056]: SHT3xD:
[12:04:12][C][sensor.sht3xd:057]:   Address: 0x44
[12:04:12][E][sensor.sht3xd:059]: Communication with SHT3xD failed!
[12:04:12][C][sensor.sht3xd:061]:   Update Interval: 10000 ms
[12:04:12][C][sensor.sht3xd:063]:   Temperature 'AH02_TH Temperature'
[12:04:12][C][sensor.sht3xd:063]:     Unit of Measurement: '°C'
[12:04:12][C][sensor.sht3xd:063]:     Accuracy Decimals: 1
[12:04:12][C][sensor.sht3xd:064]:   Humidity 'AH02_TH Humidity'
[12:04:12][C][sensor.sht3xd:064]:     Unit of Measurement: '%'
[12:04:12][C][sensor.sht3xd:064]:     Accuracy Decimals: 1
[12:04:12][C][sensor.sht3xd:064]:     Icon: 'mdi:water-percent'
[12:04:12][C][ota:129]: Over-The-Air Updates:
[12:04:12][C][ota:130]:   Address: REDACTED
[12:04:12][C][mqtt.client:062]: MQTT:
[12:04:12][C][mqtt.client:064]:   Server Address: REDACTED
[12:04:12][C][mqtt.client:065]:   Username: 'REDACTED'
[12:04:12][C][mqtt.client:066]:   Client ID: 'ah02_th-5ccf7f151481'
[12:04:12][C][mqtt.client:068]:   Discovery prefix: 'homeassistant'
[12:04:12][C][mqtt.client:069]:   Discovery retain: YES
[12:04:12][C][mqtt.client:071]:   Topic Prefix: 'ah02_th'
[12:04:12][C][mqtt.client:073]:   Log Topic: 'ah02_th/debug'
[12:04:12][C][mqtt.client:076]:   Availability: 'ah02_th/status'
[12:04:12][C][sensor.mqtt:030]: MQTT Sensor 'AH02_TH Temperature':
[12:04:12][C][sensor.mqtt:032]:   Expire After: 600s
[12:04:12][C][sensor.mqtt:034]:   State Topic: 'ah02_th/sensor/ah02_th_temperature/state'
[12:04:12][C][sensor.mqtt:030]: MQTT Sensor 'AH02_TH Humidity':
[12:04:12][C][sensor.mqtt:032]:   Expire After: 600s
[12:04:12][C][sensor.mqtt:034]:   State Topic: 'ah02_th/sensor/ah02_th_humidity/state'
[12:04:12][C][binary_sensor.mqtt:025]: MQTT Binary Sensor 'AH02_TH Button':
[12:04:12][C][binary_sensor.mqtt:026]:   State Topic: 'ah02_th/binary_sensor/ah02_th_button/state'
[12:04:12][C][light.mqtt_json:071]: MQTT Light 'AH02_TH Blue LED':
[12:04:12][C][light.mqtt_json:072]:   State Topic: 'ah02_th/light/ah02_th_blue_led/state'
[12:04:12][C][light.mqtt_json:072]:   Command Topic: 'ah02_th/light/ah02_th_blue_led/command'
[12:04:12][I][application:113]: Running through first loop()
[12:04:12][I][application:140]: First loop finished successfully!
[12:06:01][I][ota:154]: Boot seems successful, resetting boot loop counter.
```

**Additional information:**
This error is happening using 1.11.0-dev version of both esphomeyaml and esphomeib. The same configuration and hardware works properly in version 1.10.1. I'm still using MQTT communication rather than the native API at this time. I've switched to DEV to take advantage of the fix for MQTT discovery of lights failing with Home Assistant 0.87. I'll attempt to do some more debugging of this tomorrow.

OttoWinter, 2019-02-10:
Ok turns out this was the issue: https://github.com/esphome/esphome-core/commit/b11f29e1f592e605ba26829546a86115d6ef4878#diff-c64e80705a5a19a52469fc74ac312e73R49

It should be fixed by manually setting `frequency: 1000 Hz` in your i2c config. 1kHz is quite slow though for i2c and so I want to find a better default. Can you try running through a couple of frequencies to figure out what is the highest frequency that still works?

Thanks!

closed 2019-02-10 by OttoWinter

mtl010957, 2019-02-11:
Thanks, adding the frequency value did the trick! I tested my hardware combination and the upper limit is 100000 Hz. Anything above that value (even just 100001 Hz.) causes the same error on startup. 100000 Hz has been perfectly stable across multiple reboots and power up scenarios so I'd call it the safe upper limit. I didn't notice any response lag with frequencies down to 1000 Hz in this situation, the data coming from the one single SHT31-D is pretty small.