esphome/issues #44: Sonoff Dual (R1) no relays

chaolue, 2019-02-10:
Hugely impressed with your work. In the process of transitioning over my ESP8266 (mostly Sonoff) devices from Tasmota and have hit a problem setting up the Sonoff Dual (R1). 

It appears that using a GPIO Switch doesn't work because the Dual (R1) uses the TX/RX (GPIO1/3) pins for the relays. I have no trouble with relays on my other Sonoff (Basic, S20, SV) devices.

Have tried using `
logger:
baud_rate: 0 ` without success.

chaolue, 2019-02-11:
I realise now that the Dual (R1) communicates via UART to turn on and off the relays.

Closer to making it work but not exactly as pretty as I would like. Been playing with Template Switches but still not quite right.

For others looking for what to do with Dual (R1), here's a way to make it work:

```yaml
uart:
  baud_rate: 19200
  tx_pin: GPIO1

switch:
  - platform: uart
    name: "Light 1&2 Off"
    data: [0xA0, 0x04, 0x00, 0xA1]
    id: switch_lights_off
  - platform: uart
    name: "Light 1 On Light 2 Off"
    data: [0xA0, 0x04, 0x01, 0xA1]
    id: switch_light_1_on
  - platform: uart
    name: "Light 1 Off Light 2 On"
    data: [0xA0, 0x04, 0x02, 0xA1]
    id: switch_light_2_on
  - platform: uart
    name: "Light 1&2 On"
    data: [0xA0, 0x04, 0x03, 0xA1]
    id: switch_lights_on 
```
Can lambdas be used on UART `data:`?
Are there plans to extend the UART Switch to have `data_on:` and `data_off:`?

JasperZ, 2019-02-14:
I also have a Sonoff Dual (R1) which runs tasmota for now but I would like to see ESPHome on it.
My problem is that it's in a place where I can't access it easily. Nevertheless I startet writing a config for it which I think should work. But I'm to scared to flash it right now.

It would be great if you could give this config a try and tell me if it works!

**Untested Sonoff Dual (R1) config:**
```
esphomeyaml:
  name: jasper_room_lights
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: !secret wlan_ssid
  password: !secret wlan_password
  domain: .iot

api:
  password: jasper_room_lights_api_password
  
# Enable logging
logger:
  baud_rate: 0

ota:
  password: !secret jasper_room_lights_ota_password

uart:
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 19200

switch:
  - platform: uart
    id: relay_12_off
    data: [0xA0, 0x04, 0x00, 0xA1]
    
  - platform: uart
    id: relay_1_on
    data: [0xA0, 0x04, 0x01, 0xA1]
    
  - platform: uart
    id: relay_2_on
    data: [0xA0, 0x04, 0x02, 0xA1]
    
  - platform: uart
    id: relay_12_on
    data: [0xA0, 0x04, 0x03, 0xA1]
    
  - platform: template
    id: relay1
    name: "jasper_room_lights_relay1"
    turn_on_action:
      if:
        condition:
          switch.is_off: relay2
        then:
          - switch.turn_on: relay_1_on
        else:
          - switch.turn_on: relay_12_on
    turn_off_action:
      if:
        condition:
          switch.is_off: relay2
        then:
          - switch.turn_on: relay_12_off
        else:
          - switch.turn_on: relay_2_on
    optimistic: true
        
  - platform: template
    id: relay2
    name: "jasper_room_lights_relay2"
    turn_on_action:
      if:
        condition:
          switch.is_off: relay1
        then:
          - switch.turn_on: relay_2_on
        else:
          - switch.turn_on: relay_12_on
    turn_off_action:
      if:
        condition:
          switch.is_off: relay1
        then:
          - switch.turn_on: relay_12_off
        else:
          - switch.turn_on: relay_1_on
    optimistic: true
```

OttoWinter, 2019-03-24:
Ok so since dual is supported through the config seen above I will close this issue. If someone wants, they can create a cookbook entry for dual r1s with the sample config here.

closed 2019-03-24 by OttoWinter