esphome/issues #45: MQTT connection socket error

Michael0yodi, 2019-02-12:
Device: D1 Mini, Sonoff Basic R2

Device is losing connection all the time, its restored automatically. the same behavior on both devices, tested power saving modes and other wifi settings without success, device responds to ping while losing mqtt connection.

Home Assistant: 0.87.1
Mosquitto broker hassio:
Latest esphomeyaml 

testmini-dc4f221d5129, disconnecting.
1549955558: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549955558: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549955862: Socket error on client testmini-dc4f221d5129, disconnecting.
1549955862: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549955862: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956166: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956166: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956166: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956188: Saving in-memory database to /data/mosquitto.db.
1549956470: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956470: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956470: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956774: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956774: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956774: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549957078: Socket error on client testmini-dc4f221d5129, disconnecting.
1549957078: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549957078: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').

Home assistant logbook:
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:43
Test GP1 turned off
08:43
Test GP1 changed to unavailable
08:37
Test GP1 turned off
08:37
Test GP1 changed to unavailable
08:32
Test GP1 turned off
08:32
Test GP1 changed to unavailable
08:27
Test GP1 turned off
08:27
Test GP1 changed to unavailable
08:22
Test GP1 turned off
08:22
Test GP1 changed to unavailable
08:17
Test GP1 turned off
08:17
Test GP1 changed to unavailable
08:12
Test GP1 turned off
08:12
Test GP1 changed to unavailable
08:07
Test GP1 turned off
08:07
Test GP1 changed to unavailable

OttoWinter, 2019-02-12:
The HA logbook logs are expected (see availability, you can turn that off). That's not the issue here.

The mosquitto logs unfortunately also do not contain much information. 

The main issue is that the ESP SDK is kind of unstable for certain network setups. That's an external issue in espressif's (closed source) code and therefore we cannot do anything against that.

For the logs, what would help the most would be USB logs directly from the device. For connection issues you obviously won't get any good logs over the air, because there's not network connection.

@bk8190 Let's keep this thread here to MQTT only please.

Michael0yodi, 2019-02-12:
What is the proper way to provide the necessary logs via usb?

aero83, 2019-02-13:
Got the chance to check serial debug.
It seems to be rebooting with error below, with the whole log below. Is there something I have configured incorrectly in HA?  I'm pretty new with it. 
```
[E][api:063]: No client connected to API. Rebooting...
[I][helpers:277]: Forcing a reboot... Reason: 'api'
```

```
[C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[I][application:053]: Running through setup()...
[C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Sonoff Basic Button'...
[D][binary_sensor:027]: 'Sonoff Basic Button': Sending state OFF
[C][switch.gpio:023]: Setting up GPIO Switch 'Sonoff Basic Relay'...
[D][switch:055]: 'Sonoff Basic Relay': Sending state OFF
[C][output.esp8266_pwm:023]: Setting up ESP8266 PWM Output...
[C][light.state:149]: Setting up light 'Sonoff Basic Green LED'...
[C][wifi:029]: Setting up WiFi...
[D][wifi:260]: Starting scan...
[D][wifi:275]: Found networks:
[I][wifi:308]: - 'NSA' (2C:30:33:F6:EA:27) ▂▄▆█
[D][wifi:309]:     Channel: 6
[D][wifi:310]:     RSSI: -27 dB
[D][wifi:312]: - 'ATTwSINXi2' (C0:A0:0D:3C:B8:20) ▂▄▆█
[D][wifi:312]: - 'Banditmachine' (A4:56:CC:09:BB:9E) ▂▄▆█
[D][wifi:312]: - '' (A4:56:CC:09:BB:9F) ▂▄▆█
[D][wifi:312]: - '' (A4:56:CC:09:BB:A1) ▂▄▆█
[D][wifi:312]: - '' (B2:FC:0D:DC:D3:31) ▂▄▆█
[D][wifi:312]: - 'My home' (A4:56:CC:97:50:06) ▂▄▆█
[D][wifi:312]: - '' (A4:56:CC:97:50:07) ▂▄▆█
[D][wifi:312]: - '' (A4:56:CC:97:50:09) ▂▄▆█
[D][wifi:312]: - '' (A4:56:CC:97:50:0B) ▂▄▆█
[D][wifi:312]: - 'KKGNET' (C0:C1:C0:DC:6A:FD) ▂▄▆█
[D][wifi:312]: - 'Brent' (58:6D:8F:1D:BD:BB) ▂▄▆█
[D][wifi:312]: - 'NETGEAR81' (B0:B9:8A:3A:30:B3) ▂▄▆█
[D][wifi:312]: - '' (7A:E1:03:88:83:99) ▂▄▆█
[D][wifi:312]: - 'ATT2LKF634 2.4' (E0:22:02:56:08:ED) ▂▄▆█
[D][wifi:312]: - '87C883' (F8:F5:32:87:C8:83) ▂▄▆█
[D][wifi:312]: - 'WoodruffMesh' (50:0F:F5:37:5C:B9) ▂▄▆█
[I][wifi:179]: WiFi Connecting to 'NSA'...
[I][wifi:349]: WiFi connected!
[C][wifi:240]:   SSID: 'NSA'
[C][wifi:241]:   IP Address: 192.168.10.135
[C][wifi:243]:   BSSID: 2C:30:33:F6:EA:27
[C][wifi:245]:   Hostname: 'ripper_lamp'
[C][wifi:250]:   Signal strength: -39 dB ▂▄▆█
[C][wifi:251]:   Channel: 6
[C][wifi:252]:   Subnet: 255.255.255.0
[C][wifi:253]:   Gateway: 192.168.10.1
[C][wifi:254]:   DNS1: 192.168.10.10
[C][wifi:255]:   DNS2: 192.168.10.10
[C][api:023]: Setting up Home Assistant API server...
[C][ota:129]: Over-The-Air Updates:
[C][ota:130]:   Address: 192.168.10.135:8266
[C][ota:132]:   Using Password.
[C][mqtt.client:024]: Setting up MQTT...
[I][mqtt.client:170]: Connecting to MQTT...
[I][mqtt.client:209]: MQTT Connected!
[W][mqtt.client:384]: Publish failed for topic='ripper_lamp/switch/sonoff_basic_relay/state' will retry later..
[W][mqtt.client:384]: Publish failed for topic='ripper_lamp/light/sonoff_basic_green_led/state' will retry later..
[I][application:089]: setup() finished successfully!
[I][application:097]: You're running esphomelib v1.10.1 compiled on Feb 12 2019, 08:14:21
[C][wifi:341]: WiFi:
[C][wifi:240]:   SSID: 'NSA'
[C][wifi:241]:   IP Address: 192.168.10.135
[C][wifi:243]:   BSSID: 2C:30:33:F6:EA:27
[C][wifi:245]:   Hostname: 'ripper_lamp'
[C][wifi:250]:   Signal strength: -39 dB ▂▄▆█
[C][wifi:251]:   Channel: 6
[C][wifi:252]:   Subnet: 255.255.255.0
[C][wifi:253]:   Gateway: 192.168.10.1
[... 130 lines cut ...]
[C][api:073]:   Address: 192.168.10.135:6053
[C][ota:129]: Over-The-Air Updates:
[C][ota:130]:   Address: 192.168.10.135:8266
[C][ota:132]:   Using Password.
[C][mqtt.client:058]: MQTT:
[C][mqtt.client:060]:   Server Address: 192.168.10.45:1883 (192.168.10.45)
[C][mqtt.client:061]:   Username: ''
[C][mqtt.client:062]:   Client ID: 'ripper_lamp-5ccf7f958d57'
[C][mqtt.client:064]:   Discovery prefix: 'homeassistant'
[C][mqtt.client:065]:   Discovery retain: YES
[C][mqtt.client:067]:   Topic Prefix: 'ripper_lamp'
[C][mqtt.client:069]:   Log Topic: 'ripper_lamp/debug'
[C][mqtt.client:072]:   Availability: 'ripper_lamp/status'
[C][binary_sensor.mqtt:025]: MQTT Binary Sensor 'Sonoff Basic Button':
[C][binary_sensor.mqtt:026]:   State Topic: 'ripper_lamp/binary_sensor/sonoff_basic_button/state'
[C][switch.mqtt:048]: MQTT switch 'Sonoff Basic Relay':
[C][switch.mqtt:049]:   State Topic: 'ripper_lamp/switch/sonoff_basic_relay/state'
[C][switch.mqtt:049]:   Command Topic: 'ripper_lamp/switch/sonoff_basic_relay/command'
[C][light.mqtt_json:072]: MQTT Light 'Sonoff Basic Green LED':
[C][light.mqtt_json:073]:   State Topic: 'ripper_lamp/light/sonoff_basic_green_led/state'
[C][light.mqtt_json:073]:   Command Topic: 'ripper_lamp/light/sonoff_basic_green_led/command'
[I][application:114]: Running through first loop()
[I][application:141]: First loop finished successfully!
[I][ota:154]: Boot seems successful, resetting boot loop counter.
```

OttoWinter, 2019-02-13:
@aero83 You have both mqtt and api enabled. When no client connects to the native api within 5 minutes a watchdog kicks in and reboots the device. Please disable the native api if you don't use it.

Michael0yodi, 2019-02-13:
@OttoWinter Disabling API solved my issue. I did not reflect on this setting since (If I recall correctly) it where added in the guide when creating the initial configuration, even if stated in the documentation that it force a reboot every 5 minute, perhaps this can be further elaborated somehow or not included as default in the generated .yaml, anyway thanks for the support! :)

aero83, 2019-02-14:
Thank you. Solved issue, thought it might be a noob mistake.

OttoWinter, 2019-02-14:
Ok then, I guess this issue can be closed then because the issue has been resolved.

For documentation: A PR in the docs repo to add a note would be welcome :)

closed 2019-02-14 by OttoWinter