esphome/issues #48: Monochromatic light. PWM. Dimming

erazor666, 2019-02-13:
I am not 100% sure this is an actual issue with esphome, but i'm not sure what is the cause.

I have some pwm driven monochromatic lights and they function like expected mostly. However when dimming them the dim 'curve' is a litte strange. From 0 to about 50-60% intensity they are kind of dull and not lighting up much, and from there and up to 100% it feels like too much is happening. The distrubution of light seems not to be completely linear.

My theories is something esphome related or the mosfet gates are not the optimal ones for this application. I'm using irfz44n mosfets and a wemos d1 esp8266. I dont know if i should expect other results using a esp32?

erazor666, 2019-02-13:
@yawor 
I have some mosfets that say just IRZ44N also, i'm browsing aliexpress but there is a lot of 'sloppy' naming on things there.
For instance: https://www.aliexpress.com/item/10PCS-IRLZ44N-TO-220-IRLZ44-TO220-IRLZ44NPBF-free-shipping/32714396199.html
I don't know how significant the xxxxN is. I'm by no means an expert on the subject.

yawor, 2019-02-13:
@erazor666 well I'm neither :smile:. I don't think the N part is that significant in this case.
From what I've found out the N variant has slightly lower max Drain-Source current and voltage, but also has a slightly lower Drain-Source resistance when turned on (about 0.01 Ohm) and better response times. I think you'll be good with either.

erazor666, 2019-02-13:
@yawor 
I was looking around my stash, i also seem to have some MOSFETS labeled IRLB8721, have you any experience with them?

yawor, 2019-02-14:
@erazor666 according to the datasheet I've found (exactly for IRLB8721PbF) it should work. The typical Gate-Source threshold max voltage is 2.35 v so it turns on fully above that - no problem driving it from 3.3 v logic. The Source-Drain resistance is about 0.02 Ohm when opened. Max DS voltage is 30 v so no problem with driving 12 v LEDs. Max continuous current is 62 amps but it is given for Gate-Source voltage of 10 v, so it'll be lower for 3.3 v, but it's similar to IRLZ44, so I think it's safe to use it as a replacement. From what I understand it's also much faster, but at this speeds this doesn't really matter for PWM frequencies.

erazor666, 2019-02-14:
@yawor Wow you are just a well of knowledge, i will try out these instead of the IRFZ44 ones to verify if they are causing any strange behavior. There is no consequences in supplying them with too much voltage for the Gate-Source threshold? (since esp8266 logic is 3.3v, not that much over, but still) They seem safe to use for a 5m led strip for instance, the cheap ones are usually 5-10w meter (12V. ~0.85a per meter)

yawor, 2019-02-14:
The threshold voltage is a parameter which tells you at what voltage a mosfet starts to open (min threshold voltage) and when it's fully opened (max threshold voltage). It doesn't mean that that you can't go above max threshold voltage. This is limited by max Gate-Source voltage, which for IRLB8721PbF is 20 volts, so at 3.3 v you're safe :smile:.

erazor666, 2019-02-16:
i swapped out the IRFZ44 (for IRLB8721) without seeing that much difference.
At 100% light, they are as expected still, good light. It seems that most of the 'action' is still going on in the 100% to 50-60% range. and then not that much, so if the last 40-50% of the dimmer range could be 'removed' it would be more as expected.
There is light at say 10%, 20% etc, but its not really much difference in it.  

`light:
  - platform: monochromatic
    name: "led1"
    output: pwm_output
    gamma_correct: 0
  - platform: monochromatic
    name: "led2"
    output: pwm_output2
    gamma_correct: 0`

would this disable it?

OttoWinter, 2019-03-24:
Closing due to inactivity. I believe the issue here is user/hw error, esphome does drive the output correctly.

closed 2019-03-24 by OttoWinter