esphome/issues #49: Broken builds after commit

marcinkowalczyk, 2019-02-13:
Hi,

 There is broken builds comming after commit https://github.com/esphome/esphome-core/commit/896a32142fe6c4b7897218d6aa2190620e13c356  "Rename esphomelib to esphome-core"

it fails to resolve 

from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/mqtt/mqtt_client_component.h:12:29: fatal error: AsyncMqttClient.h: No such file or directory

I've checed b31dabb30f6195651f92875df1d27134d0690104 and it works like a charm.

OttoWinter, 2019-02-13:
Please include how you use esphome (what esphomeyaml version etc). If you're using esphomelib version dev, you must also upgrade esphomeyaml to dev (the two must be in sync)

marcinkowalczyk, 2019-02-13:
Hi,

```
root@7bce6681-esphomeyaml-edge:/config/esphomeyaml# esphomeyaml termometry-dol.yaml version
Version: 1.10.0-dev
root@7bce6681-esphomeyaml-edge:/config/esphomeyaml# mcedit termometry-dol.yaml

root@7bce6681-esphomeyaml-edge:/config/esphomeyaml# esphomeyaml termometry-dol.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d termometrydol
Processing termometrydol (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Error: Could not find `AsyncMqttClient` dependency for `esphomelib` library
================================================================================================================================================= [ERROR] Took 3.10 seconds =================================================================================================================================================
root@7bce6681-esphomeyaml-edge:/config/esphomeyaml#

```

config is simple:

```esphomeyaml:
  name: termometrydol
  platform: ESP32
  board: nodemcu-32s
  esphomelib_version: 1.10.0

wifi:
  ssid: 'SAGEM_6230_IoT'
  password: '7E7CFD66'

logger:
  level: DEBUG

api:

ota:

esp32_ble_tracker:
```

OttoWinter, 2019-02-13:
> If you're using esphomelib version dev, you must also upgrade esphomeyaml to dev (the two must be in sync)

Please install the latest esphomeyaml dev. (the one you're using is not the latest; also the `esphomeyaml` command has changed to `esphome`)

marcinkowalczyk, 2019-02-13:
Hi,

Seems it compiles now - but this mean, that hass.io docker images needs to be updated

OttoWinter, 2019-02-14:
The Hass.io image has also been updated (rebuild -edge image). For the stable and beta versions that will go out with the next release, but those are also not affected by this (because they pin the esphomelib version)

closed 2019-02-14 by OttoWinter