esphome/issues #50: Error in esphomeyaml with output custom component

ProASV, 2019-02-14:
Python 2.7.15
Windows 10

Example one:
esp32_test.yaml :

#------ test yaml file ------
esphomeyaml:
  name: esp32_test
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'TEST'
  password: 'TESTTESTTEST'

logger:
api:
ota:

output:
  - platform: custom
    type: float
    lambda: |-
      return 0;

    outputs:
      id: custom_float
#------ test yaml file ------

Compiling:

esphomeyaml esp32_test.yaml compile
INFO Reading configuration...
Failed config

output.custom: [source esp32_test.yaml:18]
  platform: custom

  [type] is an invalid option for [output.custom].
  type: float
  lambda: return 0;

  'platform' is a required option for [outputs].
  outputs:  [source esp32_test.yaml:24]
    id: custom_float

Example two:

#------ test yaml file ------
esphomeyaml:
  name: esp32_test
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'TEST'
  password: 'TESTTESTTEST'

logger:

# Enable Home Assistant API
api:

ota:

output:
  - platform: custom
#    type: float
    lambda: |-
      return 0;

    outputs:
      id: custom_float

#------ test yaml file ------

Compiling:
esphomeyaml esp32_test.yaml compile

INFO Reading configuration...
Failed config

output.custom: [source esp32_test.yaml:18]

  type not specified!. Got '{"platform": "custom", "lambda": "return 0;", "outputs": {"id": "custom_float"}}'
  platform: custom
  lambda: return 0;
  outputs:
    id: custom_float

brandond, 2019-02-14:
What version of esphome are you using? This passes config validation for me, although it's not valid as the lambda needs to return an instance of your custom component, not a value.

```yaml
output:
  - platform: custom
    type: float
    lambda: |-
      return nullptr;
    outputs:
      id: custom_float
```

ProASV, 2019-02-14:
Release version (pip2 install -U esphomeyaml) of esphome. The actual content of this  lambda is not so important in this case. My real code was:
```
output:
  - platform: custom
    type: 'float'
    lambda: |-
        auto my_custom_float_output = new NooliteFloatOutput(4);
        App.register_component(my_custom_float_output);
        return {my_custom_float_output};
    outputs:
        id: custom_float_bedroom_bra_left

```

brandond, 2019-02-15:
I would try dev. You'll have to update your YAML and command line, but I suspect it'll work there.

ProASV, 2019-02-15:
I see you have changed the implementation in dev version:
Was (esphomeyaml/components/output/custom.py ):
```BINARY_SCHEMA = output.PLATFORM_SCHEMA.extend({
    cv.GenerateID(): cv.declare_variable_id(CustomBinaryOutputConstructor),
    vol.Required(CONF_LAMBDA): cv.lambda_,
    vol.Required(CONF_OUTPUTS):
        cv.ensure_list(output.BINARY_OUTPUT_SCHEMA.extend({
            cv.GenerateID(): cv.declare_variable_id(output.BinaryOutput),
        })),
})

FLOAT_SCHEMA = output.PLATFORM_SCHEMA.extend({
    cv.GenerateID(): cv.declare_variable_id(CustomFloatOutputConstructor),
    vol.Required(CONF_LAMBDA): cv.lambda_,
    vol.Required(CONF_OUTPUTS):
        cv.ensure_list(output.FLOAT_OUTPUT_PLATFORM_SCHEMA.extend({
            cv.GenerateID(): cv.declare_variable_id(output.FloatOutput),
        })),
})
```

Now ():

```
BINARY_SCHEMA = output.PLATFORM_SCHEMA.extend({
    cv.GenerateID(): cv.declare_variable_id(CustomBinaryOutputConstructor),
    vol.Required(CONF_LAMBDA): cv.lambda_,
    vol.Required(CONF_TYPE): 'binary',
    vol.Required(CONF_OUTPUTS):
        cv.ensure_list(output.BINARY_OUTPUT_SCHEMA.extend({
            cv.GenerateID(): cv.declare_variable_id(output.BinaryOutput),
        })),
})

FLOAT_SCHEMA = output.PLATFORM_SCHEMA.extend({
    cv.GenerateID(): cv.declare_variable_id(CustomFloatOutputConstructor),
    vol.Required(CONF_LAMBDA): cv.lambda_,
    vol.Required(CONF_TYPE): 'float',
    vol.Required(CONF_OUTPUTS):
        cv.ensure_list(output.FLOAT_OUTPUT_SCHEMA.extend({
            cv.GenerateID(): cv.declare_variable_id(output.FloatOutput),
        })),
})
```

So, we can close this issue.

closed 2019-02-15 by ProASV