esphome/issues #52: Unable to compile with dev branch

mvturnho, 2019-02-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
Linux Ubuntu
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266 NodeMCU
**Affected component:**

I think it is the combination of esphome and esphome-core

**Description of problem:**
After upgrading with:
`pip install --upgrade git+https://github.com/esphome/esphome`

and using the dev version of esphome-core from my yaml file 
```yaml
  esphomelib_version:
    # Repository defaults to https://github.com/OttoWinter/esphomelib.git
    repository: https://github.com/esphome/esphome-core.git
    #repository: https://github.com/mvturnho/esphome-core.git
    branch: dev
```
**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: mpr121
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version:
    # Repository defaults to https://github.com/OttoWinter/esphomelib.git
    repository: https://github.com/esphome/esphome-core.git
    #repository: https://github.com/mvturnho/esphome-core.git
    branch: dev
  platformio_options:
        upload_speed: 115200
        lib_ldf_mode: on

wifi:
  ssid: 'XXXXXXX'
  password: 'XXXXXXXXXXXXXXXXXXXXX'
  domain : .visconti.nl

mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

i2c:
  sda: D2
  scl: D4
  scan: True

binary_sensor:
  - platform: gpio
    name: "dingdong"
    pin:
      number: D6
      inverted: False
      mode: INPUT

```

**Traceback (if applicable):**

```
esphomeyaml mpr121.yaml compile                                  
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d mpr121
Processing mpr121 (platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=346 @ 0.8.2
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ ^1.1.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing esphome-core
git version 2.17.1
Cloning into '/home/mvturnho/esphomelib/mpr121/.piolibdeps/_tmp_installing-KiRvkC-package'...
remote: Enumerating objects: 376, done.
remote: Counting objects: 100% (376/376), done.
remote: Compressing objects: 100% (327/327), done.
remote: Total 376 (delta 47), reused 352 (delta 46), pack-reused 0
Receiving objects: 100% (376/376), 408.73 KiB | 1.22 MiB/s, done.
Resolving deltas: 100% (47/47), done.
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <Wire> 1.0
|-- <esphome-core> 1.11.0-dev #9fc874e
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
Compiling .pioenvs/mpr121/src/main.cpp.o
Generating LD script .pioenvs/mpr121/ld/eagle.app.v6.common.ld
In file included from src/main.cpp:3:0:
.piolibdeps/esphome-core/src/esphomelib/application.h:2:2: warning: #warning "This header file is deprecated, please include esphome.h instead." [-Wcpp]
#warning "This header file is deprecated, please include esphome.h instead."
^
In file included from .piolibdeps/esphome-core/src/esphome/application.h:5:0,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/defines.h:21:0: warning: "USE_OTA" redefined [enabled by default]
#define USE_OTA
^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .piolibdeps/esphome-core/src/esphome/application.h:5:0,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/defines.h:22:0: warning: "USE_I2C" redefined [enabled by default]
#define USE_I2C
^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .piolibdeps/esphome-core/src/esphome/application.h:5:0,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/defines.h:23:0: warning: "USE_BINARY_SENSOR" redefined [enabled by default]
#define USE_BINARY_SENSOR
^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .piolibdeps/esphome-core/src/esphome/application.h:5:0,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/defines.h:24:0: warning: "USE_GPIO_BINARY_SENSOR" redefined [enabled by default]
#define USE_GPIO_BINARY_SENSOR
^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .piolibdeps/esphome-core/src/esphome/application.h:5:0,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/defines.h:135:0: warning: "USE_MQTT" redefined [enabled by default]
#define USE_MQTT
^
<command-line>:0:0: note: this is the location of the previous definition
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:13:38: error: 'ESPHOMELIB_LOG_LEVEL_DEBUG' was not declared in this scope
logcomponent->set_global_log_level(ESPHOMELIB_LOG_LEVEL_DEBUG);
^
Compiling .pioenvs/mpr121/lib5b8/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
*** [.pioenvs/mpr121/src/main.cpp.o] Error 1

```

**Additional information and things you've tried:**

brandond, 2019-02-14:
You want to run 'esphome' now, not 'esphomeyaml'. The old command runs the old module.

@OttoWinter could probably handle this a bit better by including a 'dummy' Python module in the old name that included a replacement. Since the module name has changed, pip doesn't remove the old one, and you end up with both esphome and esphomeyaml available.

mvturnho, 2019-02-14:
@brandond  Thanks, I did not think of this. also the docs do not mention that you need to change your yaml file yet. So for other users that run into this problem. My `yaml` file had to be modified for the `esphomeyaml:` section to `exphome:` like this:

```yaml
esphome:
  name: mpr121
  platform: ESP8266
  board: nodemcuv2
  esphome_core_version:
    repository: https://github.com/esphome/esphome-core.git
    branch: dev
```

OttoWinter, 2019-02-15:
@brandond Yeah, that's probably good. Didn't know about the pip not removing thing. Fixed by https://github.com/esphome/esphome/commit/50af0da13f6e94430b90d11c83d279f08228bc4a

closed 2019-02-15 by OttoWinter