esphome/issues #53: Can't configure module Ethernet on Olimex ESP32-PoE

readylan, 2019-02-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
I'm using Ubuntu 16.04 LTS

The board is [ESP32-POE from Olimex](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware).

**Affected component:**

Ethernet Component
https://esphome.io/components/ethernet.html?highlight=ethernet

**Description of problem:**
I can compile my definition file (.yaml) and flash the board but inmediatly after, the wizard throws non-stop errors that are repeated continuously:
```
Hard resetting via RTS pin...
======================================================= [SUCCESS] Took 55.83 seconds =======================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[14:15:25][I][logger:071]: Log initialized
[14:15:25][C][ota:461]: There have been 8 suspected unsuccessful boot attempts.
[14:15:25][I][application:053]: Running through setup()...
[14:15:25][C][ethernet:028]: Setting up Ethernet...
[14:15:25]E (22) tcpip_adapter: tcpip_adapter is not initialized!
[14:15:25]abort() was called at PC 0x400dd87f on core 1
[14:15:25]
[14:15:26]Backtrace: 0x40091a54:0x3ffb1df0 0x40091c57:0x3ffb1e10 0x400dd87f:0x3ffb1e30 0x400d3eff:0x3ffb1e70 0x400d4076:0x3ffb1eb0 0x400d3576:0x3ffb1ef0 0x400d3327:0x3ffb1f10 0x400d2159:0x3ffb1f60 0x4012bad7:0x3ffb1fa0
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d ./esp32poe -t idedata
WARNING Decoded 0x40091a54: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x40091c57: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x400dd87f: tcpip_adapter_dhcpc_start at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c:1191
WARNING Decoded 0x400d3eff: esphomelib::EthernetComponent::start_connect_()
WARNING Decoded 0x400d4076: esphomelib::EthernetComponent::setup()
WARNING Decoded 0x400d3576: esphomelib::Component::setup_()
WARNING Decoded 0x400d3327: esphomelib::Application::setup()
WARNING Decoded 0x400d2159: setup()
WARNING Decoded 0x4012bad7: loopTask(void*)
[14:15:28]
[14:15:28]Rebooting...
[14:15:28]ets Jun  8 2016 00:22:57
[14:15:28]
[14:15:28]rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[14:15:28]configsip: 0, SPIWP:0xee
[14:15:28]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:15:28]mode:DIO, clock div:2
[14:15:28]load:0x3fff0018,len:4
[14:15:28]load:0x3fff001c,len:952
[14:15:28]load:0x40078000,len:6084
[14:15:28]load:0x40080000,len:7944
[14:15:28]entry 0x40080310
[14:15:28][I][logger:071]: Log initialized
[14:15:28][C][ota:461]: There have been 9 suspected unsuccessful boot attempts.
[14:15:28][I][application:053]: Running through setup()...
[14:15:28][C][ethernet:028]: Setting up Ethernet...
[14:15:28]E (22) tcpip_adapter: tcpip_adapter is not initialized!
[14:15:28]abort() was called at PC 0x400dd87f on core 1
[14:15:28]
[14:15:28]Backtrace: 0x40091a54:0x3ffb1df0 0x40091c57:0x3ffb1e10 0x400dd87f:0x3ffb1e30 0x400d3eff:0x3ffb1e70 0x400d4076:0x3ffb1eb0 0x400d3576:0x3ffb1ef0 0x400d3327:0x3ffb1f10 0x400d2159:0x3ffb1f60 0x4012bad7:0x3ffb1fa0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40091a54: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x40091c57: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x400dd87f: tcpip_adapter_dhcpc_start at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c:1191
WARNING Decoded 0x400d3eff: esphomelib::EthernetComponent::start_connect_()
WARNING Decoded 0x400d4076: esphomelib::EthernetComponent::setup()
WARNING Decoded 0x400d3576: esphomelib::Component::setup_()
WARNING Decoded 0x400d3327: esphomelib::Application::setup()
WARNING Decoded 0x400d2159: setup()
WARNING Decoded 0x4012bad7: loopTask(void*)
[14:15:28]
[14:15:28]Rebooting...
```
I'm not sure of my definition is valid as I can only have an Arduino sketch running properly using board's ethernet. [This sketch](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720_internal_clock/ETH_LAN8720_internal_clock.ino) initialices the board this way:
```
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     15

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    2
```
while my file.yaml have this:
```
esphomeyaml:
  name: esp32poe
  platform: ESP32
  board: esp-wrover-kit

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
```
I have tried many pin combinations without success.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: esp32poe
  platform: ESP32
  board: esp-wrover-kit

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

readylan, 2019-02-15:
```
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   12

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18
```
(these are the correct values in the arduino sketch)

setola, 2019-02-20:
Hi there,
I can confirm this behavior.

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
I'm running Ubuntu 18LTS with docker
```bash
$lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

$docker --version
Docker version 18.09.2, build 6247962
```

**ESP (ESP32/ESP8266, Board/Sonoff):**
My board is a [Olimex ESP32-PoE Rev C](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware); here's the [schematics](https://github.com/OLIMEX/ESP32-POE/raw/master/HARDWARE/ESP32-PoE-hardware-revision-C/ESP32-PoE_Rev_C_Color.pdf)

**Additional information and things you've tried:**

* I tried to guess the pin configuration from code like [this](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet/ESP32_PoE_Ethernet/main/ethernet/olimex_ethernet.c)
* I tried to remove power and re-apply it (because of [this topic](https://www.olimex.com/forum/index.php?topic=6699.0))
* for safety reasons (missing galvanic insulation on board) I'm **not** using POE power, but only usb power from a dell xps.

**Problem-relevant YAML-configuration entries:**
```yaml
# Example configuration entry
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 0
  power_pin: GPIO12

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.90
    gateway: 192.168.1.1
    subnet: 255.255.255.0
```

**Traceback:**
I think is very much the same as the first post... anyway here's for completeness:
```
[15:05:34]Rebooting...
[15:05:34]ets Jun  8 2016 00:22:57
[15:05:34]
[15:05:34]rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[15:05:34]configsip: 0, SPIWP:0xee
[15:05:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:05:34]mode:DIO, clock div:2
[15:05:34]load:0x3fff0018,len:4
[15:05:34]load:0x3fff001c,len:952
[15:05:34]load:0x40078000,len:6084
[15:05:34]load:0x40080000,len:7944
[15:05:34]entry 0x40080310
[15:05:34][I][logger:071]: Log initialized
[15:05:34][C][ota:461]: There have been 6 suspected unsuccessful boot attempts.
[15:05:34][I][application:053]: Running through setup()...
[15:05:34][C][switch.gpio:023]: Setting up GPIO Switch 'led'...
[15:05:34][D][switch:055]: 'led': Sending state OFF
[15:05:34][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'input 4'...
[15:05:34][D][binary_sensor:027]: 'input 4': Sending state OFF
[15:05:34][C][ethernet:028]: Setting up Ethernet...
[15:05:34]E (51) tcpip_adapter: tcpip_adapter is not initialized!
[15:05:34]abort() was called at PC 0x400e2e1b on core 1
[15:05:34]
[15:05:34]Backtrace: 0x40091aa4:0x3ffb1d80 0x40091ca7:0x3ffb1da0 0x400e2e1b:0x3ffb1dc0 0x400d5349:0x3ffb1e00 0x400d552e:0x3ffb1e40 0x400d463a:0x3ffb1e80 0x400d39bf:0x3ffb1ea0 0x400d2586:0x3ffb1ef0 0x40133c0f:0x3ffb1fa0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40091aa4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x40091ca7: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
WARNING Decoded 0x400e2e1b: tcpip_adapter_dhcpc_stop at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c:1191
WARNING Decoded 0x400d5349: esphomelib::EthernetComponent::start_connect_()
WARNING Decoded 0x400d552e: esphomelib::EthernetComponent::setup()
WARNING Decoded 0x400d463a: esphomelib::Component::setup_()
WARNING Decoded 0x400d39bf: esphomelib::Application::setup()
WARNING Decoded 0x400d2586: setup()
WARNING Decoded 0x40133c0f: loopTask(void*)
```

OttoWinter, 2019-02-20:
I believe this should be fixed by - could you try the latest beta (https://beta.esphome.io)

setola, 2019-02-20:
Just to be sure: for testing the latest beta I can use the image *esphome/esphome:beta*, isn't it?
```bash
$docker-compose exec esphome esphome test1.yaml version
Version: 1.11.0b2
```

On this version my board doesn't reboot, but she's still not able to connect neither via DHCP or static ip.
To exclude a cable issue, I've used the one connection my laptop docking station.

```
INFO Reading configuration...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[17:31:38]<here I've manually reset the board so there a stream of random chars>
[I][logger:122]: Log initialized
[17:31:39][C][ota:480]: There have been 8 suspected unsuccessful boot attempts.
[17:31:39][I][application:055]: Running through setup()...
[17:31:39][C][ethernet:028]: Setting up Ethernet...
[17:31:54][W][ethernet:044]: Connecting via ethernet failed! Re-connecting...
```
if I set clk_mode
```yaml
ethernet:
  clk_mode: GPIO17_OUT
```
I get this:
```
[17:38:41]E (32136) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
[17:38:42]E (33136) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
[17:38:43]E (34136) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
```
On the schematics the `XTAL1/CLKIN` pin of the *LAN8701* is connected to `GPIO17/EMAC_CLK_OUT_180` og the *ESP32*, so presume (not sure) `GPIO17_OUT` should be the correct value.

Is it possible to have more verbose output?

OttoWinter, 2019-02-20:
@setola yes the image is correct.
for verbose output see level setting in logger

setola, 2019-02-22:
Thank you @OttoWinter,
it's working on version 1.11.0b3:
```bash
$docker-compose exec esphome esphome test1.yaml version
Version: 1.11.0b3
```

For info, this is the pin configuration I'm using:
```yaml
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
```

added this to documentation with [PR#166](https://github.com/esphome/esphome-docs/pull/166)

readylan, 2019-02-22:
Hello !
I've tried now with that config and my board is working as well :)
```
Hard resetting via RTS pin...
======================================================= [SUCCESS] Took 61.48 seconds =======================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:59:32][W][ethernet:044]: Connecting via ethernet failed! Re-connecting...
..... many more warnings during 7 seconds ......
[15:59:39][I][ethernet:055]: Connected via Ethernet!
[15:59:39][C][ethernet:229]:   IP Address: 192.168.1.170
[15:59:39][C][ethernet:230]:   Hostname: 'puerta'
[15:59:39][C][ethernet:231]:   Subnet: 255.255.255.0
[15:59:39][C][ethernet:232]:   Gateway: 192.168.1.1
[15:59:39][C][ethernet:235]:   DNS1: 192.168.1.1
[15:59:39][C][ethernet:237]:   DNS2: 0.0.0.0
[15:59:39][C][ethernet:241]:   MAC Address: 24:0A:C4:9E:FF:F7
[15:59:39][C][ethernet:242]:   Is Full Duplex: YES
[15:59:39][C][ethernet:243]:   Link Up: YES
[15:59:39][C][ethernet:244]:   Link Speed: 100
[15:59:39][C][api:023]: Setting up Home Assistant API server...
[15:59:39][C][ota:129]: Over-The-Air Updates:
[15:59:39][C][ota:130]:   Address: puerta.local:3232
[15:59:39][I][application:090]: setup() finished successfully!
[15:59:39][I][application:098]: esphome-core version 1.12.0-dev compiled on Feb 22 2019, 15:56:54
[15:59:39][C][logger:159]: Logger:
[15:59:39][C][logger:160]:   Level: DEBUG
[15:59:39][C][logger:161]:   Log Baud Rate: 115200
[15:59:39][C][logger:162]:   Hardware UART: UART0
[15:59:39][C][ethernet:069]: Ethernet:
[15:59:39][C][ethernet:229]:   IP Address: 192.168.1.170
[15:59:39][C][ethernet:230]:   Hostname: 'puerta'
[15:59:39][C][ethernet:231]:   Subnet: 255.255.255.0
[15:59:39][C][ethernet:232]:   Gateway: 192.168.1.1
[15:59:39][C][ethernet:235]:   DNS1: 192.168.1.1
[15:59:39][C][ethernet:237]:   DNS2: 0.0.0.0
[15:59:39][C][ethernet:241]:   MAC Address: 24:0A:C4:9E:FF:F7
[15:59:39][C][ethernet:242]:   Is Full Duplex: YES
[15:59:39][C][ethernet:243]:   Link Up: YES
[15:59:39][C][ethernet:244]:   Link Speed: 100
[15:59:39][C][ethernet:071]:   Power Pin: GPIO12 (Mode: OUTPUT)
[15:59:39][C][ethernet:072]:   MDC Pin: 23
[15:59:39][C][ethernet:073]:   MDIO Pin: 18
[15:59:39][C][ethernet:074]:   Type: LAN8720
[15:59:39][C][api:089]: API Server:
[15:59:39][C][api:090]:   Address: puerta.local:6053
[15:59:39][C][ota:129]: Over-The-Air Updates:
[15:59:39][C][ota:130]:   Address: puerta.local:3232

[15:59:39][I][application:113]: Running through first loop()
[15:59:39][I][application:140]: First loop finished successfully!
```
With latest bleeding edge version. Thank you !

closed 2019-03-17 by OttoWinter

sobiso, 2019-05-28:
Hi,
I'm using LAN8720 module. I was trying to connect like they showing on https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
but it does not work.
So i trying to use connection like Olimex ESP32-PoE Board (GPIO17_OUT), but still without any success.
Got this: 
`[09:03:50][VV][preferences:036]: LOAD 0: valid=YES, 0=0x00000001 1=0x42F4583F (Type=233825507, CRC=0x42F4583F)
[09:03:50][C][ota:484]: There have been 1 suspected unsuccessful boot attempts.
[09:03:50][VV][preferences:049]: SAVE 0: 0=0x00000002 1=0x13D89D5A (Type=233825507, CRC=0x13D89D5A)
[09:03:50][I][application:055]: Running through setup()...
[09:03:50][V][application:056]: Sorting components by setup priority...
[09:03:50][C][ethernet:026]: Setting up Ethernet...
[09:04:05][W][ethernet:042]: Connecting via ethernet failed! Re-connecting...
[09:04:06]E (15996) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
[09:04:07]E (16996) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff`

OttoWinter, 2019-05-28:
@sobiso That message is from esp-idf - so either there is a bug in the ESP SDK by espressif or you have wired it up wrong. I don't own that module so I can't tell - for me at least some other olimex Ethernet boards have worked fine.