esphome/issues #54: LEDC set_level not working

legomind, 2019-02-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

LEDC (https://esphome.io/components/output/ledc.html)

**Description of problem:**
The output component works just fine when the brightness is set in home assistant.
But when I use set_level as described here: (https://esphome.io/components/output/index.html#output-set-level-action), the brightness of the light does not change in home assistant.

I have tried both lamba and pure yaml methods, neither work.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: dht
    pin: 22
    model: DHT22
    update_interval: 10s
    temperature:
      name: 'Temperature'
      id: habitat_temp
      filters:
        - lambda: return x * (9.0/5.0) + 32.0;
      unit_of_measurement: '°F'
      on_value:
        - then:
          - lambda: !lambda |-
              ESP_LOGD("main", "%f => Recommended dim level %f", x, (x >= 100) ? 60.0 : 100.0);
              id(habitat_heatlamp).set_level((x >= 100) ? 0.5 : 1);
    humidity:
      name: 'Humidity'
      id: habitat_humid

output:
  - platform: ledc
    pin: 23
    frequency: 10khz
    id: habitat_heatlamp

light:
  - platform: monochromatic
    name: 'Heatlamp'
    output: habitat_heatlamp
    default_transition_length: 0s
    gamma_correct: 0
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-02-15:
I would give the light an ID and set the level on that, rather then try to bypass the light and set the level on the output directly. I am not super familiar with that code, but I suspect that the light is overriding the level on that output, since it expects to have exclusive control over it.

legomind, 2019-02-16:
@brandond Thanks. I should have thought of that. Using the documentation here: (https://esphome.io/components/light/) worked great.

Perhaps a note on the output component docs would be a good idea.

closed 2019-02-16 by legomind