esphome/issues #58: Code compiles but remains in bootloop on device

mvturnho, 2019-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
pip on windows 10 
<esphome-core> 1.12.0-dev #5d70f27
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 nodemcu
**Affected component:**

No clue

**Description of problem:**
After upgarding to latest esphome and esphome-core using:
```
pip install --upgrade git+https://github.com/esphome/esphome.git

Successfully installed esphome-1.12.0.dev0
```

Then I modified my `dummy.yaml` file. Compile gives no errors.
After upload to the device it remains in a boot loop where it should setup the components.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: dummy
  platform: ESP8266
  board: nodemcuv2
  esphome_core_version:
        branch: dev

wifi:
  ssid: 'XXXXXXXXXX'
  password: 'XXXXXXXXXXXXXX'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

i2c:
  sda: D2
  scl: D4
  scan: True
  
binary_sensor:
  - platform: gpio
    name: "dummy"
    pin:
      number: D6
      inverted: False
      mode: INPUT
    
light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: D3
    num_leds: 16
    rgb_order: GRB
    name: 'LED Strip'

```

**Traceback (if applicable):**

```
[17:34:08][I][logger:122]: Log initialized
[17:34:08][C][ota:480]: There have been 0 suspected unsuccessful boot attempts.
[17:34:08][I][application:055]: Running through setup()...
[17:34:08][C][light.fast_led:040]: Setting up FastLED light...
[17:34:08][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'dummy'...
[17:34:16][D][binary_sensor:037]: 'dummy':
[17:34:16] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:34:16]
[17:34:16]wdt reset
[17:34:16]load 0x4010f000, len 1384, room 16
[17:34:16]tail 8
[17:34:16]chksum 0x2d
[17:34:16]csum 0x2d
[17:34:16]vbb28d4a3
[17:34:16]~ld
[17:34:16][I][logger:122]: Log initialized
[17:34:16][C][ota:480]: There have been 1 suspected unsuccessful boot attempts.
[17:34:16][I][application:055]: Running through setup()...
[17:34:16][C][light.fast_led:040]: Setting up FastLED light...
[17:34:16][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'dummy'...
[17:34:25][D][binary_sensor:037]: 'dummy':
[17:34:25] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:34:25]
[17:34:25]wdt reset
[17:34:25]load 0x4010f000, len 1384, room 16
[17:34:25]tail 8
[17:34:25]chksum 0x2d
[17:34:25]csum 0x2d
[17:34:25]vbb28d4a3
[17:34:25]~ld
[17:34:25][I][logger:122]: Log initialized
[17:34:25][C][ota:480]: There have been 2 suspected unsuccessful boot attempts.
[17:34:25][I][application:055]: Running through setup()...
[17:34:25][C][light.fast_led:040]: Setting up FastLED light...
[17:34:25][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'dummy'...
[17:34:33][D][binary_sensor:037]: 'dummy':
[17:34:33] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:34:33]
[17:34:33]wdt reset
[17:34:33]load 0x4010f000, len 1384, room 16
[17:34:33]tail 8
[17:34:33]chksum 0x2d
[17:34:33]csum 0x2d
[17:34:33]vbb28d4a3
[17:34:33]~ld
[17:34:33][I][logger:122]: Log initialized
[17:34:33][C][ota:480]: There have been 3 suspected unsuccessful boot attempts.
[17:34:33][I][application:055]: Running through setup()...
[17:34:33][C][light.fast_led:040]: Setting up FastLED light...
[17:34:33][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'dummy'...
[17:34:41][D][binary_sensor:037]: 'dummy':
[17:34:41] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:34:41]
[17:34:41]wdt reset
[17:34:41]load 0x4010f000, len 1384, room 16
[17:34:41]tail 8
[17:34:41]chksum 0x2d
[17:34:41]csum 0x2d
[17:34:41]vbb28d4a3
[17:34:41]~ld
[17:34:41][I][logger:122]: Log initialized
[17:34:42][C][ota:480]: There have been 4 suspected unsuccessful boot attempts.
[17:34:42][I][application:055]: Running through setup()...
[17:34:42][C][light.fast_led:040]: Setting up FastLED light...
[17:34:42][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'dummy'...

```

**Additional information and things you've tried:**

uninstalling the esphome and deling all files from the compiled directory.

erase flash with:
```
$ esptool.py erase_flash
esptool.py v2.5.1
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:de:a3:85
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.1s
Hard resetting via RTS pin...
```

mvturnho, 2019-02-19:
I checked my install and after the upgrade to the latest dev version of esphome I have both versions installed esphomelib and esphome.

What would be the proper install method for the esphome dev version?

I am working on the mpr121 sensor and want it to work with the latest dev version. I am now unable to test my code since no sensor works from my current dev install. Also tried the bme280. It gives the same problem. 

After the sensor works in esphome-core I want to start with the python esphome component for the mpr121.

mvturnho, 2019-02-19:
@OttoWinter I would like to continue implementing the mpr121 device,
but since the latest dev release I am unable to run my code succesfully on my esp8266 nodemcu, I tried several different devices, but with the same result. Even with no components added I do not get through the setup when the device boots.

Should I develop using the master or still use the dev branch?

For this issue I did a complete reinstall using:

```
pip install git+https://github.com/esphome/esphome
..
Installing collected packages: esphome
  Running setup.py install for esphome: started
    Running setup.py install for esphome: finished with status 'done'
Successfully installed esphome-1.12.0.dev0
```
then run:
`esphome dummy.yaml wizard`

resuling in my `dummy.yaml`

```yaml
esphome:
  name: dummy
  platform: ESP8266
  board: esp12e

wifi:
  ssid: 'KELPNET_IOT'
  password: 'GROENEPAARDENSCHOENEN'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

then run:

`esphome dummy.yaml compile`

where the output seems succesfull.

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d dummy
Processing dummy (platform: espressif8266@1.8.0; framework: arduino; board: esp12e)

LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing esphome-core
git version 2.19.1.windows.1
Cloning into 'C:\Users\michi\Documents\esphomelib\dummy\.piolibdeps\_tmp_installing-65hg_2-package'...
remote: Enumerating objects: 378, done.
remote: Counting objects: 100% (378/378), done.
remote: Compressing objects: 100% (374/374), done.
remote: Total 378 (delta 47), reused 30 (delta 1), pack-reused 0
Receiving objects: 100% (378/378), 412.17 KiB | 1.25 MiB/s, done.
Resolving deltas: 100% (47/47), done.
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
[... 325 lines cut ...]
[17:42:51] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:42:51]
[17:42:51]wdt reset
[17:42:51]load 0x4010f000, len 1384, room 16
[17:42:51]tail 8
[17:42:51]chksum 0x2d
[17:42:51]csum 0x2d
[17:42:51]vbb28d4a3
[17:42:51]~ld
[17:42:51][I][logger:122]: Log initialized
[17:42:51][C][ota:480]: There have been 1 suspected unsuccessful boot attempts.
[17:42:51][I][application:055]: Running through setup()...
[17:42:51][C][wifi:029]: Setting up WiFi...
[17:43:00]
[17:43:00] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:43:00]
[17:43:00]wdt reset
[17:43:00]load 0x4010f000, len 1384, room 16
[17:43:00]tail 8
[17:43:00]chksum 0x2d
[17:43:00]csum 0x2d
[17:43:00]vbb28d4a3
[17:43:00]~ld

```

mvturnho, 2019-02-19:
flasing my old compiled stuff works fine as long as I do not use the compile option.

OttoWinter, 2019-02-19:
I've seen this happen before, but never figured out why. Especially lines like this have baffled me last time too:
```
[17:34:25][D][binary_sensor:037]: 'dummy':
[17:34:25] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
```

the log line *should* ready `'dummy': Sending state OFF` - but for reasons that are beyond me the last part of that is somehow cut off right in the middle of the message - and a WDT reset occurs after a while. Now I think this is just a symptom after an issue somewhere else (not logger)

I have a couple of theories what could be causing this I think:

 - Last time I saw this, it disappeared when MQTT was disabled (but from your dummy.yaml it appears as though that's not the case here).
 - another option is that the WDT reset is somewhere in the ESP SDK and with the new wifi scan introduced in 1.10 - What happens when you enable `fast_connect` mode in wifi? (that makes it use the old non-scan wifi connect)
 - Maybe the board also has some effect - maybe try with `nodemcuv2` or even `esp01_1m`.

mvturnho, 2019-02-19:
@OttoWinter I think I found it. 
Since the esphome works for you and others I thought it could not be that, so I figured it could be the tools or libraries from platformio.

I updated my platformio and found that some platforms and libraries were detached.
So I uninstalled those libs and my espresif platforms (esp32 and esp8266)

```
platformio lib -g uninstall FastLED
platformio lib -g uninstall "ESP Async WebServer"
platformio platform uninstall espressif8266
```

```
$ platformio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.181225    [Up-to-date]
Updating contrib-piohome                 @ 2.0.0          [Up-to-date]
Updating tool-pioplus                    @ 2.0.2          [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.8.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20402.4      [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.190218 [Up-to-date]

Library Manager
===============
Updating Adafruit MCP23017 Arduino Library @ 1.0.3          [Up-to-date]
Updating Adafruit MPR121                 @ 1.0.2          [Up-to-date]
Updating Adafruit PWM Servo Driver Library @ 1.0.2          [Up-to-date]
Updating ArduinoJson-esphomelib          @ 5.13.3         [Up-to-date]
Updating ArduinoJson                     @ 5.13.4         [Up-to-date]
Updating AsyncMqttClient                 @ 0.8.2          [Up-to-date]
Updating AsyncTCP                        @ 1.0.3          [Up-to-date]
Updating BH1750                          @ a49b08550a     [Up-to-date]
Updating BME280                          @ 3.0.0          [Up-to-date]
Updating ESP Async WebServer             @ 1.2.0          [Up-to-date]
Updating ESP Async WebServer             @ 1.1.1          [Detached]
Updating ESPAsyncTCP                     @ 1.2.0          [Up-to-date]
Updating FastLED                         @ 3.2.6          [Up-to-date]
Updating FastLED                         @ 3.2.0          [Detached]
Updating Grove 4-Digit Display           @ 1.0.0          [Up-to-date]
Updating I2Cdevlib-esphomelib            @ 1.1.0          [Up-to-date]
Updating M5Stack                         @ 0.2.5          [Up-to-date]
Updating M5ez                            @ 2.0.1          [Up-to-date]
Updating MPR121Button                    @ 1.0.0          [Up-to-date]
Updating MQTT                            @ 2.4.3          [Up-to-date]
Updating NeoPixelBus                     @ 2.4.2          [Up-to-date]
Updating PCA9685                         @ 26adc95c5b     [Up-to-date]
Updating RadioHead                       @ 1.85           [Up-to-date]
Updating esphomelib                      @ 1.10.1         [Up-to-date]
Updating ezTime                          @ 0.7.9          [Up-to-date]
Updating r89m Buttons                    @ 2.0.1          [Up-to-date]

platformio platform uninstall espressif8266
Uninstalling espressif8266 @ 1.8.0:     [OK]
Uninstalling framework-arduinoespressif8266 @ 2.20402.4:        [OK]
Uninstalling tool-espotapy @ 1.1.3:     [OK]
Uninstalling tool-esptool @ 1.413.0:    [OK]
Uninstalling tool-mkspiffs @ 1.200.0:   [OK]
Uninstalling toolchain-xtensa @ 1.40802.190218:         [OK]
The platform 'espressif8266' has been successfully uninstalled!

```
then when compiling my dummy project all platforms and libs were doenloaded and reinstaled. After flashing to the esp8266 It seemed to boot.

brandond, 2019-02-19:
I have had an issue once or twice where I (in desperation) deleted ~/.platformio in addition to the stuff removed by esphome's `clean` command. It did seem to fix the issue, but I certainly can't explain why. Something to keep in the toolbox, I guess.

mvturnho, 2019-02-20:
Thanks for your help. I close the issue.

closed 2019-02-20 by mvturnho