esphome/issues #59: GPIO Pin 0 does not support pullup pin mode (i2c)

rradar, 2019-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8285

**Affected component:**

i2c

**Description of problem:**

I tried to compile a old sketch with the new esphome version 1.10.1 and get the following error message:

> Failed config
> 
> i2c: [source espm3-i2c.yaml:19]
>   
>   GPIO Pin 0 does not support pullup pin mode. Please choose another pin. Got '0'
>   sda: 0
>   scl: 2
>   scan: False

I need to use these pins for i2c communication because this board (esp m3) offers no others.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: ...
  platform: ESP8266
  board: esp8285

[...]

i2c:
  sda: 0
  scl: 2
  scan: False
```

**Additional information and things you've tried:**

It always worked with older versions of esphome...

OttoWinter, 2019-02-20:
Technically the error message is correct, pin 0 does not have pullups.

I guess I will re-allow non-pullup pins (but the user then needs to apply external pullups on the bus)

rradar, 2019-02-21:
Hey @OttoWinter 

thank's for your feedback. I actually don't understand what the missing pullup does (or doesn't)? What would be needed (hardware wise) and why it wasn't needed with the old esphome versions? :confused: :question: 

I never had problems using the esp-m3 (esp8285) with i2c devices such as bme280, oled display, bh1750 and others on the only available i2c pins GPIO0 and GPIO2.

I'm converting all my devices from mqtt over to the new api and running in this issue is sadly a show stopper as I just can't compile it with the new version. :crying_cat_face: Is there any option available to force compiling? Because I know it works perfect and reliable! :smiley_cat:

closed 2019-02-21 by OttoWinter

rradar, 2019-02-23:
For me and myself: https://medium.freecodecamp.org/a-simple-explanation-of-pull-down-and-pull-up-resistors-660b308f116a