esphome/issues #60: Can't work offline / How to (long-term) cache dependencies?

rradar, 2019-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266

**Affected component:**

no idea

**Description of problem:**

I have serious troubles getting my >true< offline environment working with esphome.

Eventhoug I'm building a sketch while I'm >somewhere< online - I only can build the same or even stripped down (nothing new in the yaml) sketch for a short period.

Sometimes it even doesn't work at all.

Is there any way to force caching dependencies (and download all at once?) and also force esphome to use this ones instead trying an non existing online connection?

Typical error message is the following one (compiled this sketch in the last week successfully with internet - now the exactly same one doesn't compile offline?)

**Traceback (if applicable):**

```

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d d1old
Processing d1old (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
LibraryManager: Installing v1.10.1
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 107, in cli
    result = (envname, ep.process())
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 207, in process
    result = self._run()
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 302, in _run
    self.verbose)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 325, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/commands/lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/managers/lib.py", line 352, in install
    force=force)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 721, in install
    name, url, requirements, track=True)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 508, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/home/userwithoutname/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 199, in download
    raise e
ConnectionError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /OttoWinter/esphomelib/archive/v1.10.1.zip (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f50180e4c50>: Failed to establish a new connection: [Errno -2] Name or service not known',))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

**Additional information and things you've tried:**

OttoWinter, 2019-02-20:
That is a fundamental problem in how platformio works - I don't know of a way that these dependencies can be cached.

I don't think esphome can solve this, maybe try `pio lib -g install` for all libraries listed in `lib_deps`? Idk

rradar, 2019-02-23:
Thank's for your feedback @OttoWinter 

I tried `pio lib -g install` but it just outputs the library storage directory.

Would it make sense to open a issue over at platformio? Maybe they have an idea?

DeeBeeKay, 2019-03-23:
Hi, 

I am very much a noob trying some stuff out. I've had ConnectionError: problems with platformio for the last 3 days (max retries exceeded). Correct me if I'm wrong (please), but it seems like esphome is using a particular API key or something, which would imply that all esphome users are using the same API key, which is causing platformio to say "Nay."

I am not familiar with platformio.org, but it appears that the service supports user accounts. Would not a potential solution be to allow individual esphome users to enter their own platformio account details? That way, the connection count per user would drop.

Let me know if I'm barking up the wrong tree, but at this point, I haven't even been able to compile once yet, which is super frustrating.

OttoWinter, 2019-03-23:
@DeeBeeKay No, esphome does not use any special api key anything like that. The issue you're seeing is likely a broken DNS config (see `/etc/resolv.conf`; broken in some docker versions).

DeeBeeKay, 2019-03-23:
Thank you so much for this very helpful comment, @OttoWinter ! I edited the /etc/resolv.conf and pointed it at my pihole instead of the 127.0.0.11 it was pointed at. Compiling like a boss now. Finally I get to play!

DeeBeeKay, 2019-03-25:
@OttoWinter I'm posting an update simply in case someone else has this issue, and also, I can't figure out what happened.

So, my problem was, I couldn't compile, because of MaxConnection issues. You suggested that the problem lay with my /etc/resolv.conf file. I edited it manually, and found that it was pointing to 127.0.0.11 for name resolution. I changed this to [my pihole address], and rebooted the host my hassio instance is on.

This appeared to fix the problem - and, to be fair, the problem is resolved: I can compile just fine. But, I just noticed that the /etc/resolv.conf file is still open in Configurator (the hassio plugin I used for editing), and it's reverted to 127.0.0.11. 

But everything still works.

... which ... it ... shouldn't?

I dunno. I'm happy that I can tinker on with my project. I don't expect a response to this post, I'm just putting it here as feedback, and because, well, it's a mystery. And mysteries bother me.

brandond, 2019-03-25:
@DeeBeeKay that's a docker thing. Docker runs an internal resolver to look up service hostnames. It in turn uses the system resolver, or whatever alternative DNS server you've configured it to use. I'm not sure what platform you're running hassio on, but I've seen a lot of issues with docker + systemd-resolve. See this issue for example: https://github.com/docker/libnetwork/issues/2187

OttoWinter, 2019-03-27:
I'm going to close this issue because it's an issue with the docker install and not esphome directly.

closed 2019-03-27 by OttoWinter