esphome/issues #61: NodeMcu V3 and SRX882 causes "Exception type: unknown"

hubertbanas, 2019-02-18:
NodeMcu V3 + SRX882

**Affected component:**
https://esphome.io/components/remote_receiver

**Description of problem:**
Exception type: unknown

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: nodemcu1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: !secret mqtt_broker

# Enable logging
logger:

# Enable Home Assistant API
#api:

ota:

remote_receiver:
  pin:
    number: D1
  dump: rc_switch
```

**Traceback (if applicable):**

```
[15:17:04][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[15:17:04][I][application:053]: Running through setup()...
[15:17:04][C][remote.receiver:295]: Setting up Remote Receiver...
[15:17:04][C][wifi:029]: Setting up WiFi...
[15:17:04]
[15:17:04]Exception (0):
WARNING Exception type: unknown
[15:17:04]epc1=0x4020a208 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d nodemcu1 -t idedata
WARNING Decoded 0x4020a208: interruptFunctional(void*)
[15:17:05]
[15:17:05]ctx: cont 
[15:17:05]sp: 3ffffae0 end: 3fffffd0 offset: 01a0
[15:17:05]
[15:17:05]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:17:05]3ffffc80:  feefeffe 3ffffd42 3ffffcb0 3ffffcf0  
[15:17:05]3ffffc90:  40216a44 3ffffd90 3ffffd31 00000020  
WARNING Decoded 0x40216a44: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c:180
[15:17:05]3ffffca0:  40106975 00000005 00000000 4010695d  
WARNING Decoded 0x40106975: interrupt_handler
WARNING Decoded 0x4010695d: interrupt_handler
[15:17:05]3ffffcb0:  c0037005 00000002 0000000a 3ffeb16a  
[15:17:05]3ffffcc0:  3ffffd33 3ffea508 3ffffd61 00000022  
[15:17:05]3ffffcd0:  401068de 401068a4 3fffc258 4000050c  
WARNING Decoded 0x401068de: interrupt_handler
WARNING Decoded 0x401068a4: interrupt_handler
[15:17:05]3ffffce0:  c0037005 00000030 0000001f ffffffff  
[15:17:05]3ffffcf0:  40201f5f 00000000 00000030 00000022  
WARNING Decoded 0x40201f5f: uart_write
[... 114 lines cut ...]
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableReceive(0);  // Receiver on interrupt 0 => that is pin #2
}

void loop() {
  if (mySwitch.available()) {
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}
```

The ReceiveDemo_Advanced produces the dump as well:

```
Decimal: 6717205 (24Bit) Binary: 011001100111111100010101 Tri-State: not applicable PulseLength: 575 microseconds Protocol: 6
Raw data: 13245,472,409,915,870,454,871,451,423,903,430,897,886,439,884,438,437,891,441,888,892,432,890,434,888,436,888,436,886,437,886,437,885,440,434,894,440,887,445,885,896,428,447,883,896,428,446,883,898,
```

Please advise.

OttoWinter, 2019-02-18:
Does this happen when the RF receiver is not connected?

hubertbanas, 2019-02-18:
It only happens when RF receiver is connected.

OttoWinter, 2019-02-18:
Ok so something is either wrong with the gcc compiler, the linker or the arduino project.

epc1 points to interruptFunctional which is defined in the Arduino core. However, the address is 0x4020a208 (not 0x401... as it should be), which puts it in flash and not iram. So when the method is called from an ISR it will fail because of the location in memory (flash cannot be accessed from ISR).

Still weird I didn't encounter this before (tested it and worked fine). I guess I'll just use `__attachInterruptArg` directly

hubertbanas, 2019-02-18:
Thanks for looking into it. I will be happy to provide any debug if needed.

closed 2019-02-19 by OttoWinter