esphome/issues #62: Waveshare e-paper display causes crashes

pauln, 2019-02-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
- Home Assistant 0.87.0 (Docker)
- esphome 1.12.0-dev (also 1.11.0b1 and 1.10.1)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (`esp12e` board)
Waveshare e-paper driver board

**Affected component:**

[Waveshare e-paper display](https://esphome.io/components/display/waveshare_epaper.html)

**Description of problem:**
I'm using a 7.5" monochrome Waveshare e-paper display (though the same happens with the 2-colour versions).  Upon boot, the initial display is drawn successfully - and usually one subsequent draw will succeed (after pulling data from Home Assistant via the API, if I'm lucky).  The next refresh results in a `Soft WDT reset`, which results in the initial state (i.e. without data from Home Assistant) being redrawn, starting the cycle over again.

In order to isolate the issue, I've disabled not only the Home Assistant sensors, but the API and even wifi; even drawing a small amount of static content to the display triggers this behaviour.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: einkdisplay
  platform: ESP8266
  board: esp12e
  esphome_core_version: dev

logger:

# Configure SPI pins
spi:
  clk_pin: 14 #D5
  mosi_pin: 13 #D7

# Fonts
font:
  - file: "Roboto-Regular.ttf"
    id: roboto_20
    size: 20

# Configure e-ink display
display:
  - platform: waveshare_epaper
    cs_pin: 15 #D8
    dc_pin: 4 #D2
    busy_pin: 16 #D0
    reset_pin: 5 #D1
    model: 7.50in
    update_interval: 1min
    lambda: |-
      it.rectangle(44, 102, 215, 94);
      it.print(52, 110, id(roboto_20), TextAlign::TOP_LEFT, "Bedroom");
      it.rectangle(258, 102, 340, 94);
      it.print(265, 110, id(roboto_20), TextAlign::TOP_LEFT, "Weather");
      it.rectangle(44, 195, 554, 94);
      it.print(52, 203, id(roboto_20), TextAlign::TOP_LEFT, "3D Printer");
```

**Traceback (if applicable):**

```
[20:33:57][I][logger:122]: Log initialized
[20:33:57][I][application:055]: Running through setup()...
[20:33:57][C][spi:093]: Setting up SPI bus...
[20:33:57][I][application:090]: setup() finished successfully!
[20:33:57][I][application:098]: esphome-core version 1.12.0-dev compiled on Feb 19 2019, 20:31:49
[20:33:57][C][spi:105]: SPI bus:
[20:33:57][C][spi:106]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[20:33:57][C][spi:108]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[20:33:57][C][logger:159]: Logger:
[20:33:57][C][logger:160]:   Level: DEBUG
[20:33:57][C][logger:161]:   Log Baud Rate: 115200
[20:33:57][C][logger:162]:   Hardware UART: UART0
[20:33:57][C][display.waveshare_epaper:677]: Waveshare E-Paper
[20:33:57][C][display.waveshare_epaper:677]:   Rotations: 0 °
[20:33:57][C][display.waveshare_epaper:677]:   Dimensions: 640px x 384px
[20:33:57][C][display.waveshare_epaper:678]:   Model: 7.5in
[20:33:57][C][display.waveshare_epaper:679]:   Reset Pin: GPIO5 (Mode: OUTPUT)
[20:33:57][C][display.waveshare_epaper:680]:   DC Pin: GPIO4 (Mode: OUTPUT)
[20:33:57][C][display.waveshare_epaper:681]:   Busy Pin: GPIO16 (Mode: INPUT)
[20:33:57][C][display.waveshare_epaper:682]:   Update Interval: 60000 ms
[20:33:57][I][application:113]: Running through first loop()
[20:33:59][I][application:140]: First loop finished successfully!
[20:35:32]
[20:35:32]Soft WDT reset
[20:35:32]
[20:35:32]ctx: cont 
[20:35:32]sp: 3ffffcb0 end: 3fffffd0 offset: 01b0
[20:35:32]
[20:35:32]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[20:35:32]3ffffe60:  00000034 3ffef974 3ffeecd8 00000121  
[20:35:32]3ffffe70:  0000000a 3fff04f0 00000008 00000001  
[20:35:32]3ffffe80:  0000008d 00000007 00000013 00000012  
[20:35:32]3ffffe90:  00000000 4bc6a7f0 0001772c 00000000  
[20:35:32]3ffffea0:  00000000 00000000 4bc6a7f0 3fff0434  
[20:35:32]3ffffeb0:  3fff0494 00000001 3fff04e4 40206aa8  
WARNING Decoded 0x40206aa8: esphome::SPIDevice::enable()
[20:35:32]3ffffec0:  00000001 00000033 3fff04e4 40206ae4  
WARNING Decoded 0x40206ae4: esphome::SPIDevice::write_byte(unsigned char)
[20:35:32]3ffffed0:  00000001 00000033 3fff04e4 40205781  
WARNING Decoded 0x40205781: esphome::display::WaveshareEPaper::data(unsigned char)
[20:35:32]3ffffee0:  00000000 00006d68 3fff04c4 40205ea7  
WARNING Decoded 0x40205ea7: esphome::display::WaveshareEPaper7P5In::display()
[20:35:32]3ffffef0:  000000c0 00000003 00016eea 4020527e  
WARNING Decoded 0x4020527e: esphome::display::DisplayBuffer::do_update()
[20:35:32]3fffff00:  3fff04f0 00000000 4bc6a7f0 3fff04c4  
[20:35:32]3fffff10:  00000000 00016eea 3fff04c4 402052f0  
WARNING Decoded 0x402052f0: esphome::display::WaveshareEPaper::update()
[20:35:32]3fffff20:  00000000 4bc6a7f0 00000000 4020777c  
WARNING Decoded 0x4020777c: std::_Function_handler<void (), esphome::PollingComponent::setup_()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at component.cpp
[20:35:32]3fffff30:  3fff052c 3fff7db0 00000000 402049b6  
WARNING Decoded 0x402049b6: esphome::Component::loop_internal()
[20:35:32]3fffff40:  3ffee660 3fff7d8c 401068e6 570a3d70  
WARNING Decoded 0x401068e6: millis
[20:35:32]3fffff50:  00000000 3fff04c4 4020721c 3fff04c4  
WARNING Decoded 0x4020721c: esp_yield
[20:35:32]3fffff60:  3fff052c 00000002 3fff04c4 40204aec  
WARNING Decoded 0x40204aec: esphome::Component::loop_()
[20:35:32]3fffff70:  3fff0530 00000002 3ffeec38 402034c1  
WARNING Decoded 0x402034c1: esphome::Application::loop()
[20:35:32]3fffff80:  00000000 3fff0530 feefef00 feefeffe  
[20:35:32]3fffff90:  00000012 00000003 00000708 3ffeed98  
[20:35:32]3fffffa0:  3fffdad0 00000000 3ffeed90 40203224  
WARNING Decoded 0x40203224: loop
[20:35:32]3fffffb0:  3fffdad0 00000000 3ffeed90 402072a8  
WARNING Decoded 0x402072a8: loop_wrapper() at core_esp8266_main.cpp
[20:35:32]3fffffc0:  feefeffe feefeffe 3ffe8524 40100721  
WARNING Decoded 0x40100721: cont_wrapper
[20:35:32]<<<stack<<<
[20:35:33]
[20:35:33] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:35:33]
[20:35:33]load 0x4010f000, len 1384, room 16 
[20:35:33]tail 8
[20:35:33]chksum 0x2d
[20:35:33]csum 0x2d
[20:35:33]vbb28d4a3
[20:35:33]~ld
```

**Additional information and things you've tried:**
I've tried adjusting the refresh time, and confirmed that it's definitely related to the display being drawn and not something else coincidentally happening when the display is due to be updated.  I've also tried removing the configuration for the reset and busy pins, since they're listed as optional - just on the off chance that the Waveshare driver board does something odd with one or both of those pins which somehow triggers this issue.  No change - the WDT reset still happens.

I've also removed all display-related config while retaining all wifi/API/Home Assistant sensor config, and everything remains stable (no WDT resets).

OttoWinter, 2019-02-20:
Huh, that is weird. ESPHome does feed the soft WDT in the waveshare epaper implementation, maybe there is an infinite loop somewhere.

could you try two things for me:

- Enable VERY_VERBOSE logs - this will help identify what component is causing the issue and how long the WDT is not fed
- Replace [this line](https://github.com/esphome/esphome-core/blob/dev/src/esphome/helpers.cpp#L386) with `yield();` in `<NODE_NAME>/.piolibdeps/<some_path>/helpers.cpp` ?

pauln, 2019-02-21:
Enabling `VERY_VERBOSE` logs resulted in a non-stop stream of logs from the SPI component (`Enabling SPI Chip` / `Wrote 0b[byte]` / `Disabling SPI Chip`), which slowed the process down enough (due to the sheer volume of data that's sent when updating the 7.5" screen) that it still hadn't managed to push a single update after more than a minute.

I disabled those specific log entries, and it was able to perform display updates - but still suffered from the WDT reset issue, and didn't give any helpful logs despite the `VERY_VERBOSE` setting.

Having seen that every byte of data being sent was going through the enable / write / disable cycle, I then modified the `WaveshareEPaper` class to add `start_data()` and `stop_data()` functions (based on what the `data()` function does) and updated `WaveshareEPaper7P5In::display()` use those in conjunction with `write_byte()` rather than calling `data()` for each individual byte in the display buffer.  This seems to have resolved the issue; after running through several display cycles using the stripped back static display config, I swapped back to my full config with Wifi and Home Assistant sensors - and that's now run through more than a dozen display cycles (including receiving updates from Home Assistant via the API) without any issues.  I'll submit a pull request with my changes; I've only updated the 7.5" model's `display()` to use the new start/stop functions for now, as its display method is more involved than other models (it writes 8 bytes per display buffer byte, whereas all others only write a single byte), but I suppose other models may also benefit from removing the repeated enable/disable cycles.  I don't have any other models/sizes to test against, though, so I'd rather leave that to somebody who does.

brandond, 2019-02-21:
I suspect some of the other displays could use the same thing; the ssd1306 display updates glacially slowly on esphomelib as well, while it can update without even stuttering on Arduino.

OttoWinter, 2019-02-21:
@pauln Yeah the enable/disable stuff is not that efficient, but when I looked at the datasheet it specifically said that each data byte would need to be transferred with a dedicated enable/disable cycle. I never tested it so it may also work without it - it would need to be tested at least with a couple more displays, I have a 2.9inch one I can test with later.

But the other thing about that is that `enable/disable` and even `data` should not slow things down that much (when logging is not set to very verbose).

`:data` is quite lightweight, it only sets a GPIO pin, and calls `enable` / `disable`

https://github.com/esphome/esphome-core/blob/67fe25b5b1cd5bad9d134abc8e9bb4bd4d605fd2/src/esphome/display/waveshare_epaper.cpp#L81-L86

enable/disable are also quite lightweight (when logging is not high; with normal log level the statement does not appear in firmware)

https://github.com/esphome/esphome-core/blob/67fe25b5b1cd5bad9d134abc8e9bb4bd4d605fd2/src/esphome/spi_component.cpp#L78-L91

The `write_byte` method surely does much more work...

https://github.com/esphome/esphome-core/blob/67fe25b5b1cd5bad9d134abc8e9bb4bd4d605fd2/src/esphome/spi_component.cpp#L18-L65

The other thing is that the WDT should not trigger at all, the soft WDT (that is triggering here) is regularly fed:

https://github.com/esphome/esphome-core/blob/67fe25b5b1cd5bad9d134abc8e9bb4bd4d605fd2/src/esphome/display/waveshare_epaper.cpp#L664

(Sidenote: One way to make this class more efficient would be to make `SPIComponent` use the internal SPI interface on the ESP if possible, kind of like the UARTComponent does it)

@brandond Do you mean the i2c version of the SSD1306? If so, try upping the i2c frequency (the default is very low because higher ones don't work with some sensors). `700KHz` is the highest for ESP8266 (and other options are 600KHz, 500KHz, 400KHz, 300KHz, 200KHz, 100KHz, 50KHz)

pauln, 2019-02-22:
> when I looked at the datasheet it specifically said that each data byte would need to be transferred with a dedicated enable/disable cycle

Interesting - I can confirm that at least the 7.5" screen (which is all that I have, and all that my PR affects) is definitely happy with all data bytes being sent within a single enable/disable cycle.  Looking at section 7-3-1 of the [7.5" display's spec](https://www.waveshare.com/w/upload/b/b6/7.5inch-e-paper-specification.pdf), the diagrams do seem to depict disabling between bytes - but the text doesn't seem to mention it, and I think that the diagrams are showing that you should disable when not sending, rather than disable between bytes.

> But the other thing about that is that `enable/disable` and even `data` should not slow things down that much

I don't think that it's necessarily the enable/disable cycle (or the `data` function) that's to blame.  The 7.5" screen [does some additional processing](https://github.com/esphome/esphome-core/blob/67fe25b5b1cd5bad9d134abc8e9bb4bd4d605fd2/src/esphome/display/waveshare_epaper.cpp#L644-L665) and ends up sending multiple bytes for every byte in the display buffer - but only feeds the WDT in the outer loop.  The extra overhead of `data` vs `write_byte` seems to be just enough to (sometimes) trigger the WDT.  I did also try feeding the WDT in the inner loop instead of the outer loop, but that resulted in a WDT reset happening as soon as the first draw is triggered (rather than the 2nd or 3rd); I'm not sure why, but with my changes (`[start/stop]_data`, `write_byte`) in place, I can also move the `feed_wdt` call to the inner loop without any problems.

OttoWinter, 2019-02-22:
I just looked at the datasheet for the displays I have and you appear to be correct, the cs pin can be left enabled during transmissions.

Still though we're talking about a few microseconds here, that does not explain the underlying issue and I would like to fix that instead of fixing the symptoms.

Yes there is an inner loop with 8 executions but surely that can never be enough to trigger a soft wdt reset... I still think something is wrong with the feed_wdt function. Could you try replacing the ESP.feedWDT() call in there with yield() as in my first answer?

I have no problem with the fix you made, as it definitely improves execution speed in a tight loop. But I'd ideally still like to find out what exactly is breaking the wdt, since the feed_wdt() function is called in a couple of places.

pauln, 2019-02-22:
Replacing the `ESP.feedWDT()` call with `yield()` results in the WDT reset, with or without my patch.  I still have the SPI enable/write_byte/disable logging removed, as otherwise they completely drown out any other log messages - but here are the logs with and without my patch:

`yield()` without patch:
```
[00:02:50]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[00:02:50][I][logger:122]: Log initialized
[00:02:50][I][application:055]: Running through setup()...
[00:02:50][V][application:056]: Sorting components by setup priority...
[00:02:50][C][spi:093]: Setting up SPI bus...
[00:02:51][VV][component:059]: set_interval(name='update', interval=15000, offset=6736)
[00:02:51][I][application:090]: setup() finished successfully!
[00:02:51][I][application:098]: esphome-core version 1.12.0-dev compiled on Feb 22 2019, 23:55:31
[00:02:51][C][spi:105]: SPI bus:
[00:02:51][C][spi:106]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[00:02:51][C][spi:108]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[00:02:51][C][logger:159]: Logger:
[00:02:51][C][logger:160]:   Level: VERY_VERBOSE
[00:02:51][C][logger:161]:   Log Baud Rate: 115200
[00:02:51][C][logger:162]:   Hardware UART: UART0
[00:02:51][C][display.waveshare_epaper:686]: Waveshare E-Paper
[00:02:51][C][display.waveshare_epaper:686]:   Rotations: 0 °
[00:02:51][C][display.waveshare_epaper:686]:   Dimensions: 640px x 384px
[00:02:51][C][display.waveshare_epaper:687]:   Model: 7.5in
[00:02:51][C][display.waveshare_epaper:688]:   Reset Pin: GPIO5 (Mode: OUTPUT)
[00:02:51][C][display.waveshare_epaper:689]:   DC Pin: GPIO4 (Mode: OUTPUT)
[00:02:51][C][display.waveshare_epaper:690]:   Busy Pin: GPIO16 (Mode: INPUT)
[00:02:51][C][display.waveshare_epaper:691]:   Update Interval: 15000 ms
[00:02:51][I][application:113]: Running through first loop()
[00:02:51][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=4294946070 (now=607)
[00:02:53][I][application:140]: First loop finished successfully!
[00:02:59][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=4294961070 (now=8781)
[00:03:14][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=8774 (now=23788)
[00:03:16]
[00:03:16]Soft WDT reset
[00:03:16]
[00:03:16]ctx: cont 
[00:03:16]sp: 3ffffc90 end: 3fffffd0 offset: 01b0
[00:03:16]
[00:03:16]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[00:03:16]3ffffe40:  00000034 3ffefd0c 3ffefd0c 00000121  
[00:03:16]3ffffe50:  0000000a 3fff0928 00000008 00000001  
[00:03:16]3ffffe60:  0000008d 00000007 00000013 00000012  
[00:03:16]3ffffe70:  00000000 4bc6a7f0 0000642d 00000000  
[00:03:16]3ffffe80:  00000000 00000000 4bc6a7f0 3fff082c  
[00:03:16]3ffffe90:  3fff08ac 00000001 3fff091c 40206b58  
WARNING Decoded 0x40206b58: esphome::SPIDevice::enable()
[00:03:16]3ffffea0:  00000002 00000033 3fff091c 40206b94  
WARNING Decoded 0x40206b94: esphome::SPIDevice::write_byte(unsigned char)
[00:03:16]3ffffeb0:  00000002 00000033 3fff091c 4020582d  
WARNING Decoded 0x4020582d: esphome::display::WaveshareEPaper::data(unsigned char)
[00:03:16]3ffffec0:  00000000 00005f90 3fff08fc 40205f57  
WARNING Decoded 0x40205f57: esphome::display::WaveshareEPaper7P5In::display()
[00:03:16]3ffffed0:  000000f0 00000003 00005cec 4020532a  
WARNING Decoded 0x4020532a: esphome::display::DisplayBuffer::do_update()
[00:03:16]3ffffee0:  3fff0928 3ffffef0 0000000c 3fff8214  
[00:03:16]3ffffef0:  00005cec 00000000 3fff08fc 4020539c  
WARNING Decoded 0x4020539c: esphome::display::WaveshareEPaper::update()
[00:03:16]3fffff00:  00000086 3ffe893a 3fff08fc 40207c64  
WARNING Decoded 0x40207c64: std::_Function_handler<void (), esphome::PollingComponent::setup_()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at component.cpp
[00:03:16]3fffff10:  3fff8238 00000000 3fff08fc 40204a34  
WARNING Decoded 0x40204a34: esphome::Component::loop_internal()
[00:03:16]3fffff20:  3fff81f8 00000000 00003a98 00002246  
[00:03:16]3fffff30:  00005cec 00000000 4bc6a7f0 00000000  
[00:03:16]3fffff40:  00000000 3fff082c 40106c6a c624dd2f  
WARNING Decoded 0x40106c6a: millis
[00:03:16]3fffff50:  00000000 3fff08fc 00005cec 3fff08fc  
[00:03:16]3fffff60:  3fff0974 00000002 3fff08fc 40204b70  
WARNING Decoded 0x40204b70: esphome::Component::loop_()
[00:03:16]3fffff70:  3fff0978 00000002 3ffeef28 402034c9  
[... 86 lines cut ...]
[00:05:40]3fffff60:  3fff0974 00000002 3fff08fc 40204b70  
WARNING Decoded 0x40204b70: esphome::Component::loop_()
[00:05:40]3fffff70:  3fff0978 00000002 3ffeef28 402034c9  
WARNING Decoded 0x402034c9: esphome::Application::loop()
[00:05:40]3fffff80:  00000000 3fff0978 feefef00 feefeffe  
[00:05:40]3fffff90:  00000012 00000003 00000708 3ffef088  
[00:05:40]3fffffa0:  3fffdad0 00000000 3ffef080 4020322c  
WARNING Decoded 0x4020322c: loop
[00:05:40]3fffffb0:  3fffdad0 00000000 3ffef080 40207684  
WARNING Decoded 0x40207684: loop_wrapper() at core_esp8266_main.cpp
[00:05:40]3fffffc0:  feefeffe feefeffe 3ffe8524 40100a2d  
WARNING Decoded 0x40100a2d: cont_wrapper
[00:05:40]<<<stack<<<
[00:05:40]
[00:05:40] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[00:05:40]
[00:05:40]load 0x4010f000, len 1384, room 16 
[00:05:40]tail 8
[00:05:40]chksum 0x2d
[00:05:40]csum 0x2d
[00:05:40]vbb28d4a3
[00:05:40]~ld
```

I also added some additional logging to waveshare_epaper.cpp, and the WDT resets definitely seem to be happening in the `display()` loop.  I haven't been able to narrow it down any further than that yet, as adding any logging within the loop has the same issue as using VERY_VERBOSE logging without commenting out the SPI logging - it generates so much logging that it slows `display()` down too much to be practical.  I'll try to dig deeper over the weekend to see if the WDT resets still happen when the logging is that verbose, and if so, whether I can isolate the actual cause.

OttoWinter, 2019-02-22:
Oh shit, I know what it is now.

https://github.com/esphome/esphome-core/blob/dev/src/esphome/helpers.cpp#L392

This last_feed check is there so that feeds don't happen too often (on esp32 will slow things down). But... the variable is set outside of the if branch - so no feed ever happens _facepalm_.

Thank you for testing again! I will test with that fixed (and try with a really long loop which should trigger soft wdt). Then I will merge your patch for the data function - I looked at some datasheet and you're correct, the cs pin can be kept high the entire time - that should speed things up a bit.

closed 2019-02-22 by OttoWinter

pauln, 2019-02-23:
Good catch!  That'd certainly do it...

I've bumped my display to use v1.11.0, and it's been perfectly stable so far.  Thanks for taking the time to investigate this issue with me!