esphome/issues #64: ESP32 not correctly setting hostname

ndarilek, 2019-02-19:
Trying to get this working with a generic ESP32 board. My config is as follows:

```
esphomeyaml:
  name: doorbell
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: ICanHasWifi
  password: ...

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

And my connection logs:

```
[22:23:22][0;32m[I][wifi:179]: WiFi Connecting to 'ICanHasWifi'...[0m
[22:23:24][0;32m[I][wifi:349]: WiFi connected![0m
[22:23:24][0;35m[C][wifi:240]:   SSID: 'ICanHasWifi'[0m
[22:23:24][0;35m[C][wifi:241]:   IP Address: 192.168.0.138[0m
[22:23:24][0;35m[C][wifi:243]:   BSSID: B8:27:EB:43:86:7A[0m
[22:23:24][0;35m[C][wifi:245]:   Hostname: 'doorbell'[0m
[22:23:24][0;35m[C][wifi:250]:   Signal strength: -65 dB [0;33m▂▄▆[0;37m█[0m[0m
[22:23:24][0;35m[C][wifi:251]:   Channel: 4[0m
[22:23:24][0;35m[C][wifi:252]:   Subnet: 255.255.255.0[0m
[22:23:24][0;35m[C][wifi:253]:   Gateway: 192.168.0.1[0m
[22:23:24][0;35m[C][wifi:254]:   DNS1: 192.168.0.1[0m
[22:23:24][0;35m[C][wifi:255]:   DNS2: 0.0.0.0[0m
[22:23:24][0;35m[C][api:023]: Setting up Home Assistant API server...[0m
[22:23:24][0;35m[C][ota:129]: Over-The-Air Updates:[0m
[22:23:24][0;35m[C][ota:130]:   Address: 192.168.0.138:3232[0m
[22:23:24][0;32m[I][application:089]: setup() finished successfully![0m
[22:23:24][0;32m[I][application:097]: You're running esphomelib v1.10.1 compiled on Feb 19 2019, 20:33:30[0m
[22:23:24][0;35m[C][wifi:341]: WiFi:[0m
[22:23:24][0;35m[C][wifi:240]:   SSID: 'ICanHasWifi'[0m
[22:23:24][0;35m[C][wifi:241]:   IP Address: 192.168.0.138[0m
[22:23:24][0;35m[C][wifi:243]:   BSSID: B8:27:EB:43:86:7A[0m
[22:23:24][0;35m[C][wifi:245]:   Hostname: 'doorbell'[0m
[22:23:24][0;35m[C][wifi:250]:   Signal strength: -65 dB [0;33m▂▄▆[0;37m█[0m[0m
[22:23:24][0;35m[C][wifi:251]:   Channel: 4[0m
[22:23:24][0;35m[C][wifi:252]:   Subnet: 255.255.255.0[0m
[22:23:24][0;35m[C][wifi:253]:   Gateway: 192.168.0.1[0m
[22:23:24][0;35m[C][wifi:254]:   DNS1: 192.168.0.1[0m
[22:23:24][0;35m[C][wifi:255]:   DNS2: 0.0.0.0[0m
[22:23:24][0;35m[C][logger:099]: Logger:[0m
[22:23:24][0;35m[C][logger:100]:   Level: DEBUG[0m
[22:23:24][0;35m[C][logger:101]:   Log Baud Rate: 115200[0m
[22:23:24][0;35m[C][api:072]: API Server:[0m
[22:23:24][0;35m[C][api:073]:   Address: 192.168.0.138:6053[0m
[22:23:24][0;35m[C][ota:129]: Over-The-Air Updates:[0m
[22:23:24][0;35m[C][ota:130]:   Address: 192.168.0.138:3232[0m
[22:23:24][0;32m[I][application:114]: Running through first loop()[0m
[22:23:24][0;32m[I][application:141]: First loop finished successfully![0m
```

So it seems like the hostname should be set correctly. However, my device isn't reachable at doorbell.local. It is, however, reachable at espressif.local. I've also looked in my router's web interface, and it does indeed show the hostname as espressif.local.

To be clear, I know that I can manually set a hostname and resolve this myself. But I initially thought this was an MDNS issue and was prepared to write it off since I know platforms have some broken MDNS implementations at times. But my research suggests that espressif.local is the hostname an ESP32 gets if one isn't set, so I'm wondering if there is a race condition where the hostname isn't fully set before DHCP launches?

Happy to help debug this further. Thanks for any help.

brandond, 2019-02-20:
I think we're calling it too early (https://github.com/esphome/esphome-core/blob/dev/src/esphome/wifi_component.cpp#L1095), according to https://github.com/espressif/esp-idf/issues/1803 it shouldn't be called until after SYSTEM_EVENT_ETH_START is received.

ndarilek, 2019-02-20:
I'd be happy to try testing this out, though I'm not very familiar with 
the tooling. Is it possible to fork the core and tell my own firmwares 
to temporarily use that fork?

brandond, 2019-02-20:
@ndarilek https://esphome.io/components/esphomeyaml.html#esphomeyaml-esphomelib-version

ndarilek, 2019-02-20:
Hmm, sorry for the noise.

As a reminder, this is on wifi. Is the implication that we need to set 
the hostname after the ethernet interface is started so it correctly 
propagates to the DHCP API? I just started digging into the referenced 
issue.

Thanks.

brandond, 2019-02-20:
Yes, for wired you'll find the hostname call here: https://github.com/esphome/esphome-core/blob/dev/src/esphome/ethernet_component.cpp#L172

OttoWinter, 2019-02-20:
See https://github.com/esphome/esphome-core/pull/399 and try with beta at https://beta.esphome.io (see changelog for installation)

Please reopen if still an issue

closed 2019-02-20 by OttoWinter

ndarilek, 2019-02-21:
Thanks for the quick response! I'm using the HassOS addon. Is it 
necessary to install the beta version of that, or can I set 
esphomelib_version to "dev" in my node's config?

Also, is there a timeframe for this to move to stable? I probably won't 
be actually setting this up for a few weeks--I'm just wondering what the 
release cycle looks like.

Thanks again.

NearlCrews, 2019-03-13:
This is closed but I'm having the exact same issue on the dev version. Here's my configuration:

`esphome:
  name: nodemcu_plantmon
  platform: ESP32
  board: nodemcu-32s
  esphome_core_version: dev
  arduino_version: dev

wifi:
  ssid: 'SSID'
  password: 'password'
  domain: '.domain.local'

api:

# Enable logging
logger:

ota:

esp32_ble_tracker:

sensor:
  - platform: wifi_signal
    name: "NodeMCU PlantMon WiFi"

binary_sensor:
  - platform: status
    name: "NodeMCU PlantMon Status"

time:
  - platform: sntp
    id: sntp_time
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

switch:
  - platform: restart
    name: "NodeMCU PlantMon Restart`

I've tried both dev and stable for the arduino and esphome libraries. Here's what the DHCP request looks like:

`DHCPREQUEST for 10.10.10.10 from 24:0d:c1:1d:86:e0 via mvneta1`

Here's one from an ESP8266 that correctly sets the hostname:

`DHCPREQUEST for 10.10.10.11 from dd:4c:24:10:8c:ec (nodemcu_sump)`

Please let me know if there's any other information you require.

ndarilek, 2019-03-20:
Should we open a new issue, then? Or should we re-open this one?

I have to admit that I didn't test whether this works or not. I'm 
learning lots of new things, so took it as a given that if I was told it 
was fixed then it was. But if it *isn't* then it should be re-opened.

If it isn't re-opened, maybe a new issue should be opened referencing 
this one. For someone starting with electronics, this was a pretty big 
stumbling block for me, and if it isn't working then it probably 
shouldn't be closed.

Thanks.

OttoWinter, 2019-03-20:
@ndarilek See https://github.com/esphome/issues/issues/125#issuecomment-472790737

ndarilek, 2019-03-20:
Cool, thanks for the clarification!