esphome/issues #66: Cannot set expire_after on template binary sensor

brandond, 2019-02-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip + virtualenv

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp01_1m

**Affected component:**
Template binary sensor

**Description of problem:**

Error during compile reporting that expire_after is an invalid option for this platform.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: template
    id: motion
    name: "PIR00 Motion Sensor"
    filters: []
    expire_after:
    device_class: motion
    lambda: "return {};"
```

**Traceback (if applicable):**
```
  [expire_after] is an invalid option for [binary_sensor.template].
```

OttoWinter, 2019-02-20:
binary sensors (and template binary sensors) do not support expire_after - where is that mentioned in the docs?

brandond, 2019-02-20:
For some reason I was under the impression that binary sensors were a subclass of sensor; I realized after opening this that they are not. Do they support expiry on the HA side?

OttoWinter, 2019-02-22:
AFAIK they don't, but I won't support it directly through ESPHome MQTT discovery either - those additional settings should be done directly in HA with the HA config script. (plan is to remove that option from sensors too soon)

closed 2019-02-22 by OttoWinter