esphome/issues #67: DHT sensor's values become Unknown

akasma74, 2019-02-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 (Wemos D1 mini)

**Affected component:**
DHT22 sensor

**Description of problem:**
I have a Wemos D1 mini with a switch and DHT sensor.
Sporadically the sensor data become Unknown in HA and I have to power cycle the Wemos to get them back.
It's not that the whole thing is unavailable as it reacts to the switch, but the sensor's data is not being reported.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
- platform: gpio
  name: "bathroom_extractor_fan"
  pin:
    number: D0 # Only GPIO16 supports pulldown
    mode: INPUT_PULLDOWN_16

sensor:
- platform: dht
  pin: D4
  temperature:
    name: "bathroom_shower_temperature"
  humidity:
    name: "bathroom_shower_humidity"
  update_interval: 30s

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-02-20:
Logs from the thing please. The error messages are useful

akasma74, 2019-02-20:
Later this morning the whole Wemos became unavailable and I had to power cycle it.
By the way, is there a way to reboot a node remotely instead of disconnecting it from power?

The node worked fine for the most of the day, but now shows this:

INFO Reading configuration...
INFO Starting log output from bathroom_extractor_fan_sensor.local using esphomelib API
INFO Connecting to bathroom_extractor_fan_sensor.local:6053 (192.168.0.53)
INFO Successfully connected to bathroom_extractor_fan_sensor.local
[20:40:21][I][application:097]: You're running esphomelib v1.10.1 compiled on Feb 16 2019, 12:44:50
[20:40:21][C][wifi:341]: WiFi:
[20:40:21][C][wifi:240]:   SSID: 'Level2'
[20:40:21][C][wifi:241]:   IP Address: 192.168.0.53
[20:40:21][C][wifi:243]:   BSSID: E8:94:F6:86:4D:96
[20:40:21][C][wifi:245]:   Hostname: 'bathroom_extractor_fan_sensor'
[20:40:21][C][wifi:250]:   Signal strength: -83 dB ▂▄▆█
[20:40:21][C][wifi:251]:   Channel: 6
[20:40:21][C][wifi:252]:   Subnet: 255.255.255.0
[20:40:21][C][wifi:253]:   Gateway: 192.168.0.1
[20:40:21][C][wifi:254]:   DNS1: 194.168.4.100
[20:40:21][C][wifi:255]:   DNS2: 194.168.8.100
[20:40:21][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'bathroom_extractor_fan'
[20:40:21][C][binary_sensor.gpio:024]:   Pin: GPIO16 (Mode: INPUT_PULLDOWN_16)
[20:40:21][C][logger:099]: Logger:
[20:40:21][C][logger:100]:   Level: DEBUG
[20:40:21][C][logger:101]:   Log Baud Rate: 115200
[20:40:21](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[20:40:21][C][sensor.dht:036]:   Pin: GPIO2 (Mode: OUTPUT)
[20:40:21](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[20:40:21][C][sensor.dht:045]:   Update Interval: 30000 ms
[20:40:21][C][sensor.dht:047]:   Temperature 'bathroom_shower_temperature'
[20:40:21][C][sensor.dht:047]:     Unit of Measurement: '°C'
[20:40:21][C][sensor.dht:047]:     Accuracy Decimals: 1
[20:40:21][C][sensor.dht:048]:   Humidity 'bathroom_shower_humidity'
[20:40:21][C][sensor.dht:048]:     Unit of Measurement: '%'
[20:40:21][C][sensor.dht:048]:     Accuracy Decimals: 0
[20:40:21][C][sensor.dht:048]:     Icon: 'mdi:water-percent'
[20:40:21][C][api:072]: API Server:
[20:40:21][C][api:073]:   Address: 192.168.0.53:6053
[20:40:21][C][ota:129]: Over-The-Air Updates:
[20:40:21][C][ota:130]:   Address: 192.168.0.53:8266
[20:40:21][C][ota:132]:   Using Password.
[20:40:28][W][sensor.dht:141]: Requesting data from DHT failed!
[20:40:28][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:40:58][W][sensor.dht:141]: Requesting data from DHT failed!
[20:40:58][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:41:28][W][sensor.dht:141]: Requesting data from DHT failed!
[20:41:28][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:41:58][W][sensor.dht:141]: Requesting data from DHT failed!
[20:41:58][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:42:28][W][sensor.dht:141]: Requesting data from DHT failed!
[20:42:28][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:42:58][W][sensor.dht:141]: Requesting data from DHT failed!
[20:42:58][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

I almost 100% believe that if I just disconnect and reconnect it, everything will come back to normal as it was before. If it's wiring, it shouldn't send temperature/humidity readings at all, but it certainly did that.

akasma74, 2019-02-21:
Ok, I re-soldered all wires and on my workbench everything worked fine. I then moved it to its permanent place, next room, and it started to play up - losing connection, getting invalid response from the sensor etc. I had to disconnect it from power and reconnect again, and it did the trick most of the time, but not always. Surprisingly, pressing Reset on Wemos seemed to help nothing as I got Host unreachable from logger.
In the end it suddenly stopped playing up and I left it with the logger running.
I noticed that it ran well for 4 hours, that's what it was in the beginning:

> INFO Reading configuration...
> INFO Starting log output from 192.168.0.53 using esphomelib API
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> [12:42:52][I][application:097]: You're running esphomelib v1.10.1 compiled on Feb 20 2019, 21:47:25
> [12:42:52][C][wifi:341]: WiFi:
> [12:42:52][C][wifi:240]:   SSID: 'Level2'
> [12:42:52][C][wifi:241]:   IP Address: 192.168.0.53
> [12:42:52][C][wifi:243]:   BSSID: E8:94:F6:86:4D:96
> [12:42:52][C][wifi:245]:   Hostname: 'bathroom_extractor_fan_sensor'
> [12:42:52][C][wifi:250]:   Signal strength: -79 dB ▂▄▆█
> [12:42:52][C][wifi:251]:   Channel: 8
> [12:42:52][C][wifi:252]:   Subnet: 255.255.255.0
> [12:42:52][C][wifi:253]:   Gateway: 192.168.0.1
> [12:42:52][C][wifi:254]:   DNS1: 0.0.0.0
> [12:42:52][C][wifi:255]:   DNS2: 0.0.0.0
> [12:42:52][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'bathroom_extractor_fan'
> [12:42:52][C][binary_sensor.gpio:024]:   Pin: GPIO16 (Mode: INPUT_PULLDOWN_16)
> [12:42:52][C][logger:099]: Logger:
> [12:42:52][C][logger:100]:   Level: DEBUG
> [12:42:52][C][logger:101]:   Log Baud Rate: 115200
> [12:42:52](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
> [12:42:52][C][sensor.dht:036]:   Pin: GPIO2 (Mode: OUTPUT)
> [12:42:52][C][sensor.dht:042]:   Model: DHT22 (or equivalent)
> [12:42:52][C][sensor.dht:045]:   Update Interval: 30000 ms
> [12:42:52][C][sensor.dht:047]:   Temperature 'bathroom_shower_temperature'
> [12:42:52][C][sensor.dht:047]:     Unit of Measurement: '°C'
> [12:42:52][C][sensor.dht:047]:     Accuracy Decimals: 1
> [12:42:52][C][sensor.dht:048]:   Humidity 'bathroom_shower_humidity'
> [12:42:52][C][sensor.dht:048]:     Unit of Measurement: '%'
> [12:42:52][C][sensor.dht:048]:     Accuracy Decimals: 0
> [12:42:52](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
> [12:42:52][C][api:072]: API Server:
> [12:42:52][C][api:073]:   Address: 192.168.0.53:6053
> [12:42:52][C][ota:129]: Over-The-Air Updates:
> [12:42:52][C][ota:130]:   Address: 192.168.0.53:8266
> [12:42:52][C][ota:132]:   Using Password.
> [12:43:12][D][sensor.dht:066]: Got Temperature=20.5°C Humidity=58.3%
> [12:43:42][D][sensor.dht:066]: Got Temperature=20.3°C Humidity=57.9%
> [12:44:12][D][sensor.dht:066]: Got Temperature=20.1°C Humidity=57.4%
> [12:44:25][I][ota:154]: Boot seems successful, resetting boot loop counter.
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> [12:44:42][D][sensor.dht:066]: Got Temperature=19.9°C Humidity=59.1%
> [12:45:12][D][sensor.dht:066]: Got Temperature=19.8°C Humidity=61.4%

Then it was, only once

> [13:37:44][D][api:241]: 'Home Assistant 0.88.0 (192.168.0.34)' disconnected.

and then

> [13:50:50][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!

with plenty of readings in between.

And finally

> [16:44:42][D][sensor.dht:066]: Got Temperature=19.1°C Humidity=63.3%
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
[... 33 lines cut ...]
>     self.on_disconnect()
>   File "/Users/dmitry/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 444, in try_connect
>     cli.connect()
>   File "/Users/dmitry/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 198, in connect
>     self._socket.settimeout(0.1)
> AttributeError: 'NoneType' object has no attribute 'settimeout'

And now the node itself is available, I can see if the button is pressed (no info in logs though), but DHT sensor values stopped coming to HA since the time of those errors.

Apparently there's a link between node's wifi connection and functioning of DHT sensor.

I tried most of the tips to improve wifi stability, and currently my wifi config looks like this:
```
wifi:
  ssid: 'Level2'
  password: '*****'
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.0.53
    gateway: 192.168.0.1
    subnet: 255.255.255.0

```

My wireless router is pretty close to the node as you can see by signal level, and still.. have no idea what it is.. :(

akasma74, 2019-02-22:
Now I'm pretty sure it's a software problem.
I desoldered some extra wires from the sensor, and it all worked fine.
Then I changed some WiFi settings and uploaded OTA, which resulted in invalid readings coming back to me straight away. I didn't change anything apart from the yaml.

Perhaps the node is not resetted properly after upload?
I also noticed that pressing Reset button on it doesn't help either, I have to electrically disconnect the node from power and reconnect it to get a clean reboot.

OttoWinter, 2019-02-22:
1. Do you have a pull up resistor on the pin? (as the error says)
2. Please read the docs about setting a model manually
3. The DHTs have a pretty tight timing, if the wires have too much capacitance or are flaky things will fail.
4. The disconnection errors are unrelated and have nothing to do with this error. There are issues open for those already, please report those there (it gets hard to track them otherwise).
5. "Is there any way of software restarting ESP?". Yes, please see switch section on website.

akasma74, 2019-02-22:
1. I use [this board](https://www.aliexpress.com/item/Free-shipping-DHT22-single-bus-digital-temperature-and-humidity-sensor-module-electronic-building-blocks-AM2302-for/1878263207.html), and I have no extra resistors as it has a resistor and a capacitor. It works fine most of the time with Tasmota and Esphome so I doubt it's because of some missing resistors.
2. I did that. Adding `model: dht22` makes no difference, that's actually what I'm using for the last 2 days.
3. My wires are copper 0.56mm, less than 8 cm each
4. No problem
5. Thanks, still getting to know all ins and outs on the website.

Anyway, did you see my last comment about getting invalid readings straight after OTA uploading?
I think that's the most important indication that it is not hardware-related and has nothing to do with wires/resistors.
At least I just cannot believe that something changes in connectivity between the sensor and the ESP when I just OTA upload a new firmware. It's somewhere in the code.
Perhaps it clears up when I disconnect if for a long period of time (few minutes), but it doesn't work when I press the Reset button? Sounds like mystery to me as I'm not that clued up in the hardware side of things.

akasma74, 2019-02-23:
Some more food for thought.
The sensor worked fine tonight until the node started going offline/online for many times during the early hours.
Since then it showed the last value and invalid readings in the logs.
Just now I power cycled it and.. I got the reading back!

> INFO Reading configuration...
> INFO Starting log output from 192.168.0.53 using esphomelib API
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> [10:37:16][I][application:097]: You're running esphomelib v1.10.1 compiled on Feb 22 2019, 15:28:24
> [10:37:16][C][wifi:341]: WiFi:
> [10:37:16][C][wifi:240]:   SSID: 'Level2'
> [10:37:16][C][wifi:241]:   IP Address: 192.168.0.53
> [10:37:16][C][wifi:243]:   BSSID: E8:94:F6:86:4D:96
> [10:37:16][C][wifi:245]:   Hostname: 'bathroom_extractor_fan_sensor'
> [10:37:16][C][wifi:250]:   Signal strength: -77 dB ▂▄▆█
> [10:37:16][C][wifi:251]:   Channel: 11
> [10:37:16][C][wifi:252]:   Subnet: 255.255.255.0
> [10:37:16][C][wifi:253]:   Gateway: 192.168.0.1
> [10:37:16][C][wifi:254]:   DNS1: 0.0.0.0
> [10:37:16][C][wifi:255]:   DNS2: 0.0.0.0
> [10:37:16][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'bathroom_extractor_fan'
> [10:37:16][C][binary_sensor.gpio:024]:   Pin: GPIO16 (Mode: INPUT_PULLDOWN_16)
> [10:37:16][C][logger:099]: Logger:
> [10:37:16][C][logger:100]:   Level: DEBUG
> [10:37:16][C][logger:101]:   Log Baud Rate: 115200
> [10:37:17][C][sensor.dht:035]: DHT:
> [10:37:17][C][sensor.dht:036]:   Pin: GPIO2 (Mode: OUTPUT)
> [10:37:17][C][sensor.dht:042]:   Model: DHT22 (or equivalent)
> [10:37:17][C][sensor.dht:045]:   Update Interval: 30000 ms
> [10:37:17][C][sensor.dht:047]:   Temperature 'bathroom_shower_temperature'
> [10:37:17][C][sensor.dht:047]:     Unit of Measurement: '°C'
> [10:37:17][C][sensor.dht:047]:     Accuracy Decimals: 1
> [10:37:17][C][sensor.dht:048]:   Humidity 'bathroom_shower_humidity'
> [10:37:17][C][sensor.dht:048]:     Unit of Measurement: '%'
> [10:37:17][C][sensor.dht:048]:     Accuracy Decimals: 1
> [10:37:17][C][sensor.dht:048]:     Icon: 'mdi:water-percent'
> [10:37:17][C][api:072]: API Server:
> [10:37:17][C][api:073]:   Address: 192.168.0.53:6053
> [10:37:17][C][ota:129]: Over-The-Air Updates:
> [10:37:17][C][ota:130]:   Address: 192.168.0.53:8266
> [10:37:17][C][ota:132]:   Using Password.
> [10:37:43][W][sensor.dht:141]: Requesting data from DHT failed!
> [10:37:43][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
> [10:38:13][W][sensor.dht:141]: Requesting data from DHT failed!
> [10:38:13][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 2 seconds
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 4 seconds
> WARNING Disconnected from API.
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> [10:39:22][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!
> [10:39:28][D][sensor.dht:066]: Got Temperature=20.2°C Humidity=99.9%
> [10:39:58][D][sensor.dht:066]: Got Temperature=20.6°C Humidity=99.9%
> [10:40:00][D][binary_sensor:027]: 'bathroom_extractor_fan': Sending state ON
> [10:40:28][D][sensor.dht:066]: Got Temperature=20.7°C Humidity=99.9%
>

So over time something goes wrong with the DHT code and only a full restart cures it for a while.
That's bad, I'm about to replace esphome with Tasmota to see if it makes things any better.

akasma74, 2019-02-26:
It's solved now. Hardware configuration issue.
According to this [document](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins), it is not advisable to use pin D4.
So I moved the sensor to D1 and voila, it works as expected!

closed 2019-02-26 by akasma74