esphome/issues #69: Error OTA Update ESP8266 12e

Maikl3, 2019-02-21:
Operating environment/Installation Hass.io

I received a repeated error related to the GPIO parameter.
mode: WAKEUP_PULLDOWN and mode: WAKEUP_PULLUP.
When using these parameters, the ESP8266 12e module remains on the network, is available to be managed, but it is no longer possible to update its configuration over the air.
The process stops at 2% or 3%
The only way out is to get the module out of the device again and flash it via USB.
And it is very difficult when the module is installed in an inaccessible place.
Maybe there is a solution to this problem?

`switch: 
  - platform: gpio
    pin:
      number: GPIO5
      mode: WAKEUP_PULLDOWN
    name: "sound"
    `
[image]

P.S I checked it on two modules of version ESP8266 12E
At first I used the parameter
mode: WAKEUP_PULLUP
On the second
mode: WAKEUP_PULLDOWN
In the second case, I needed this GPIO5 pin when switching on the module did not turn on the relay that is connected to it. How else can you get a low level when starting a module?

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-02-21:
I do not think the wakeup pin modes have anything to do with this - and from what I understand these options should not have anything to do with relays starting on power cycle (thought maybe I'm wrong)

For the OTA issue: I don't know what it si, maybe the WiFi signal is not too strong? Try power cycling the ESP, that will make it reconnect to WiFi. Also, try getting logs off of it; first by starting a new ESPHome window and showing OTA logs while uploading, and if that doesn't help I think the only way here is to get USB logs.

Maikl3, 2019-02-21:
First, thanks for the quick response and for your work. This is a very necessary project!
I have little knowledge of the module device.
But I suppose that at the moment of OTA download, this mode (wakeup_pullup / pulldown) does not allow the module state to be set correctly. Or to memorize this GPIO mode, use the memory section that is needed for OTA.
Prior to the installation of this mode, the module was beautifully flashing with OTA. The situation is repeated with another module.
The USB process is normal.
On your advice, I took a snapshot of the process in the second window.

[image]

OttoWinter, 2019-02-21:
Ok so sadly the API logs show too little information to see what's going on, but if it only happens after it has been installed that likely means the wifi signal is not too good (-73db is not that good).

So to debug this issue USB logs during the OTA upload would be needed.

Maikl3, 2019-02-26:
During the week, I will try to get the log via USB. Perhaps this will help to improve this project.

Maikl3, 2019-03-12:
I managed to make a log via usb-com during OTA.
The signal level is low but sufficient for stable reception.
Only after applying the WAKEUP_PULLUP mode, there is a problem with the subsequent OTA update.

[18:29:27] ets Jan  8 2013,rst cause:4, boot mode:(3,2)
[18:29:27]
[18:29:27]wdt reset
[18:29:27]load 0x4010f000, len 1384, room 16 
[18:29:27]tail 8
[18:29:27]chksum 0x2d
[18:29:27]csum 0x2d
[18:29:27]vbb28d4a3
[18:29:27]~ld
[18:29:27]
[18:29:27][C][ota:480]: There have been 2 suspected unsuccessful boot attempts.
[18:29:27][I][application:055]: Running through setup()...
[18:29:27][C][switch.gpio:023]: Setting up GPIO Switch 'sound'...
[18:29:27][D][switch:035]: 'sound' Turning OFF.
[18:29:27][D][switch:055]: 'sound': Sending state OFF
[18:29:27][D][switch:035]: 'sound' Turning OFF.
[18:29:27][C][wifi:029]: Setting up WiFi...
[18:29:27][I][wifi:183]: WiFi Connecting to 'tictic'
[18:29:33]
[18:29:33][0335m[C][wifI:244]:   SSID: 'tictic'
[18:29:33][C][wifi:245]:   IP Address: 192.168.1.39
[18:29:33][C][wifi:247]:   BSSID: 28:28:5D:A3:CA:B8
[18:29:33][C][wifi:248]:   Hostname: 'pow01'
[18:29:33][C][wifi:252]:   Signal strength: -75 dB ▂▄▆█
[18:29:33][C][wifi:253]:   Channel: 2
[18:29:33][C][wifi:254]:   Subnet: 255.255.255.0
[18:29:33][C][wifi:255]:   Gateway: 192.168.1.1
[18:29:33][C][wifi:256]:   DNS1: 192.168.1.1
[18:29:33][C][wifi:257]:   DNS2: 0.0.0.0
[18:29:33][C][api:023]: Setting up Home Assistant API server...
[18:29:33][C][ota:129]: Over-The-Air Updates:
[18:29:33][C][ota:130]:   Address: pow01.local:8266
[18:29:33][C][ota:132]:   Using Password.
[18:29:33][W][ota:136]: Last Boot was an unhandled reset, will proceed t
[18:29:33][I][application:090]: setup() finished successfully!
[18:29:33][I][application:098]: esphome-core version 1.11.2 compiled on Mar 12 2019, 00:23:19
[18:29:33][C][wifi:348]: WiFi:
[18:29:33][C][wifi:244]:   SSID: 'tictic'
[18:29:33][C][wifi:245]:   IP Address: 192.168.1.39
[18:29:33][C][wifi:247]:   BSSID: 28:28:5D:A3:CA:B8
[18:29:33][C][wifi:248]:   Hostname: 'pow01'
[18:29:33][C][wifi:252]:   Signal strength: -75 dB ▂▄▆█
[18:29:33][C][wifi:253]:   Channel: 2
[18:29:33][C][wifi:254]:   Subnet: 255.255.255.0
[18:29:33][C][wifi:255]:   Gateway: 192.168.1.1
[18:29:33][C][wifi:256]:   DNS1: 192.168.1.1
[18:29:33][C][wifi:257]:   DNS2: 0.0.0.0
[18:29:33][C][switch.gpio:048]: GPIO Switch 'sound'
[18:29:33][C][switch.gpio:049]:   Pin: GPIO5 (Mode: WAKEUP_PULLDOWN)
[18:29:33][C][switch.gpio:065]:   Restore Mode: Restore (Defaults to OFF)
[18:29:33][C][logger:159]: Logger:
[18:29:33][C][logger:160]:   Level: DEBUG
[18:29:33][C][logger:161]:   Log Baud Rate: 115200
[18:29:33][C][logger:162]:   Hardware UART: UART0
[18:29:33][C][api:089]: API Server:
[18:29:33][C][api:090]:   ``ress: pow01.local:6053
[18:29:33]ښjmumх129]: Over-The-Air Updates:
[18:29:33][C][ota:130]:   Address: pow01.local:8266
[18:29:33][C][ota:132]:   Using Password.
[18:29:33][W][ota:136]: Last Boot was an unhandled reset, will proceed tmode in 8 restarts[
[18:29:33][I][application:140]: First loop finished successfully!
[18:29:38][0;36 T535]: Client 'Home Assistant 0.89.1 (192.168.1.36)' connected succEssfully!
[18:31:27][I][ota:154]: Boot seems successful, resetting boot loop counter.

Maikl3, 2019-03-26:
I updated the log a bit. Definitely this mode should not be used on ESP8266 modules.

MyNameWasUnavailable, 2019-04-30:
I am having the exact same issue, have you been able to work around this?

venix1, 2019-05-04:
Ran into this as well when setting WAKEUP_PULLUP.  After flashing that firmware, OTA updates failed at 2 and 3%.  With some patience, it is possible to flash a corrected firmware using OTA safe mode.  This defaults to enabled and activates after 10 failed boots.  A failed boot is when the device is not online for a full 2 minutes per the default configuration.  Ultimately, I just attempted OTA updates 11 times, with each attempt taking about 1 minute.  By the 11th attempt it entered safe mode and updated.  I'm sure there is an easier way but trying to power cycle the device didn't produce results.

Maikl3, 2019-05-04:
Unfortunately, the problem with this mode is not solved: (
But safe mode is already good.

diraimondo, 2019-06-04:
I can confirm this problem on a Sonoff Basic (R2) using the WAKEUP_PULLUP on GPIO01.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter