esphome/issues #70: Use or ESPAsyncTCP at 1.1.3 causes issues

s00500, 2019-02-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
macOS, no docker
'.
-->

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 but makes no difference

**Affected component:**
Generation of platformio.ini

**Description of problem:**
The ESPAsyncTCP is constrainerd to 1.1.3, thes causes an install error:

```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "/usr/local/lib/python2.7/site-packages/platformio/managers/lib.py", line 352, in install
    force=force)
  File "/usr/local/lib/python2.7/site-packages/platformio/managers/package.py", line 723, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/usr/local/lib/python2.7/site-packages/platformio/managers/lib.py", line 213, in _install_from_piorepo
    if app.get_setting("enable_ssl") else dl_data['url'], requirements)
  File "/usr/local/lib/python2.7/site-packages/platformio/managers/package.py", line 527, in _install_from_url
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "/usr/local/lib/python2.7/site-packages/platformio/managers/package.py", line 601, in _install_from_tmp_dir
    pkg_dir = join(self.package_dir, target_dirname)
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/posixpath.py", line 73, in join
    path += '/' + b
UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 28: ordinal not in range(128)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

And also some weird compile issues when TLS is enabeled 

```
ESPAsyncTCP_ID305/src/tcp_axtls.c:422:1: error: unknown type name 'bool'
```

Is there any particular reason why this constrain is used? when removing it and using version 1.2.0 it works just fine!

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-02-21:
The traceback you posted is from platformio, which appears to be failing to install a package due to there being non-ascii characters in your path. Can you provide your full configuration yaml, as well as the local filesystem path you're using? I wouldn't be surprised to see compile errors if platformio can't install things...

OttoWinter, 2019-02-22:
Yes this is a platformio issue, not an esphome one (nor do I think it has anything to do with that specific version of espasynctcp).

I am not able to replicate the issue, so I cannot fix it because I would like to know what really is the issue here. I suspect in your case the fix you mentioned is to just remove the pinning, but pinning it to 1.2.0 would probably end up the same.

ESPHome *will* pin all dependencies (I do not want to have broken builds all of the sudden like with asynctcp on esp32s). Also there was/is no reason afaik to upgrade to 1.2.0 - since all the bug fixes I see there do not affect esphome (except maybe the ssl error you mentioned, but that seems weird because I'm pretty sure that worked last time i tested). Don't change a running system :)

s00500, 2019-02-22:
Ok, so I was not proposing to unpin it, just pin it to 1.2.0

When I manually go into the created dir, change the version and then run pio run it works without any of the two issues!

Here is my configuration:

```
esphome:
  name: lrfire
  platform: ESP8266
  board: nodemcuv2
  esphome_core_version: dev

wifi:
  ssid: "<SSID removed, no non ascii chars>"
  password: "<password removed, no non ascii chars>"
  fast_connect: true

logger:

ota:
  password: "<password removed, no non ascii chars>"

mqtt:
  broker: <server removed, no non ascii chars>
  username: <username removed, no non ascii chars>
  password: <password removed, no non ascii chars>
  port: 8883
  ssl_fingerprints:
    - <fingerprint removed>

binary_sensor:
  - platform: gpio
    pin:
      number: 4
      mode: INPUT_PULLUP
    name: "FireButton"

output:
  - platform: gpio
    pin: 2
    id: fireled

light:
  - platform: binary
    name: "Fireled"
    output: fireled
```

esphome is at 1.11beta3
PlatformIO, version 3.6.4
(platform espressif8266 is pinned to 1.8.0 so there should be no issues...)

What PlatformIO version are you using?

s00500, 2019-02-22:
@brandond 

Some additions here, when generating a yaml file with the wizard and running compile right after it the install of 305 works (interestingly)
When I then add mosquitto tls configuration I run into the strange compile errors. See the output here:

```
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs/testinstall/libf99/ESP8266WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
Compiling .pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
Compiling .pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/SyncClient.cpp.o
Compiling .pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/tcp_axtls.c.o
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.c:34:0:
.piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.h:87:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp);
^
.piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.c:422:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp){
^
*** [.pioenvs/testinstall/libb39/ESPAsyncTCP_ID305/tcp_axtls.c.o] Error 1
```

There is also a related closed issue to that on ESPAsyncTCP https://github.com/me-no-dev/ESPAsyncTCP/issues/82

s00500, 2019-02-25:
What PlatformIO version are you using?

rwifall, 2019-03-24:
I also have the issue where the compile fails when I try to use ssl_fingerprints with mqtt.
My setup is esphome installed using pip on a stock raspberry pi raspbian.
`Version: 1.12.1`

If you remove the ssl_fingerprints section below, then it all compiles fine.

Here is the yaml that causes the error and then the resulting error:

```
esphome:
  name: garage_temperature
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "SSID"
  password: "password"

# Enable logging
logger:

ota:
  password: 'password'

mqtt:
  broker: something.somewhere.com
  username: myusernmae
  password: mypassword
  port: 8883
  ssl_fingerprints:
    - FINGERPRINTHERE

sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Garage Temperature"
    humidity:
      name: "Garage Humidity"
    update_interval: 300s
```

Compiling causes:
```
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.1.3
|   |   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/garage_temperature/src/main.cpp.o
Compiling .pioenvs/garage_temperature/lib764/ESPAsyncTCP_ID305/tcp_axtls.c.o
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.c:34:0:
.piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.h:87:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp);
^
.piolibdeps/ESPAsyncTCP_ID305/src/tcp_axtls.c:422:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp){
^
Compiling .pioenvs/garage_temperature/libc73/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRelPacket.cpp.o
*** [.pioenvs/garage_temperature/lib764/ESPAsyncTCP_ID305/tcp_axtls.c.o] Error 1
Compiling .pioenvs/garage_temperature/libc73/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PublishPacket.cpp.o
============================================= [ERROR] Took 10.55 seconds =============================================
```

s00500, 2019-03-24:
Yep, this can be solved by updating the ESPAsyncTCP Library

tobiash, 2019-03-30:
I have the same problem and tried if the dev version (`sha256:b0d0f2f5a3d9abc716ff2b2078edfa6b1b35c84692ddbd4c028e4ca6c6e123f5`) of the esphome container would solve it, but it does not work. My yaml is mostly equivalent to @rwifall , using mqtt and the `ssl_fingerprints` option.

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d ringer
Processing ringer (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=346 @ 0.8.2
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ ^1.1.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing esphome-core
git version 2.17.1
Cloning into '/config/ringer/.piolibdeps/_tmp_installing-rTpZgu-package'...
remote: Enumerating objects: 400, done.
remote: Counting objects: 100% (400/400), done.
remote: Compressing objects: 100% (349/349), done.
remote: Total 400 (delta 51), reused 135 (delta 48), pack-reused 0
Receiving objects: 100% (400/400), 439.79 KiB | 1.74 MiB/s, done.
Resolving deltas: 100% (51/51), done.
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.13.0-dev #30da854
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/ringer/src/main.cpp.o
Generating LD script .pioenvs/ringer/ld/eagle.app.v6.common.ld
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/ringer/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
[... 18 lines cut ...]
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubCompPacket.cpp.o
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRecPacket.cpp.o
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRelPacket.cpp.o
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PublishPacket.cpp.o
Archiving .pioenvs/ringer/libd45/libESPAsyncTCP_ID305.a
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/SubAckPacket.cpp.o
Compiling .pioenvs/ringer/lib20f/AsyncMqttClient_ID346/AsyncMqttClient/Packets/UnsubAckPacket.cpp.o
Compiling .pioenvs/ringer/lib0d3/ESP8266mDNS/ESP8266mDNS.cpp.o
Compiling .pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/AsyncPrinter.cpp.o
Compiling .pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/ESPAsyncTCP.cpp.o
Compiling .pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/SyncClient.cpp.o
Compiling .pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/tcp_axtls.c.o
Compiling .pioenvs/ringer/lib6f5/Hash/Hash.cpp.o
Archiving .pioenvs/ringer/lib20f/libAsyncMqttClient_ID346.a
Compiling .pioenvs/ringer/lib6f5/Hash/sha1/sha1.c.o
In file included from .piolibdeps/ESPAsyncTCP_ID305@1.1.3/src/tcp_axtls.c:34:0:
.piolibdeps/ESPAsyncTCP_ID305@1.1.3/src/tcp_axtls.h:87:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp);
^
.piolibdeps/ESPAsyncTCP_ID305@1.1.3/src/tcp_axtls.c:422:1: error: unknown type name 'bool'
bool tcp_ssl_has(struct tcp_pcb *tcp){
^
*** [.pioenvs/ringer/lib1af/ESPAsyncTCP_ID305@1.1.3/tcp_axtls.c.o] Error 1
```

closed 2019-03-30 by OttoWinter