esphome/issues #71: WW / CW Slider not working in Home Assistant

anthonyangel, 2019-02-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio, installed on a Ubuntu 18 host

**ESP (ESP32/ESP8266, Board/Sonoff):** 

Arilux AL-LC06, which is a ESP8266

**Affected component:** 

WW / CW Light, https://esphome.io/components/light/cwww

**Description of problem:**
Light is set up as per the documentation, can see in the esphome logs that color temps are correctly set as `Min Mireds: 154.0` & `Max Mireds: 370.0` but the entity in Home Assistant shows `min_mireds: 0 max_mireds: 0`. If I select the entity in the Home Assistant UI I can see the white value slider, but am unable to change the value. Manually setting a color_temp value in the dev tools does allow me to select both extreme values, but nothing in the middle.

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: esp8266_pwm
    pin: GPIO15
    id: cool_white
    frequency: 1000 Hz
  - platform: esp8266_pwm
    pin: GPIO5
    id: warm_white
    frequency: 1000 Hz

light:
  - platform: cwww
    name: "Kitchen Island"
    cold_white: cool_white
    warm_white: warm_white
    cold_white_color_temperature: 154 mireds
    warm_white_color_temperature: 370 mireds
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-02-22:
This sounds like an issue with the HA UI; is there something going on that makes you think it's related to esphome?

brandond, 2019-02-22:
Did you by any chance initially set up the light without specifying temperature, or with it set to 0? IIRC, HA only loads the MQTT discovery data once; if you change it you need to restart HA.

anthonyangel, 2019-02-22:
I'm using the esphome API, not mqtt. Will up the log level in Hass for the component to see if there's anything else useful in there.

One quick thought, the esphome logs show the temperature as a float, I'm not sure if Hass expects that value as a float or an int

OttoWinter, 2019-02-22:
Fixed in esphome dev. You can install the beta at beta.esphome.io

closed 2019-02-22 by OttoWinter