esphome/issues #73: Adding power_save_mode: light causes node disconnection from WiFi every minute

akasma74, 2019-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io v.0.88.0

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, Wemos D1 mini

**Affected component:**
[WiFi](https://esphome.io/components/wifi#wifi-power-save-mode)

**Description of problem:**
I had some sudden disconnections and tried to sort it out. According to the docs, I added power_save_mode: light to my config, which resulted in reconnections every minute. Didn't expect it from reading the docs

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  networks:
  - ssid: '***'
    password: '***'
 #  power_save_mode: light
  manual_ip:
    static_ip: 192.168.0.53
    gateway: 192.168.0.1
    subnet: 255.255.255.0
```

**Traceback (if applicable):**

```
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
[11:53:04][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!
[11:53:10][W][sensor.dht:141]: Requesting data from DHT failed!
[11:53:10][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 4 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
[11:54:09][W][sensor.dht:141]: Requesting data from DHT failed!
[11:54:09][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[11:54:39][W][sensor.dht:141]: Requesting data from DHT failed!
[11:54:39][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
[11:55:24][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
[11:56:08][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!
[11:56:14][W][sensor.dht:141]: Requesting data from DHT failed!
[11:56:14][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
WARNING Couldn't connect to API (Error connecting to 192.168.0.53: timed out). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
INFO Successfully connected to 192.168.0.53
[11:56:47][D][api:531]: Client 'Home Assistant 0.88.0 (192.168.0.34)' connected successfully!
```

**Additional information and things you've tried:**
If I remove that variable, the node stays connected for hours.

OttoWinter, 2019-02-22:
It's not entirely clear from the direct docs, but if you click on the link you will see that this power save level can also make things worse.

I updated the docs to make that a bit more clear.

closed 2019-02-22 by OttoWinter

akasma74, 2019-02-22:
I suggest to spell it like this (there is a typo in the last word when I checked the doc in question):
only helps in some cases,
	   in other it can make things worse.