esphome/issues #74: Sudden socket error

akasma74, 2019-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io v.0.88.0

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, Wemos D1 mini

**Affected component:**

**Description of problem:**
Suddenly an error popped out in the esphomeyaml log

**Problem-relevant YAML-configuration entries:**
```yaml
```

**Traceback (if applicable):**

```
WARNING Disconnected from API.
INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
Exception in thread Thread-2213:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@2/2.7.15_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@2/2.7.15_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 1073, in run
    self.function(*self.args, **self.kwargs)
  File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 136, in func
    self._fatal_error()
  File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 242, in _fatal_error
    self.on_disconnect()
  File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 444, in try_connect
    cli.connect()
  File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 198, in connect
    self._socket.settimeout(0.1)
AttributeError: 'NoneType' object has no attribute 'settimeout'
```

**Additional information and things you've tried:**

OttoWinter, 2019-02-22:
Yes that is an error in esphome, but it's very hard to fix while we still have to use python 2 - so because it only affects logging and doesn't happen too often I'm flagging this as won't fix until python 3 is used.

akasma74, 2019-02-22:
Ok, got it. Glad it's not fatal ;)

akasma74, 2019-02-22:
And is this error kind of related to python 2 as well?

> Exception in thread Thread-29711:
> Traceback (most recent call last):
>   File "/usr/local/Cellar/python@2/2.7.15_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
>     self.run()
>   File "/usr/local/Cellar/python@2/2.7.15_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 1073, in run
>     self.function(*self.args, **self.kwargs)
>   File "/Users/dmitry/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 138, in func
>     self._refresh_ping()
>   File "/Users/dmitry/devel/pio/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 126, in _refresh_ping
>     self._ping_timer.cancel()
> AttributeError: 'NoneType' object has no attribute 'cancel'

OttoWinter, 2019-02-23:
Yes - it's a race condition; it is fixable but python 2 doesn't have the tools to do so easily. Fixing it would require a massive amount of code and fixing race conditions is notoriously difficult because they cannot be reproduced reliably.

OttoWinter, 2019-07-11:
Fixed in esphome 1.14

closed 2019-07-11 by OttoWinter