esphome/issues #75: fatal error: AsyncMqttClient.h: No such file or directory

dominicmcglinn, 2019-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

OTA upload 

**Description of problem:**
Compiling .pioenvs/loungeroom/lib7b3/AsyncTCP_ID1826/AsyncTCP.cpp.o
In file included from .piolibdeps/esphome-core/src/esphome/mqtt/mqtt_component.h:9:0,
from .piolibdeps/esphome-core/src/esphome/binary_sensor/mqtt_binary_sensor_component.h:9,
from .piolibdeps/esphome-core/src/esphome/binary_sensor/binary_sensor.h:262,
from .piolibdeps/esphome-core/src/esphome/controller.h:4,
from .piolibdeps/esphome-core/src/esphome/api/api_server.h:9,
from .piolibdeps/esphome-core/src/esphome/application.h:6,
from .piolibdeps/esphome-core/src/esphomelib/application.h:3,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/mqtt/mqtt_client_component.h:12:29: fatal error: AsyncMqttClient.h: No such file or directory

*************************************************************************
* Looking for AsyncMqttClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncMqttClient.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncMqttClient.h
*
*************************************************************************

compilation terminated.
Compiling .pioenvs/loungeroom/liba6f/WiFi/ETH.cpp.o
*** [.pioenvs/loungeroom/src/main.cpp.o] Error 1
========================== [ERROR] Took 27.47 seconds ==========================

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: loungeroom
  platform: ESP32
  board: esp32dev
  esphomelib_version: dev
  arduino_version: dev
  
wifi:
  ssid: ''
  password: ''
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.0.1.106
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.0.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  power_save_mode: light
  fast_connect: on
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ''

ota:
  password: ''

mqtt:
  broker: '10.0.1.44'
  username: 'Admin'
  password: ''

sensor:
  - platform: wifi_signal
    name: "loungeroom WiFi Signal Sensor"
    update_interval: 60s
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO27
    num_leds: 150
    rgb_order: GRB
    name: "KidsRoom Light"
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-02-22:
ESPHome 1.11.0 has just been released, can you check with that? I just checked with my install and that worked.

Also, I would not use `arduino_version: dev` on ESP8266s right now, the arduino core appears to be very unstable at the moment.

dominicmcglinn, 2019-02-22:
I saw the release but the hassio addon hadn't updated yet.

dominicmcglinn, 2019-02-23:
ive been using an old version of the addon repo so just installed new repo link from https://github.com/esphome/hassio will test and see how I go

dominicmcglinn, 2019-02-23:
1.11.0 fixed it had to edit my yaml to esphome_core_version from esphomelib_version

closed 2019-02-23 by dominicmcglinn