esphome/issues #80: Dashboard online indicator does not work across subnets

OttoWinter, 2019-02-24:
**Affected component:** ESPHome Dashboard
**Description of problem:**

The dashboard now uses multicast DNS for the online indicator, for normal setups that works fine since multicast DNS is sent to the entire subnet the ESPHome server is in.

However, if the ESPs and the dashboard are in *different* subnets mDNS does not work since multicast UDP apparently only works in a single subnet (kinda makes sense).

So everybody that has that sort of set up sees constant "offline" in the dashboard even though the node can be pinged. **This is entirely visual - the operation of the node is not affected in any way**.

I still want to keep mDNS the default because it is likely the easiest solution for the "normal" user - for power-user setups with different subnets etc there will be a new option to force using ICMP ping packets for node status (as was done before 1.11).

This will be addressed, but not in this 1.11 release - I don't want to introduce new features in a patch release. Also because this is entirely visual (and before 1.10 you didn't even have status indicators)

Workarounds:

 - Some routers have the option to relay mDNS over subnets
 - Follow the instructions here: https://esphome.io/guides/faq#docker-reference

2016for, 2019-02-24:
Awesome. Thanks for the perfect explanation.

OttoWinter, 2019-03-05:
Closed in https://github.com/esphome/esphome/pull/461

closed 2019-03-05 by OttoWinter