esphome/issues #81: String scalar not parsed properly?

john-arvid, 2019-02-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266Generic

**Affected component:**

**Description of problem:**
Setup wizard does not parse input properly.
In my special case I use a single quote in my wifi name. This is being parsed as it is to the yaml file. 
Since the yaml file is being created using single quotes for the scalar this causes problems.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'John-Arvid's'
  password: 'xxx'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xxx'

ota:
  password: 'xxx'

```

**Traceback (if applicable):**

```
INFO Reading configuration...
ERROR Error while reading config: while parsing a block mapping
  in "/config/esphome/test.yaml", line 7, column 3
expected <block end>, but found '<scalar>'
  in "/config/esphome/test.yaml", line 7, column 21
```

**Suggestion:**
Adding another single quote or change the single quoted scalar to a double quoted scalar fixes this problem. Maybe this should be added in the setup wizard?

OttoWinter, 2019-02-24:
The setup wizard just evaluates a quick format string for the sample config (https://github.com/esphome/esphome/blob/2e5fd7e90dd792e4aeb3802bc656f3f19588ccf5/esphome/wizard.py#L47-L61).

I'm tempted to mark this as a won't fix - it's rather easy to fix in the YAML by the user itself and it would be quite difficult to fix in the generator code. The reason is that you can get arbitrarily complex with escaping. What if the user uses double quotes, what if the user has emojis etc etc.
And we can't use the real yaml dumper because I want to have a very specific style of the base config + comments.

Maybe the best "solution" would just be to change the default to double quotes. At least that's probably less likely to affect users.

john-arvid, 2019-02-24:
I agree, both with the won't fix and default to double quotes. 
And if double quotes are inserted it could be escaped with \

Found a good article here: [http://blogs.perl.org/users/tinita/2018/03/strings-in-yaml---to-quote-or-not-to-quote.html](url)

At least I have opened this issue and other might find it at a later time.

closed 2019-03-05 by OttoWinter

gitolicious, 2019-04-13:
Too bad my wifi password actually contains double quotes... Now I ended up with an invalid config generated by the wizard. For me it might be an easy fix, for others this might be a disappointment on their first touch to ESPHome and shy them away.

john-arvid, 2019-04-13:
I checked a bit more regarding this problem and I think I found a solution that works with everything except three single quotes after another. (Cause the solution is to put it in three singel quotes)
I will create a pull request.

john-arvid, 2019-04-13:
Or not today, never done a pull request before so I would have to ask someone else to do it for me.
Just replace
```
wifi:
  ssid: "{ssid}"
  password: "{psk}"
```

with

```
wifi:
  ssid: '''{ssid}'''
  password: '''{psk}'''
```
and test if it works

gitolicious, 2019-04-14:
Unfortunately this doesn't work either. The yaml file created by the wizard acually includes
```yaml
wifi:
  ssid: '''my " ssid'''
  password: '''my " psk'''
```
which looks ok, but the generated C++ code is
```cpp
wifiap.set_ssid("'my \042 ssid'");
wifiap.set_password("'my \042 psk'");
```