esphome/issues #82: after update to 1.11.1 unable to connect

bpmittal, 2019-02-24:
yesterday i update my esphome , removed the repository then install the new repo then again install 
esphome, 
then i saw many of my nodes showing offline but was connecting and working ok , then i see this warning 
```
INFO Reading configuration...
WARNING The esphomeyaml section has been renamed to esphome in 1.11.0. Please replace 'esphomeyaml:' in your configuration by 'esphome:'.
```

then i only changed the esphomeyaml: to esphome:
and then update it with OTA , but after that i am unable to connect my node any more , i have erased and write with USB but same unable to connect it anyhow .

then i thought my node could be faulty , then i build new setting and all the configurationsfor another 2 of my nodes which were working , but after writing through USB all of them didnt work , 

today i removed esphome completely from my hassio then again install it but same problem , all the other nodes with old esphome are working in this same hassio and same network .

help required. if any other details required i can give it here .

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

novikovav69, 2019-02-25:
I have the same problem. Some devices have been updated and turned on, but some are missing from the network.

bpmittal, 2019-02-25:
i have found the solution of this problem , after update esphome to 1.11.1 , any wifi network without password is not connecting , as in my case my wifi network is working on mac filtering .

please rectify this problem in esphome .

novikovav69, 2019-02-26:
I have a password protected network, but the problem is the same. Doing an update and losing the device. My configuration:
`esphome:
  name: s4
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'Aurelia2_4'
  password: '*******'
  fast_connect: true
  manual_ip:
    static_ip: 192.168.101.214
    gateway: 192.168.101.1
    subnet: 255.255.255.0

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: 'Poliglot57'

ota:
  password: 'Poliglot57'

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "s4 Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "s4Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes`

w2cker, 2019-02-26:
I have the same problem. Did an ota update on a sonoff basic. Now wifi cannot connect anymore. The problem might be BSSID 00:00:00:00:00

Serial log:
[0;33m[W][wifi:406]: Restarting WiFi adapter...[0m
[0;36m[D][wifi:262]: Starting scan...[0m
[0;36m[D][wifi:277]: Found networks:[0m
[0;32m[I][wifi:310]: - 'fryslan' [5m(C8:CB:B8:F3:C5:71) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;36m[D][wifi:311]:     Channel: 1[0m
[0;36m[D][wifi:312]:     RSSI: -81 dB[0m
[0;32m[I][wifi:310]: - 'fryslan' [5m(C8:CB:B8:F3:D2:B1) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;36m[D][wifi:311]:     Channel: 11[0m
[0;36m[D][wifi:312]:     RSSI: -82 dB[0m
[0;32m[I][wifi:310]: - 'fryslan' [5m(C8:CB:B8:F3:D3:B1) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;36m[D][wifi:311]:     Channel: 6[0m
[0;36m[D][wifi:312]:     RSSI: -84 dB[0m
[0;36m[D][wifi:315]: - [5m'wemos3'[6m [5m(C8:CB:B8:F3:C5:70) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'ZiggoD7F29'[6m [5m(38:B1:DB:78:04:26) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'Ziggo'[6m [5m(3A:B1:DB:78:04:27) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'H369AA61C07'[6m [5m(7C:39:53:A6:1C:07) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'THUIS WAP 5Hz'[6m [5m(84:4B:F5:64:F4:27) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m''[6m [5m(54:FA:3E:B9:CF:A1) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'ZiggoDE466F8'[6m [5m(AC:22:05:87:F8:6F) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'Ziggo'[6m [5m(3A:43:1D:38:A8:5F) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'Ziggo1961214'[6m [5m(38:43:7D:38:A8:5F) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'CelenoInitialAP5DDDB3'[6m [5m(54:FA:3E:5D:DD:B3) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'Bosma_EXT'[6m [5m(A0:63:91:BF:CD:4E) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'Ziggo-gastDE466F8'[6m [5m(AE:22:35:87:F8:6F) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'HZN240434070'[6m [5m(54:FA:3E:B3:39:61) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'devolo-739'[6m [5m(30:D3:2D:9B:97:39) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:315]: - [5m'STED'[6m [5m(00:13:10:1F:29:53) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;32m[I][wifi:183]: WiFi Connecting to 'fryslan'...[0m
[0;33m[W][wifi:741]: Event: Disconnected ssid='fryslan' bssid=00:00:00:00:00:00 reason='AP Not Found'[0m
[0;33m[W][wifi:376]: Error while connecting to network.[0m
[0;33m[W][wifi:406]: Restarting WiFi adapter...[0m
[0;32m[I][wifi:183]: WiFi Connecting to 'fryslan'...[0m
[0;33m[W][wifi:741]: Event: Disconnected ssid='fryslan' bssid=00:00:00:00:00:00 reason='AP Not Found'[0m
[0;33m[W][wifi:376]: Error while connecting to network.[0m
[0;33m[W][wifi:406]: Restarting WiFi adapter...[0m

w2cker, 2019-02-28:
I have been doing some more testing. If I add sensors, no wifi can be found anymore. If I remove all but 
esphome:
logger:
wifi:
api:
ota:
from the device_configuration.yaml file wifi is connecting again. Maybe some cpu overload?

w2cker, 2019-03-04:
I have done some more testing with a sonoff basic. 
Based on a clean Kali linux machine, with a clean install of esphome. I also updated the install of platformio.

The problem is the logger. When the logger is enabled Wifi cannot connect. When the logger is disabled all is working fine. Ota is also working fine without the logger.

OttoWinter, 2019-03-05:
So the original issue from @bpmittal has been fixed in 1.11.2 - connecting to open wifi networks.

@w2cker With what log level is that?

w2cker, 2019-03-06:
@OttoWinter The config is just logger: The level will be default, debug.
I will another device to check if its only the Sonoff basic, or a more general problem.

mizhgan, 2019-03-22:
Hello.
I can confirm same issue as @w2cker said.
I have 6 Sonoffs Basic, one of them is from old revision (with orange label). The old one works just fine with standart config:
```
esphome:
  name: sonoff_basic_1
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: '*******'
  password: '*******'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic 1 Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic 1 Relay"
    pin: GPIO12
    id: relay

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "Sonoff Basic 1 Green LED"
    output: basic_green_led
```
The new ones (with black labels) can`t connect to WiFi with logger component enabled. When i commented out logger from config and manually reflash this Sonoffs - they are start working like a charm.

shadowdrake, 2019-03-23:
Coming to add - having the same problem with a Sonoff Basic.

If I flash it via UART it works fine. If I make any configuration file changes and upload OTA the device will no longer connect to my wifi network. There is no error message. No other settings have changed. For example I flashed the config below via UART and the device works. I added the text_sensor to try it out and flashed via OTA and this is what I get in my log repeating until the end of time. The device never connects until it is flashed via UART again.

[image]

My config file for this device is this:

```
esphome:
  name: sonoff_basic01
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "KittenWar"
  password: "$password"
  ap:
    ssid: "ESP_sonoff_basic01"

# Enable logging
logger:

# Enable Home Assistant API
#api:

ota:

web_server:

mqtt:
  broker: $broker
  username: homeassistant
  password: $password
#  birth_message:
#  will_message:
  discovery: 

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic 1 Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic 1 Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "Sonoff Basic 1 IP Address"
```

OttoWinter, 2019-03-23:
@shadowdrake that is interesting, logger should not affect any of the internals - except if the silicon itself has some kind of issue when having both UART0 and wifi enabled (unlikely, because almost all fws have uart0 enabled).

Can you try enabling fast_connect wifi mode, and if possible also get some verbose logs (the disconnect reason is printed there).

tdolder, 2019-03-24:
I have found the issue at least in my case.
It seems that it goes wrong if you do not specify a level for the logger component.

with this code de module won't connect to wifi:
wifi:
> `wifi:
  ssid: Name
  password: passwd

logger:

api:
  password: '!passwd'`

With this code everything works like a charm:

`wifi:
  ssid: Name
  password: passwd

logger:
  level: VERBOSE

api:
  password: '!passwd`

solidminds, 2019-04-10:
I'm new to ESPHome and I only flashed a NodeMCU and a Sonoff Basic till this moment. Both with simpel standard settings from guide. 
The NodeMCU is working like a charm. No problems connecting.
The Sonoff Basic had Tasmota on it, but I used the compile option in ESPHome, created a binary and  flashed it from Tasmota. Everything worked..... until I flashed the Sonoff Basic for the second time, from the ESPHome Dashboard over OTA. No connection anymore. 
Cut the power on the Sonoff a few times, even restarted my router. Nothing.
Running the 31st March update, 1.12.2

Problem only on Sonoff Basic. Did exactly the same with my NodeMCU and dont have any problems.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter