esphome/issues #83: Unable to use Nextion display buttons

AalianKhan, 2019-02-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
'.
-->
running esphome version 1.11.1
**ESP (ESP32/ESP8266, Board/Sonoff):**

I am using a wemos D1 mini with a 2.4 inch Nextion Display
**Affected component:**

Nextion Touch Component/Nextion Display Component

**Description of problem:**
I am unable to use any buttons with the Nextion Display. esphome doesn't seem to see when I touch the button. Also when I use Lambda to populate some of the components for the display it gives errors but you can see it clearly did the task correctly because you can see it on the display. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: nextion
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'My SSID'
  password: 'MY PWD'
  manual_ip:
    static_ip: 192.168.0.51
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

uart: 
  rx_pin: D7
  tx_pin: D8
  baud_rate: 9600

display:
  - platform: nextion
    id: display
    lambda: |-
      it.set_component_text("t0", "hello");

binary_sensor:
  - platform: nextion
    page_id: 0
    component_id: 2
    nextion_id: display
    id: button
    on_release:
      then:
        lambda: |-
          id(display).set_component_text("t0", "pressed");

```

**Traceback (if applicable):**

```

```
**Logs**
```
[E][uart:228]: Reading from UART timed out at byte 1!
[W][display.nextion:039]: Nextion returned no ACK data!
[W][display.nextion:111]: Sending command 't0.txt="hello"' failed because no ACK was received
```
**Additional information and things you've tried:**

closed 2019-02-27 by AalianKhan

zarya, 2019-03-08:
@AalianKhan Did you solve this issue and if you did what did you do to solve it?

AalianKhan, 2019-03-08:
@zarya a intelligent guy named t-bone got this solution

@OttoWinter @Aalian I got the dual state buttons to work by modifying what was sent to esphome. I made the follow changes for the press component using a printh command.

Make sure to convert the ID to hex when creating the string.

65 = Touch Event
00 = Page No.
0B = ID 11
01 = Press  or 00=Release

[16:27:35][D][display.nextion:193]: Got touch page=0 component=11 type=PRESS
[16:27:35][D][binary_sensor:037]: 'bt0': Sending state ON
[16:27:36][D][display.nextion:193]: Got touch page=0 component=11 type=RELEASE
[16:27:36][D][binary_sensor:037]: 'bt0': Sending state OFF
[16:27:37][D][display.nextion:193]: Got touch page=0 component=4 type=PRESS
[16:27:37][D][binary_sensor:037]: 'bt1': Sending state ON
[16:27:38][D][display.nextion:193]: Got touch page=0 component=4 type=RELEASE
[16:27:38][D][binary_sensor:037]: 'bt1': Sending state OFF

AalianKhan, 2019-03-08:
[image]