esphome/issues #86: Rotary encoder not behaving as expected

michel72, 2019-02-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.88.1, esphome 1.11.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini

**Affected component:**

Rotary encoder

**Description of problem:**

I have the rotary encoder component configured as below. The issue is that when turning the encoder the value goes up to 254, but if you keep turning like 20 "clicks" extra, the value remains 254, but when turning back, you would have to turn 20 clicks backwards before the value starts decreasing from 254 to 253, 252 and so on. The same happens below 0.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: rotary_encoder
    name: "Rotary Encoder"
    pin_a: D6
    pin_b: D7
```
**Additional information and things you've tried:**
I also tried:

```
    filters:
      - or:
        - debounce: 0.1s
        - delta: 10
      - lambda: |-
         if (int(x)  < 0) return 0;
         if (int(x) > 100) return 100;
         return int(x);
```
Bit this only limits to 100, with the exact same issue.

The rotary encoder component is not very usefull this way. I would expect the component to start lowering the value immediately after turning back, and not after rotating to amount of clicks exceeding the Maximum value of 254 (or when using the filter: 100)

I'm personally looking for a value of 0 tot 25 where 0 is the absolute lower value and 25 the absolute higher value. If exceeding 25 and then rotating back, I expect the value to lower straight away.

If my explanation is not clear, please say so, I will make a video then :)

OttoWinter, 2019-02-26:
That sounds like an issue with your rotary encoder. ESPHome itself does not limit the values in any way (well except below -4294967297 or above 4294967296), so the 254/0 issue you're seeing likely is something with the rotary encoder. Also try experimenting with the `resolution` parameter.

As for limiting the value into a range, that does seem like something the platform should support, will add it in the future.

michel72, 2019-02-26:
I am using the well known KY-040, and they all behave the same way.

It there any other way I could achieve the above untill you implement this new feature? Otherwise I would have to go back to easyesp for now since the wife can’t conrol the lights now :(

OttoWinter, 2019-02-26:
something like this lambda:
```yaml
static int last_value = 0;
if (x < last_value) {
  // decrement
  last_value = std::max(last_value - 1, 0);
} else if (x > last_value) {
  // increment
  last_value = std::min(last_value + 1, 25);
}
return last_value;
```
(not checked)

michel72, 2019-02-26:
Thanks, I did just check this and I'm afraid it doesn't work. I have the same issue:

Here is the log:
```
[22:08:31][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 20.00000 steps with 0 decimals of accuracy
[22:08:31][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 21.00000 steps with 0 decimals of accuracy
[22:08:31][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 22.00000 steps with 0 decimals of accuracy
[22:08:31][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 23.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 24.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:32][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:33][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:33][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:33][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:34][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:34][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:34][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:35][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:36][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:36][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:36][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:36][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:36][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:37][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:37][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:37][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:37][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
[22:08:37][D][sensor.sensor:131]: 'Woonakmer 1 Rotary': Sending state 25.00000 steps with 0 decimals of accuracy
```

When I reach the 25 limit, it keeps logging 25, and when reversing, I need to walk back all these "25" cycles before it goes back to 24, 23 etc.

michel72, 2019-02-28:
Will this be considered a bug, or do I have to create a feature request?
I'm more ore less desperately looking for a solutions as these encoders als build in my wallboxes and reflashing them back to espeasy (and eventually back to esphome) would be quit difficult.

OttoWinter, 2019-03-05:
It's already implemented in dev, so no need for a bug/feature request.

I will close this issue as fixed/implemented.

closed 2019-03-05 by OttoWinter

HA-TB303, 2019-03-11:
I'm afraid it's does not seem to be working. I put a remark on your commit.

HA-TB303, 2019-03-17:
This still doesn't work correctly.

After the latest fix in DEV the rotary encoder again works on rotating, but now the min/max value does not seem to work as intended.

```
# encoder 1
sensor:
  - platform: rotary_encoder
    name: "Woonkamer 1 Rotary 1"
    id: woonkamer1rotary1
    pin_a: D6
    pin_b: D7
    filters:
      - or:
        - debounce: 0.1s
        - delta: 10
    resolution: 1
    min_value: 1
    max_value: 25
    on_value:
      then:
        - homeassistant.service:
            service: light.turn_on
            data_template:
              entity_id: light.woonkamer_groep
              brightness_pct: "{{ brightness_1 | int }}"
            variables:
              brightness_1: 'return id(woonkamer1rotary1).state * 4;'
```

When rotating left the lowest I can get is:

[19:29:01][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 25.00000 steps with 0 decimals of accuracy

And it will only go up ;)

[19:29:32][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 28.00000 steps with 0 decimals of accuracy
[19:29:32][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 32.00000 steps with 0 decimals of accuracy
[19:29:33][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 32.00000 steps with 0 decimals of accuracy
[19:29:33][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 46.00000 steps with 0 decimals of accuracy
[19:29:33][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 51.00000 steps with 0 decimals of accuracy
[19:29:33][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 51.00000 steps with 0 decimals of accuracy
[19:29:34][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 57.00000 steps with 0 decimals of accuracy
[19:29:34][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 60.00000 steps with 0 decimals of accuracy
[19:29:34][D][sensor.sensor:100]: 'Woonkamer 1 Rotary 1': Sending state 62.00000 steps with 0 decimals of accuracy