esphome/issues #87: What ESP board is this LED controller based on?

maxtheking, 2019-02-27:
So a while back i bought a few generic Wi-Fi LED controllers on eBay, i posted images below. They currently use the Tuya cloud, but i want to convert them using tuya-convert so i can control them locally with Home Assistant instead of them reaching out to some chinese Tuya server everytime i turn them on for example. Does anyone know what ESP board they're based on? Can't find any info about them on the internet. 
[image]
[image]

brandond, 2019-02-27:
Open it up and find out. Even if you can't find a specific board for it, you can always use one of the generic board presets that match the available flash size (esp01_1m, etc). At the very least you'll need to figure out what pins the LED outputs map to - either by trial and error, or by tracing with a multimeter.

maxtheking, 2019-02-28:
Opened up the controller, can't find any info on what controller it is though. Had to force the casing open, but this one is bricked so it doesn't matter anyway. I'll try the generic Sonoff-Tasmota on my other ones using Tuya-Convert. 
[image]
[image]

OttoWinter, 2019-03-01:
Without the name or anything else there's no way we can help here (not even sure if it's an ESP, the PCB looks a bit weird).

I will close this issue since this bug tracker is for bugs, not general questions

closed 2019-03-01 by OttoWinter

CinoNY, 2019-03-28:
@maxtheking I can confirm you can flash this with tuya-convert. Haven't gotten far with it yet, but i'm in process of adding the device to HA via mqtt.