esphome/issues #91: esphomeyaml cannot compile

tinysnake, 2019-03-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, d1_mini
**Affected component:**

basic setup with no components.

**Description of problem:**

My device is RPI3 B, running HassOS 2.10,
Hassio Supervisor version 145,
Home Assistant Version 0.88.2.

I started the esphomeyaml add-on with protection mode on.
After creating a new configuration, I clicked the compile button, a while later the compile process is failed.
Please see the log below.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: d1_01
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'ssid'
  password: 'password'

mqtt:
  broker: 'x.x.x.x'
  username: 'username'
  password: 'password'

# Enable logging
logger:

ota:
  password: 'password'
```

**Traceback (if applicable):**

```
INFO [esphomeyaml.__main__] Generating C++ source...
INFO [esphomeyaml.__main__] Compiling app...
INFO [esphomeyaml.__main__] Running:  platformio run -d /config/esphomeyaml/esp_d1_01
Processing esp_d1_01 (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.5.3 #2693649
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.1.8 #d416194
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
Compiling .pioenvs/esp_d1_01/libb0d/FastLED/FastLED.cpp.o
Compiling .pioenvs/esp_d1_01/libb0d/FastLED/bitswap.cpp.o
Compiling .pioenvs/esp_d1_01/libb0d/FastLED/colorpalettes.cpp.o
Compiling .pioenvs/esp_d1_01/libb0d/FastLED/colorutils.cpp.o
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_memory_used: symbol not found
*** [.pioenvs/esp_d1_01/libb0d/FastLED/FastLED.cpp.o] Error 127
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_memory_used: symbol not found
*** [.pioenvs/esp_d1_01/libb0d/FastLED/bitswap.cpp.o] Error 127
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_memory_used: symbol not found
*** [.pioenvs/esp_d1_01/libb0d/FastLED/colorpalettes.cpp.o] Error 127
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++: _obstack_memory_used: symbol not found
*** [.pioenvs/esp_d1_01/libb0d/FastLED/colorutils.cpp.o] Error 127
========================== [ERROR] Took 19.20 seconds ==========================
```

**Additional information and things you've tried:**
problem remains after I create another configuration

OttoWinter, 2019-03-01:
you're not using the latest version of this addon. Please see the changelog on esphome.io for updated install instructions.

tinysnake, 2019-03-01:
You're right, I added the wrong repository. Sorry!

closed 2019-03-01 by tinysnake