esphome/issues #92: [Errno 111] Connection refused

hubgitbb, 2019-03-03:
Operating environment/Installation (Hass.io/Docker/pip/etc.):
Hass.io

ESP (ESP32/ESP8266/Board/Sonoff):
ESP8266

Affected component:
Teckin SP22

Description of problem:
I flashed the teckin sp22 OTA described like here https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html
that worked fine but if i try to upload a configuration it cant connect 

INFO Reading configuration...
INFO Detected timezone 'GMT' with UTC offset 0
INFO Detected timezone 'GMT' with UTC offset 0
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/p104
Processing p104 (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.11.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size .pioenvs/p104/firmware.elf
Checking size .pioenvs/p104/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  44.9% (used 36800 bytes from 81920 bytes)
PROGRAM: [====      ]  38.5% (used 394000 bytes from 1023984 bytes)
========================= [SUCCESS] Took 10.76 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.6.54
ERROR Connecting to 192.168.6.54:8266 failed: [Errno 111] Connection refused

brandond, 2019-03-03:
What did you you upload with tuya-convert? Out of the box it comes with tasmota which is fine, but you can't replace tasmota with esphome via OTA upload.

hubgitbb, 2019-03-04:
ok i got it. it was tasmota. i replaced it with esphome via browser. thx

closed 2019-03-04 by hubgitbb