esphome/issues #93: esphome as hassio addon and /dev/ttyAMA0 dows not turn up-

blurkis, 2019-03-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running esphome as an addon to hassio on a pi3. All up2date. 
Connected a sonoff pow2 to the gpio on a turned off pi3. Booted the pi as normal and
when logging in with ssh I can see that it detects /dev/ttyAMA0 which I assume is my sonoff.
Though esphome in the addon can not detect it. The only choices are OTA or 
/dev/ttyUSB0 (FT230x) which I assumed is my connected conbee-stick.

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

blurkis, 2019-03-04:
I have used esphome to flash some nodemcu so I know that it works. And those turn up in the droplist of esphome in the esphome addon to hassio. The problem being that AMA0 does not turn up though its recognized (as far as I know, the logs says so) by the underlying system of hassio.

OttoWinter, 2019-03-05:
What is the problem with using an FTDI adapter? Also, esphome would theoretically support that but I opted to manually exclude it from the dropdown list because 99% of users would be confused by it and would think it's the FTDI adapter they just plugged in. Also don't want to add an option because I want to keep the number of options minimal.

blurkis, 2019-03-05:
Well, first off its a problem because I don't own a cable like the one you
suggest. I have ordered one but has not yet arrived. So I thought that I
would use the gpio instead since I got the impression that it should work.
Knowing it doesn't is ok. I can wait for cable. Maybe some how the docs
should say this doesn't work. Esptool as such supports it so it's easy to
think it should work.

Thanks for a great product. I use it for nodemcu and those support flashing
from normal USB cable.

OttoWinter, 2019-03-24:
I looked again and esphome does not suppress ttyAMA0, I only did that while initially building the dashboard.

This means this is an external issue in Hass.io, apparently ttyAMA0 is not passed in with `auto_uart` for addons.

I will therefore close this issue.

closed 2019-03-24 by OttoWinter