esphome/issues #95: Sonoff Basic, wifi not connecting after removing removing piece of YAML

yniezink, 2019-03-05:
When I remove the switch from the YAML below, wifi won't connect anymore on the sonoff. Putting it back and then flashing with UART device makes it work again.

No compile or YAML validation issues.

`esphome:
  name: light_couch
  platform: ESP8266
  board: esp01_1m

# Wifi settings
wifi:
  ssid: 'Niezink'
  password: 'vissenkom'

# Enable logging
logger:

# Enable Home Assistant API
api:

# Enable Over the Air updates
ota:

# Enable web server (to be turned off later)
web_server:
  port: 80

# Binary sensor, triggered by the physical button on the device
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
    on_press:
      - light.toggle: light

# Switch to control the relay
switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

output:
  - platform: gpio
    id: switch_relay
    pin:
      number: GPIO12

# Register the switch as a binary light
light:
  - platform: binary
    id: light
    name: "Licht bij de bank"
    output: switch_relay

# Status led
status_led:
  pin:
    number: GPIO13
    inverted: yes`

OttoWinter, 2019-03-05:
The switch and output both point to the same pin. Please try removing the gpio switch.

yniezink, 2019-03-05:
Hi Otto, the problem is that when I remove the switch, wifi won't connect anymore. The code above is working. The problem occurs when I remove the switch.

OttoWinter, 2019-03-05:
Are you sure about that? It sounds very unlikely to me that that would be the issue here.

yniezink, 2019-03-05:
I'm very sure, I tried it 3 times... I don't understand it either.

bruxy70, 2019-03-24:
I have similar issues. Here is the current one:
I have a config that works (on Sonoff Basic)

```
esphome:
  name: sonoffd
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: <MYSSID>
  password: <MY PASSWORD>
    
logger:
api:
  password: <PASSWORD>
ota:
  password: <PASSWORD>

binary_sensor:
  - platform: gpio
    internal: true
    id: jidelna_button
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      then:
      - switch.toggle: jidelna_fake

switch:
  - platform: template
    internal: true
    optimistic: true
    id: jidelna_fake
    turn_on_action:
    - light.turn_on: jidelna
    - light.turn_off: jidelna_led
    turn_off_action:
    - light.turn_off: jidelna
    - light.turn_on: jidelna_led

light:
  - platform: binary
    name: "Jídelna"
    id: jidelna
    output: jidelna_relay
  - platform: monochromatic
    internal: true
    output: jidelna_led_diode
    id: jidelna_led
    
output:
  - platform: gpio
    pin: GPIO12
    id: jidelna_relay
  - platform: esp8266_pwm
    id: jidelna_led_diode
    pin:
      number: GPIO13
      inverted: True

```
Then, I add this:
```
    on_click:
      min_length: 50ms
      max_length: 500ms
      then:
      - homeassistant.service:
          service: scene.turn_on
          data:
            entity_id: scene.vecere

```
And it won't connect to WiFi. I have connected it to Arduino and this is the console:

>  [0;33m[W][wifi:429]: Restarting WiFi adapter... [0m
>  [0;36m[D][wifi:271]: Starting scan... [0m
>  [0;36m[D][wifi:286]: Found networks: [0m
>  [0;32m[I][wifi:321]: - 'bruxynet'  [5m(74:4D:28:2A:DC:5F)  [6m [0;32m▂▄▆█ [0m [0m
>  [0;36m[D][wifi:322]:     Channel: 9 [0m
>  [0;36m[D][wifi:323]:     RSSI: -40 dB [0m
>  [0;32m[I][wifi:321]: - 'bruxynet'  [5m(74:4D:28:2A:B9:BC)  [6m [0;33m▂▄ [0;37m▆█ [0m [0m
>  [0;36m[D][wifi:322]:     Channel: 1 [0m
>  [0;36m[D][wifi:323]:     RSSI: -69 dB [0m
>  [0;32m[I][wifi:321]: - 'bruxynet'  [5m(74:4D:28:2A:DE:4B)  [6m [0;33m▂▄ [0;37m▆█ [0m [0m
>  [0;36m[D][wifi:322]:     Channel: 8 [0m
>  [0;36m[D][wifi:323]:     RSSI: -77 dB [0m
>  [0;36m[D][wifi:325]: -  [5m'Bedroom.a' [6m  [5m(FA:8F:CA:9E:5D:07)  [6m [0;33m▂▄ [0;37m▆█ [0m [0m
>  [0;36m[D][wifi:325]: -  [5m'Barunka' [6m  [5m(98:DE:D0:A5:FC:A9)  [6m [0;31m▂ [0;37m▄▆█ [0m [0m
>  [0;32m[I][wifi:167]: WiFi Connecting to 'bruxynet'... [0m
>  [0;33m[W][wifi:399]: Error while connecting to network. [0m
>  [0;33m[W][wifi:429]: Restarting WiFi adapter... [0m

And so on...

I had similar issues with other pieces of YAML removed or added. A working switch suddenly stopped working. When I undo the change and upload the fw via USB (after turning off the circuit breakers and digging into the wall), it works again.

OttoWinter, 2019-03-24:
@bruxy70 That must be just random. I maybe see how removing logger or stuff like that could help things, but `on_click` definitely doesn't change wifi behavior.

bruxy70, 2019-03-24:
Yes, I know it is weird, but it is 100% replicable. I just tested again!
When I un-comment the on_click automation, it won't connect on WiFi.
I have tried with a sonoff basic r2, and there it works fine.
Before, it did it on another sonoff basic when I added esp8266_restore_from_flash: true.
And before that when I added a switch for the led diode (now have it as light and it works fine)
I am excited by ESPhome, well done. But this is frustrating :(

arkest1, 2019-04-14:
I have similar issues. When I add a sensor to my code it will not connect to wifi, if I remove the sensor it works normal. this works:
`
esphome:
  name: zonnecollector_regeling
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  networks:
  - ssid: 'ssid'
    password: 'password'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

#Enable dallas
binary_sensor:
  - platform: gpio
    pin: D2
    name: "Living Room Window"
    device_class: heat

dallas:
   - pin: D5

switch:
  - platform: gpio
    name: "pomp relay"
    pin:
      number: D1
      inverted: True
    id: relay_pump
  - platform: gpio
    name: "klep relay"
    pin:
      number: D2
      inverted: True
    id: relay_valve
`
but adding the dallas or any other sensor it fails:
`esphome:
  name: zonnecollector_regeling
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  networks:
  - ssid: 'ssid'
    password: 'password'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

#Enable dallas
binary_sensor:
  - platform: gpio
    pin: D2
    name: "Living Room Window"
    device_class: heat

switch:
  - platform: gpio
    name: "pomp relay"
    pin:
      number: D1
      inverted: True
    id: relay_pump
  - platform: gpio
    name: "klep relay"
    pin:
      number: D2
      inverted: True
    id: relay_valve`

HFsi, 2019-04-14:
Try the fast connect option in WiFi core component.....I had WiFI issues with 3 different Sonoff devices and enabling it worked for me

arkest1, 2019-04-14:
This helped for me, although i do not understand why, but for now I can continue, thanks.

HFsi, 2019-04-14:
I don´t understand either, but maybe could help Otto find why

michenew, 2019-05-16:
I had Wifi issues with 2 Sonoff Basic after changing some configuration and uploading via OTA or UART.
I tried fast connect option in Wifi. Nothing.
I tried to disable logger. Nothing.
I tried logger: level: verbose. Working!
I tried logger: level: error. Working!
I tried again logger: level: debug. Nothing.
I tried again logger: level: verbose. Working!

PeteCondliffe, 2019-06-18:
I have a couple of sonoff basics that use GPIO14.
Without it it works fine and with it it fails to connect to WiFi

adding

`  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: true
    id: wall_switch
    on_press:
      - switch.toggle: relay
    on_release:
      - switch.toggle: relay`

changing logger: level to verbose as @michenew  describes works

OttoWinter, 2019-06-19:
Moving discussion to https://github.com/esphome/issues/issues/455

closed 2019-06-19 by OttoWinter