esphome/issues #100: Connectivity issues after upgrading from 1.10.1 to 1.11.2

mplinuxgeek, 2019-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

OS: Fedora 29
Python: 3.7
Install method: pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

- Tuya RGBW GU10 Globe, same PCB and hardware config as RGBW AiLight
- Wemos Mini D1 with WS2812B shiled (the newer 7 LED version) and DS18B20

**Affected component:**

Unsure, both devices share most of the same components except MY9291 on the globe and Dallas on the Wemos.

I updated a bunch of devices from 1.10.1 to 1.11.2 last night and a couple of them experienced issues.

The attached configs are the devices in question:
 - rgbw_gu10_02 is a GU10 RGBW globe that uses the same PCB as RGBW AiLights and shipped with Tuya firmware.
[rgbw_gu10_02.yaml.txt](https://github.com/esphome/issues/files/2933890/rgbw_gu10_02.yaml.txt)
 - bedroom_lamp is a Wemos Mini D1 with the Wemos RGB LED shield (7 WS2812Bs) and a DS18B20.
[bedroom_lamp.yaml.txt](https://github.com/esphome/issues/files/2933891/bedroom_lamp.yaml.txt)

Prior to updating both of these devices worked properly, after updating they both did 2 different things:
 - rgbw_gu10_02 would appear as unavailable in HA and offline in the dashboard, and I could not ping it or connect to the web interface, after connecting to the globe with a USB-Serial adapter (and tearing up pads in the process but still usable), I found it was constantly looping doing a wifi scan to find the configured networks, it would not detect the network and just keep repeating the 2 networks it could see which had blank SSID (normally it sees about 7 networks, 3 of which are in my house), I enabled the hot spot on my phone and the globe wouldn't detect it either. NOTE: I have an idential globe named rgbw_gu10_01 which didn't have the same problems.
 - bedroom_lamp would connect to wifi but would not connect to mqtt, I added reboot_timeout: 60s to the config and it would reboot after 60s of no mqtt connectivity but still not connect to mqtt after rebooting, occasionally it would connect but disconnect right away.

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Traceback (if applicable):**

```
```

Diagnosis steps of rgbw_gu10_02 using a USB-Serial adapter:
 - Reflashed 1.11.2, still failed to connect to wifi
 - Wiped flash using the erase_flash option of esptool and reflashed 1.11.2, still failed
 - Located the .bin file from the original 1.10.1 firmware I configured for the globe and flashed it with esptool, this immediately solved the connection problem and the globe responded normally to commands from Home Assistant, the dashboard showed it as online and I could connect to it's web interface again.

At this point I uninstalled 1.11.2 and installed 1.10.1, recompiled and reflashed bedroom_lamp which resolved the mqtt connectivity issues. To futher confirm I recompiled and reflashed rgbw_gu10_02 successfully.

I don't have any logs handy but I'm willing to upgrade to 1.11.2 again and test with the bedroom lamp (not the rgbw gu10 globe as I may not be able to recover it again without removing the flash chip) to gather anything you might need to diagnose this further.

I should also point out that I have 3 or 4 other devices that were upgraded to 1.11.2 and they didn't have connectivity issues.

OttoWinter, 2019-03-06:
Could you post your wifi config please and the wifi scan results. You can blank out the ssids/bssids/passwords.

mplinuxgeek, 2019-03-06:
The configs were attached above, sorry if it wasn't clear enough:
[rgbw_gu10_02.yaml.txt](https://github.com/esphome/issues/files/2933890/rgbw_gu10_02.yaml.txt)
[bedroom_lamp.yaml.txt](https://github.com/esphome/issues/files/2933891/bedroom_lamp.yaml.txt)

OttoWinter, 2019-03-06:
Ah sorry didn't see that (only looked at the paste config section)

So you're not using any hidden network or similar. Let me guess: The warning you're seeing in the logs is "AP not found" right?

I think it may be related to this change: https://github.com/esphome/esphome-core/blob/dev/src/esphome/wifi_component.cpp#L603 (if you have set a password esphome will now only connect to non-open networks)

Do you know what authentication type your network uses? Is it WEP/WPA+WPA2/WPA2 and TKIP/CCMP?

mplinuxgeek, 2019-03-06:
No open or hidden networks, I use WPA/WPA2 mixed mode on a OpenWRT flashed router.

I'm not sure its related as it wasn't listing available networks during the scan, I've got a heap of spare ESP01's and a few spare Wemos, I'll flash a bunch with the same config as the globe to see if I can replicate the issue. I feel like it's deeper than a simple check for open networks.

nnmalex, 2019-03-13:
I’ve got similar issues - one NodeMCU and two sonoff were working fine prior to the update but now all show as unavailable.
At the same time HA would also show devices as unavailable but I can log in to each device’s web server fine. 
If I reboot either device it would briefly come up in ESP Home web interface as available but soon after gets unavailable again. I had no changes to my setup besides updating HA and ESP Home to latest versions.

OttoWinter, 2019-03-13:
@nnmalex No, that is a different issue - here the device is not connecting to the network in the first place.

nnmalex, 2019-03-13:
> @nnmalex No, that is a different issue - here the device is not connecting to the network in the first place.

Ah, OK, sorry. Will open new ticket.

tdolder, 2019-03-23:
I’ve got similar issues after update to 1.11.2 (2 x sonoff basic). The other 10 sonoffs basic are still on 1.12.1 and working.

when I remove _logger:_  it connects again.

with _logger:_ enabled the console output:

[15:22:39][I][wifi:167]: WiFi Connecting to 'I-Family'...
[15:22:41][W][wifi:399]: Error while connecting to network.
[15:22:41][W][wifi:429]: Restarting WiFi adapter...
[15:22:41][I][wifi:167]: WiFi Connecting to 'I-Family'...
[15:22:47][W][wifi:399]: Error while connecting to network.
[15:22:47][W][wifi:429]: Restarting WiFi adapter...

OttoWinter, 2019-03-23:
@tdolder For that please see https://github.com/esphome/issues/issues/82

cliffordwhansen, 2019-06-06:
Continuing from #409 
@OttoWinter So here is a weird one, I deleted almost all of the config
```
substitutions:
  devicename: lounge_floor_light
  upper_devicename: Lounge Floor Light

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.18.1.100
    gateway: !secret network_gateway
    subnet: 255.255.255.0
logger:
api:
```

And now it works, and HA has all the sensors from the previous setup.

cliffordwhansen, 2019-06-06:
Not sure what was wrong, but after copying the example and changing the settings (SSID, Password, name) it is working, so I had something bad in my config

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter