esphome/issues #101: Upload to board silently fails with Docker for Mac

raccettura, 2019-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
DOIT ESP32 DEVKIT V1

**Affected component:**

**Description of problem:**
Can't upload to board

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
```
$ docker run --rm -v "${PWD}":/config --device=/dev/$(readlink tty.SLAB_USBtoUART) -it esphome/esphome livingroom.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d livingroom
Processing livingroom (platform: espressif32@1.5.0; framework: arduino; board: esp32doit-devkit-v1)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 25 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphome-core> 1.11.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <WiFi> 1.0
|   |-- <Preferences> 1.0
|   |-- <AsyncTCP> 1.0.1
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
Retrieving maximum program size .pioenvs/livingroom/firmware.elf
Checking size .pioenvs/livingroom/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  23.6% (used 77220 bytes from 327680 bytes)
PROGRAM: [========  ]  84.3% (used 1381686 bytes from 1638400 bytes)
============================================================================================= [SUCCESS] Took 17.63 seconds =============================================================================================
INFO Successfully compiled program.
INFO Resolving IP address of livingroom.local
ERROR Error resolving IP address of livingroom.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving address with mDNS: Did not respond. Maybe the device is offline.
```

AFAIK it doesn't appear to ever upload or do anything to the board.  The output is identical regardless of if the board is connected or not.  So I suspect it's not ever uploading.

brandond, 2019-03-06:
This doesn't look like a silent failure, you've got three lines of error output at the end indicating that it can't resolve the hostname using mDNS, so the OTA upload failed. If this is your first time flashing the firmware, you need to connect and configure a serial adapter - you can't do OTA until the device is already running esphome and connected to your network.

raccettura, 2019-03-06:
I did point to the device as shown in the command entered, which was derived from following:
https://esphome.io/guides/getting_started_command_line.html#installation

raccettura, 2019-03-06:
Closing this.  I think the real issue is the documentation should likely denote the need to flash before starting the getting started.

closed 2019-03-06 by raccettura

brandond, 2019-03-06:
What does that readlink subshell expansion give you? I suspect it's not evaluating to a valid device path, which is why it's trying to do an OTA upload. If it doesn't detect a valid serial device, it won't prompt and just goes right into OTA mode.

OttoWinter, 2019-03-06:
https://esphome.io/guides/getting_started_command_line.html#first-uploading

raccettura, 2019-03-07:
Going to close this out as I got this going by using compile then espupload.