esphome/issues #102: Output switch has different behavior when esp unit is powered on when switch is on vs. off

MatthewFlamm, 2019-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

venv in Ubuntu
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32doit-devkit-v1
**Affected component:**

output switch using Native HA integration

**Description of problem:**
When esp unit is powered on while the switch is off in HA, the configuration works as expected.  I get chirps of light of the right length of time and they can be adjusted in HA in near realtime!  If I power on the esp unit while the switch is on in HA, the light is either always on or always off.  If I change the values for the chirplength and chirpdelay in HA, it doesn't change the bevahior, still always on/off. Also I only get this behavior when I'm using the sensor values for chirplength and chirpdelay. If I hard code them, then it works no matter how the esp is powered on.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: homeassistant
    name: "led_delay"
    entity_id: input_number.front_door_led_chirpdelay
    id: led_delay
  - platform: homeassistant
    name: "led_length"
    entity_id: input_number.front_door_led_chirplength
    id: led_length
output:
  - platform: gpio
    id: "output_led"
    pin: 12
switch:
  - platform: output
    name: "Front Door LED"
    id: front_door_led_switch
    output: output_led
    on_turn_on:
      - while:
          condition:
            switch.is_on: front_door_led_switch
          then:
            - output.turn_on: output_led
            - delay: !lambda return id(led_length).state;
            - output.turn_off: output_led
            - delay: !lambda return id(led_delay).state;
    on_turn_off:
      - output.turn_off: output_led
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
If I use defined lengths and not sensors

OttoWinter, 2019-03-06:
Do not connect a switch and a light to the same output. They both try to take control of the output and the result is undefined behavior (same as if you'd connect two lights to the same output).

MatthewFlamm, 2019-03-06:
I don't understand.  I don't have a light connected to the output, only the switch.  Am I missing something?

OttoWinter, 2019-03-06:
Ah ok I misread.

So the reason it does not work is that when the ESP powers up, the value of `led_delay`/`led_length` is not yet known and thus equals `nan` (not a number), so the delay waits forever.

On the first boot, the output is off and so the delay never happens. After that boot the state is restored (but not the input_boolean values) and the ESP goes into this infinite delay mode.

Try this:

```yaml
output:
  - platform: gpio
    id: "output_led"
    pin: 12
switch:
  - platform: template
    name: "Front Door LED"
    id: front_door_led_switch
    turn_on_action:
      - while:
          condition:
            switch.is_on: front_door_led_switch
          then:
            - output.turn_on: output_led
            - delay: !lambda return id(led_length).state;
            - output.turn_off: output_led
            - delay: !lambda return id(led_delay).state;
    turn_off_action:
      - output.turn_off: output_led
```

MatthewFlamm, 2019-03-06:
I can see why it might be confusing since I used the word light, but should have disambiguated from light switch in esphome.

I will try this, although it isn't clear to me why a template switch should not have a problem but an output switch would.  I thought about the unknown state during startup, but this does not explain to me why after the state *is* available and the switch is turned off then on, it still doesn't work

MatthewFlamm, 2019-03-06:
After thinking about it for more than 2 seconds, I can see why it might not work even when turned off then on.  I'm still baffled on why a template switch would behave differently.

OttoWinter, 2019-03-06:
template switches do not restore state (and they cancel the old action when turn on/off is called)

MatthewFlamm, 2019-03-06:
Thanks, very subtle.

MatthewFlamm, 2019-03-08:
When I change it exactly to what you suggested, the led does not turn on at all.  But it works when I switch back to output switch, with `output: output_led`.

Do I need s

MatthewFlamm, 2019-03-08:
I don't know about this current issue with template switch, but is it possible to implement an option to cancel old action when output switch is turned on/off.  I assume this would fix the original issue.

For now, I intend to put a check for unavailable state, and if so, provide default value. This is the best path forward anyway, so I would say this isn't high priority for me.  The use case where this would be important would be if I want to transition from a very slow chirp, although it wouldn't be a chirp at that point, to a very fast one with no delay.  In that case, we don't want to have to wait until the very slow chirp finishes it's action before implementing the new behavior.

OttoWinter, 2019-03-09:
> I don't know about this current issue with template switch, but is it possible to implement an option to cancel old action when output switch is turned on/off. I assume this would fix the original issue.

That's what scripts are for, there you have the ability to control their stop/on state.

MatthewFlamm, 2019-03-10:
Thanks!!! This does what I want. I'd suggest that somehow this is cleared up in the documentation.  The only info on scripts in the current documentation is that it allows for reuse of code.  Otherwise, I consider this fixed on my side.  I will close it.

```
sensor:
  - platform: homeassistant
    name: "led_length"
    entity_id: input_number.front_door_led_length
    id: led_length
  - platform: homeassistant
    name: "led_delay"
    entity_id: input_number.front_door_led_delay
    id: led_delay
output:
  - platform: gpio
    id: "output_led"
    pin: 12
switch:
  - platform: output
    name: "Front Door LED"
    id: front_door_led_switch
    output: output_led
    on_turn_on:
      then:
        - script.execute: chirp_led
    on_turn_off:
      then:
        - script.stop: chirp_led
script:
  - id: chirp_led
    then:
      - while:
          condition:
            switch.is_on: front_door_led_switch
          then:
            - output.turn_on: output_led
            - delay: !lambda return id(led_length).state;
            - output.turn_off: output_led
            - delay: !lambda return id(led_delay).state;
```

closed 2019-03-10 by MatthewFlamm